realsenseai/realsense-ros #1305: bug: inconsistent argument `add_plug` in urdf and launch files

hanzheteng, 2020-07-26:
The most recent merged pull request #1173 brings a new bug for launch files. 

https://github.com/IntelRealSense/realsense-ros/blob/6a9cc994e3a9f63ddff1938a86ff04ca990d2551/realsense2_description/urdf/_d435.urdf.xacro#L145

This line brings a new bug -- `add_plug` is not defined by default in this file, and not introduced as an argument in launch file:

https://github.com/IntelRealSense/realsense-ros/blob/6a9cc994e3a9f63ddff1938a86ff04ca990d2551/realsense2_camera/launch/rs_d435_camera_with_model.launch#L105

I revert to version 2.2.15 and it works just fine. 

Not sure if this same issue happens to file `_d415.urdf.xacro`. Please double check. Thanks.

doronhi, 2020-07-26:
Thanks for noticing and investigating. We'll check.

doronhi, 2020-09-06:
Hi, with #1335 by @lucarinelli merged, can this issue be closed?

hanzheteng, 2020-09-06:
Sure. I have no problem with the previous version and am not able to test the latest version. If you have tested and feel good, go ahead and close it.

closed 2020-09-07 by RealSenseCustomerSupport