realsenseai/realsense-ros #1306: could you comment on occupancy package, please?

AndreV84, 2020-07-27:
could you comment on occupancy package, please?
Dockerfile
```
FROM nvcr.io/nvidia/l4t-base:r32.4.3
RUN apt-get update -y
RUN apt install -y wget software-properties-common apt-utils  curl git
RUN /bin/bash  -c 'echo "deb http://packages.ros.org/ros/ubuntu bionic main" > /etc/apt/sources.list.d/ros-latest.list'
RUN  apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key
RUN apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
RUN add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
RUN apt-get update -y
ENV TZ America/Los_Angeles
RUN DEBIAN_FRONTEND="noninteractive" apt-get install -y tzdata git curl cmake wget mc mlocate 
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-melodic-desktop clang-format ros-melodic-desktop-full python-pip python3-pip  python-rosdep python-rosinstall python-rosinstall-generator python-wstool build-essential 
RUN apt update -y
#RUN rm -rf /var/lib/apt/lists
RUN  apt-get install -y librealsense2-dev librealsense2-utils
RUN apt install -y ros-melodic-ddynamic-reconfigure
RUN dpkg-reconfigure --frontend noninteractive tzdata
RUN rm -rf /var/lib/apt/lists
WORKDIR /home/nvidia/ws
RUN wget https://github.com/IntelRealSense/librealsense/archive/v2.36.0.tar.gz && tar -xvf v2.36.0.tar.gz
RUN cd librealsense-2.36.0 && ls && mkdir build && cd build && cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true && make -j8 && make install
RUN mkdir src && cd src &&  git clone https://github.com/intel-ros/realsense.git -b development 
#  && git clone https://github.com/slamtec/rplidar_ros
COPY realsense-ros/occupancy src/occupancy
RUN rosdep init && rosdep update

RUN /bin/bash -c '. /opt/ros/melodic/setup.bash;  cd /home/nvidia/ws; rosdep install --from-paths src --ignore-src; catkin_make'
```

Full error
```
docker run -it --rm --net=host -e DISPLAY=$DISPLAY --ipc=host --privileged -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE occupancy:latest /bin/bash -c '. /opt/ros/melodic/setup.bash; source devel/setup.bash; roslaunch occupancy occupancy_live_rviz.launch'
... logging to /root/.ros/log/8944c85e-cdba-11ea-9d76-00044bcb9a37/roslaunch-jetson-agx001-54.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jetson-agx001:44051/

SUMMARY
========

PARAMETERS
 * /DepthOfInterestMax: 2.0
 * /DepthOfInterestMin: 0.2
 * /HeightOfInterestMax: 0.05
 * /HeightOfInterestMin: -0.05
 * /d400/realsense2_camera/accel_fps: 0
 * /d400/realsense2_camera/accel_frame_id: d400_accel_frame
 * /d400/realsense2_camera/accel_optical_frame_id: d400_accel_optica...
 * /d400/realsense2_camera/align_depth: True
 * /d400/realsense2_camera/aligned_depth_to_color_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/allow_no_texture_points: False
 * /d400/realsense2_camera/base_frame_id: d400_link
 * /d400/realsense2_camera/calib_odom_file: 
 * /d400/realsense2_camera/clip_distance: -2.0
 * /d400/realsense2_camera/color_fps: 30
 * /d400/realsense2_camera/color_frame_id: d400_color_frame
 * /d400/realsense2_camera/color_height: 480
 * /d400/realsense2_camera/color_optical_frame_id: d400_color_optica...
 * /d400/realsense2_camera/color_width: 640
 * /d400/realsense2_camera/depth_fps: 30
[... 248 lines cut ...]
[ INFO] [1595601182.257573136]: Device with port number 1-4 was found.
[ INFO] [1595601182.561576496]: Device with serial number  was found.

[ INFO] [1595601182.561728149]: Device with physical ID 2-1-2 was found.
[ INFO] [1595601182.561792727]: Device with name Intel RealSense D435 was found.
[ INFO] [1595601182.562865855]: Device with port number 2-1 was found.
[ INFO] [1595601182.597407208]: Device with serial number  was found.

[ INFO] [1595601182.597545902]: Device with physical ID 2-3-7 was found.
[ INFO] [1595601182.598060512]: Device with name Intel RealSense T265 was found.
[ INFO] [1595601182.599275565]: Device with port number 2-3 was found.
[ERROR] [1595601182.694879718]: The requested device with serial number T265_SN and device name containing t265 is NOT found. Will Try again.
 24/07 07:33:07,957 WARNING [547205673344] (types.cpp:78) set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
 24/07 07:33:07,957 ERROR [547205673344] (ds5-factory.cpp:823) RGB Extrinsic recovery routine failed
 24/07 07:33:07,978 WARNING [547205673344] (types.cpp:78) hwmon command 0x15( 20 0 0 0 ) failed (response 0= Success)
 24/07 07:33:07,978 WARNING [547205673344] (ds5-factory.cpp:669) Cannot read RGB calibration table
 24/07 07:33:07,978 WARNING [547205673344] (types.cpp:78) Calibration data invald, buffer too small : expected 16 , actual: 0
[ INFO] [1595601187.978966725]: Device with serial number  was found.

[ INFO] [1595601187.979229998]: Device with physical ID 1-4-10 was found.
[ INFO] [1595601187.979398644]: Device with name Intel RealSense D435I was found.

```

_Originally posted by @AndreV84 in https://github.com/IntelRealSense/realsense-ros/issues/1287#issuecomment-663577090_

AndreV84, 2020-07-29:
```
FROM nvcr.io/nvidia/l4t-base:r32.4.3
RUN apt-get update -y
RUN apt install -y wget software-properties-common apt-utils  curl git
RUN /bin/bash  -c 'echo "deb http://packages.ros.org/ros/ubuntu bionic main" > /etc/apt/sources.list.d/ros-latest.list'
RUN  apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key
RUN apt-key adv --keyserver keys.gnupg.net --recv-key C8B3A55A6F3EFCDE ||  apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
RUN add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
RUN apt-get update -y
ENV TZ America/Los_Angeles
RUN DEBIAN_FRONTEND="noninteractive" apt-get install -y tzdata git curl cmake wget mc mlocate 
RUN DEBIAN_FRONTEND="noninteractive" apt install -y ros-melodic-desktop clang-format ros-melodic-desktop-full python-pip python3-pip  python-rosdep python-rosinstall python-rosinstall-generator python-wstool build-essential 
RUN apt update -y
#RUN rm -rf /var/lib/apt/lists
RUN  apt-get install -y librealsense2-dev librealsense2-utils
RUN apt install -y ros-melodic-ddynamic-reconfigure ros-melodic-rtabmap-ros
RUN dpkg-reconfigure --frontend noninteractive tzdata
RUN rm -rf /var/lib/apt/lists
WORKDIR /home/nvidia/ws
RUN wget https://github.com/IntelRealSense/librealsense/archive/v2.36.0.tar.gz && tar -xvf v2.36.0.tar.gz
RUN cd librealsense-2.36.0 && ls && mkdir build && cd build && cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true && make -j8 && make install
RUN mkdir src && cd src &&  git clone https://github.com/intel-ros/realsense.git -b development 
#  && git clone https://github.com/slamtec/rplidar_ros
COPY realsense-ros/occupancy src/occupancy
RUN rosdep init && rosdep update

RUN /bin/bash -c '. /opt/ros/melodic/setup.bash;  cd /home/nvidia/ws; rosdep install --from-paths src --ignore-src; catkin_make
```

RealSenseSupport, 2020-07-31:
@AndreV84  Could you please provide more detailed information about this ticket? 

Camera Model:
Firmware Version:
Operating System & Version:
Platform:
SDK Version:
ROS Wrapper Version:

AndreV84, 2020-07-31:
Hello,
Thank you for following up.
tried cameras D453i & d435, also t265 and L515;
firmware versions latest;
OS Linux tegra;
hardware arm nvidia Jetson AGX ;
basically the error log contains firmware verrsion & sdk version etc.
You should be able to read all corresponding information from the first post that contains the full error log:

```
]: started with pid [83]
[ INFO] [1595601167.809086587]: RealSense ROS v2.2.15
[ INFO] [1595601167.809206848]: Built with LibRealSense v2.36.0
[ INFO] [1595601167.809302851]: Running with LibRealSense v2.36.0
[ INFO] [1595601167.835399418]:  
```
Maybe you could also let me know how to save point cloud dfrom combined t265&d435? from combined t265&d435&l515? how to save point cloud outputs from them? without using any GUI, but from terminal, using command?

RealSenseSupport, 2020-08-05:
@AndreV84  We have one branch about occupancy mapping. Not sure if you have read about it. Please see https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping#occupancy-mapping-with-t265-and-d435d415 
You can try based on this.

AndreV84, 2020-08-05:
Thank you for sharing the url!
I shall try it on Nvidia Jetson Xavier device.
Also within  docker container.
still odds are reasonable that it will work, huh?

AndreV84, 2020-08-05:
from devs
"The problem is rviz doesn’t address 3D reconstruction
Rviz is the real-time pose
You can see it in rviz
But you need something to make all the frames combine"
Is there any solution for the concern quoted?
Thanks

samurairepo, 2020-08-11:
> @AndreV84 We have one branch about occupancy mapping. Not sure if you have read about it. Please see https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping#occupancy-mapping-with-t265-and-d435d415
> You can try based on this.

This branch does not seem to be maintained and it did not work for me and several other people. I have tried on RPI4 and Nano with same error (Ubuntu 18, ros-melodic). If you follow the steps, you will get same error, namely for some reason T265 wont launch when D415 is launched. Seperately, they work fine in both realsense-viewer and other example lauch files.

RealSenseSupport, 2020-08-20:
@samurairepo Could you please provide more details about the error such as the log or snapshot? And what's the version of librealsense and FW used when you get the issue? Thanks!

RealSenseSupport, 2020-08-21:
@samurairepo I just tried this https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping#occupancy-mapping-with-t265-and-d435d415 on my Ubuntu18.04 setup and didn't any error. Could you please double check https://github.com/IntelRealSense/realsense-ros/blob/occupancy-mapping/README.md to see if any step missing in your testing? Thanks!

RealSenseSupport, 2020-09-04:
@AndreV84 @samurairepo Any other questions about this? Looking forward to your update. Thanks!

AndreV84, 2020-09-04:
Thank you for the update! I just notetd just I should try https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping#occupancy-mapping-with-t265-and-d435d415
May I know what is the processor instrtuction set of the computer u used? x84_64?

RealSenseSupport, 2020-09-09:
@AndreV84 My setup is Intel Core i3-3227U @1.9GHz x 4, 64bit. For your reference. Thanks!

RealSenseSupport, 2020-09-15:
@AndreV84  Did you get chance to try and any update? Thanks!

AndreV84, 2020-09-15:
lets close it as for today.

closed 2020-09-15 by AndreV84