realsenseai/realsense-ros #1307: rs-tracking-depth example won't start; Jetson AGX

AndreV84, 2020-07-27:
built newer sdk from sources and installed latest firmware to cameras
 ```
./rs-tracking-and-depth 
2020-07-24 16:53:10,003 WARNING [default] Could not open device failed to set power state
2020-07-24 16:53:10,009 WARNING [default] Could not open device Unable to open device interface
2020-07-24 16:53:10,060 WARNING [default] Could not open device failed to set power state
2020-07-24 16:53:10,071 WARNING [default] Could not open device Unable to open device interface
2020-07-24 16:53:15,050 WARNING [default] Could not open device failed to set power state
2020-07-24 16:53:15,062 WARNING [default] Could not open device Unable to open device interface
2020-07-24 16:53:15,112 WARNING [default] Could not open device failed to set power state
2020-07-24 16:53:15,120 WARNING [default] Could not open device Unable to open device interface
2020-07-24 16:53:15,159 WARNING [default] Could not open device failed to set power state
2020-07-24 16:53:15,169 WARNING [default] Could not open device Unable to open device interface
```

_Originally posted by @AndreV84 in https://github.com/IntelRealSense/realsense-ros/issues/1287#issuecomment-663729897_

AndreV84, 2020-07-27:
at some point after connection to powered usb hub it would return:
Couldn't open ./H_t265_d400.cfg

AndreV84, 2020-07-31:
we managed to run the rs-tracking-and-depth using 38 version of sdk;
is there any option to save pointloud from there?
is there any option to get information from three sensors l515, t265, d435? or from 4 sensors 2xd435, 1x265 & 1x l515 with it?

LinusNEP, 2021-08-25:
I have same problem (Couldn't open ./H_t265_d400.cfg).
Is there any solution yet?

AndreV84, 2021-08-25:
yes there was the file accessible from some third party repository, as I recollect; otherwise try to find the file at the filesystem

LinusNEP, 2021-08-25:
Great!
I got it work now.
Thanks

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a significant time period.

closed 2026-10-23 by MartyG-RealSense