realsenseai/realsense-ros #1309: Question: how to launch `realsense-viewer` with `ros-kinetic-librealsense2` library installed using apt

hanzheteng, 2020-07-27:
My current configuration: 
1. sudo apt install ros-kinetic-librealsense2     (version 2.36.0)
2. git clone current 2.2.15 version of realsense-ros package and dependencies (e.g., ddynamic reconfiguration) and catkin_make in my own workspace. 

Question: I am not able to run `realsense-viewer` in the terminal. It seems like if I install librealsense2 using apt, the `realsense-viewer` is not installed system-wide. Do I have to install librealsense2 from source in order to use `realsense-viewer`?

doronhi, 2020-07-28:
Yes. realsense-viewer is part of the package librealsense2-utils. It is not part of ros-kinetic-librealsense2.

hanzheteng, 2020-07-30:
Thanks for the info. Closing this issue.

closed 2020-07-30 by hanzheteng