realsenseai/realsense-ros #1312: saving and working with depth images in python

waspinator, 2020-08-01:
Using a realsense D435i I'm capturing data using ROS into bag files with the standard topics. In particular I have 

- `/aligned_depth_to_color/image_raw` and
- `/color/image_raw`

both of type `sensor_msgs/Image`. My understanding is that color images can be saved in whatever format is convenient (jpeg, png, ...) but that depth should be saved as .raw. I'm not sure how to save, open, or interpret .raw files. 

I'm reading the messages using rosbag and saving them to file using opencv.

```python
import cv2
import rosbag
from sensor_msgs.msg import Image
from cv_bridge import CvBridge

bridge = CvBridge()

bag = rosbag.Bag('my_ros_bag_file.bag', "r")
count = 0
for topic, msg, timestamp in bag.read_messages(topics=['/aligned_depth_to_color/image_raw', '/color/image_raw']):
    if topic == '/aligned_depth_to_color/image_raw':
        depth_img = bridge.imgmsg_to_cv2(msg, desired_encoding="passthrough")
        cv2.imwrite('depth_{}.raw'.format(count), depth_img)

    if topic == '/color/image_raw':
        color_img = bridge.imgmsg_to_cv2(msg, desired_encoding="bgr8")
        cv2.imwrite('depth_{}.jpg'.format(count), color_img)

    count += 1

```

but opencv doesn't seem to understand what I mean by "raw": ` could not find a writer for the specified extension in function imwrite`. Saving as 'png' produces a result, but from reading other issues it seems that doesn't save the correct depth data.

Once I have a method of saving into the "raw" format, how would I read it? For example, I would read a regular image file using `img = cv2.imread('image.jpg')` which would produce a numpy array for the color values. I expect something similar for a numpy value of depths.

Once I have my data saved to file, my end goal is to measure the distance to an object, it's width and its height. I'm able to extract the pixel locations of the bottom, top, left, and right locations, but I would have to convert those to real-world coordinates in order to measure the distances. I probably should try using `get_distance` and `rs2_deproject_pixel_to_point()`. Are there examples of how I would use do that using ROS topics or/and saved color and raw depth files? 

Thanks

MartyG-RealSense, 2020-08-04:
Hi @waspinator   Apologies for the delay in responding.  A range of methods for utilising raw data without ROS are suggested below.

***********************

One RealSense user took the approach of creating a script to convert the .raw file into a .ply point cloud using cv::mat:

https://github.com/IntelRealSense/librealsense/issues/3286#issuecomment-465648074

Another RealSense user provided a Python script that saves color as a PNG and saves depth as an array of scaled matrices saved as an .npy file:

https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225

Another approach for getting the depth data from a .raw file is suggested by a RealSense team member here:

https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-415451870

RealSenseSupport, 2020-09-15:
Your code is correct.

The depth image is single channel 16-bit image.

In the first snippet, the shape  of `depth_img`  is (1080, 1920). the dtype of depth_img is uint16. and it's a landscape image.
And the 1500th pixel in the 1000th row in the depth_image is depth_img[1000, 1500]

in the second snippet, the variable `z` should be depth_img[1000, 1500]
in this function, `x` is the column number. `y` is the rows count. the code should be
rs.rs2_deproject_pixel_to_point(intrinsics, [1500, 1000], z)

`rs2_deproject_pixel_to_point` is written in
https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L83

RealSenseSupport, 2020-09-27:
@waspinator Did you get it through? Any other questions about this?

waspinator, 2020-09-28:
yes, saving as `png` works well enough.

closed 2020-09-28 by waspinator

karlita101, 2021-03-22:
Is there another way of saving depth images as png with rosbag? ( Not familiar with it at all and looking for something user/beginner friendly)
Thank you