realsenseai/realsense-ros #1315: Display by rviz from bag file - No Image received

HAMMADIPRO, 2020-08-04:
I would like to display color and depth data from bag file recorded by Realsense Viewer with **rviz**.

`roslaunch realsense2_camera rs_from_file.launch rosbag_filename:="/home/yassine/Destktop/cap.bag"`

but I got No Image received

[image]

Intel Realsense D415
ubuntu version 18.04.4
rosdistro: melodic
rosversion: 1.14.6

MartyG-RealSense, 2020-08-04:
Hi @HAMMADIPRO    One cause of problems reading a bag with **rs_from_file** can be if a full **absolute** path to the bag is not specified, but you seem to have done that already with **/home/yassine/Destktop/cap.bag**

Have you confirmed that the bag is playable please?   You could test this by dragging and dropping the bag file into the **RealSense Viewer** program.

HAMMADIPRO, 2020-08-04:
Thank you for responding
It works perfectly with RealSense Viewer when I read bag file  with it

MartyG-RealSense, 2020-08-04:
Have you got any error messages, such as **Fixed Frame [camera_link] does not exist**  ?

HAMMADIPRO, 2020-08-04:
I get no error in terminal 
[image]

MartyG-RealSense, 2020-08-05:
Apologies for the delay in responding further, as I was off-shift but am back now.

The next thing you can try is following general advice in the link below for getting a bag file to be recognized by RViz:

https://answers.ros.org/question/241230/how-to-visualize-depthimages-from-rosbag-file-with-rviz/

MartyG-RealSense, 2020-08-16:
Hi @HAMMADIPRO   Do you require further assistance with this case please?

HAMMADIPRO, 2020-08-18:
I'm using a Bag file recorded with Realsense Viewer without depth Information and I still got this error !
[image]

`[ INFO] [1597708705.230401231]: publish topics from rosbag file: /home/yassine/Desktop/labtest.bag
[ INFO] [1597708705.345173883]: getParameters...
[ INFO] [1597708705.387602914]: setupDevice...
[ INFO] [1597708705.387632208]: JSON file is not provided
[ INFO] [1597708705.387643852]: ROS Node Namespace: camera
[ INFO] [1597708705.387659699]: Device Name: Intel RealSense D415
[ INFO] [1597708705.387672339]: Device Serial No: 911222061387
[ INFO] [1597708705.387684377]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-2/4-2:1.0/video4linux/video1
[ INFO] [1597708705.387697619]: Device FW version: 05.11.15.00
[ INFO] [1597708705.387710512]: Device Product ID: 0x0AD3
[ INFO] [1597708705.387728647]: Enable PointCloud: Off
[ INFO] [1597708705.387744369]: Align Depth: Off
[ INFO] [1597708705.387768961]: Sync Mode: Off
[ INFO] [1597708705.387845583]: Device Sensors: 
[ INFO] [1597708705.387918604]: Stereo Module was found.
[ INFO] [1597708705.387988171]: RGB Camera was found.
[ INFO] [1597708705.388014434]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1597708705.388029651]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1597708705.388070191]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1597708705.388086914]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1597708705.388113563]: num_filters: 0
[ INFO] [1597708705.388136303]: Setting Dynamic reconfig parameters.
[ INFO] [1597708705.390349366]: Done Setting Dynamic reconfig parameters.
[ INFO] [1597708705.391669532]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1597708705.391722045]: setupPublishers...
[ INFO] [1597708705.393719211]: Expected frequency for color = 30.00000
[ INFO] [1597708705.415887586]: setupStreams...
[ INFO] [1597708705.415976369]: insert Color to RGB Camera
[ERROR] [1597708705.416409655]: An exception has been thrown: No known base_stream found for transformations.
[FATAL] [1597708705.537778158]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-2] process has died [pid 12617, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/yassine/.ros/log/891be45a-e0e5-11ea-8391-606c6661e6b7/camera-realsense2_camera_manager-2.log].
log file: /home/yassine/.ros/log/891be45a-e0e5-11ea-8391-606c6661e6b7/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has died [pid 12618, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/yassine/.ros/log/891be45a-e0e5-11ea-8391-606c6661e6b7/camera-realsense2_camera-3.log].
log file: /home/yassine/.ros/log/891be45a-e0e5-11ea-8391-606c6661e6b7/camera-realsense2_camera-3*.log
`

MartyG-RealSense, 2020-08-18:
Looking at your last two provided logs, I note that you are using ROS wrapper **2.2.14** with librealsense SDK **2.36.0**.  It is important to try to match the librealsense version as closely as possible with the version recommended for a particular ROS wrapper version on the wrapper's **Releases** page.

In this case, wrapper 2.2.14 was designed for librealsense version 2.35.2.  The recommended wrapper version for librealsense 2.36.0 is wrapper **2.2.15**.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.15

closed 2020-09-03 by MartyG-RealSense