realsenseai/realsense-ros #1317: D435i | rs_camera.launch additional arguements realted to pointcloud stream

meghdeepj, 2020-08-05:
Realsense SDK: 2.35.2 (5.11.xx firware)
Realsense-ROS: 2.2.14-1
Ubuntu: 18.04 (Linux4Tegra, kernel 4.9)
Hardware: Nvidia Jetson Nano
Camera: Intel Realsense D435i

I am trying to stream pointcloud data from rs_camera.launch.

1. What is the difference between "filters:=pointcloud" and "enable_pointcloud:=true"? Is there any difference in computational resources used?

2. What are the options for "pointcloud_texture_stream"? What if I do not want any texture nor rgb data in pointcloud?

Thanks

doronhi, 2020-08-05:
There is no difference between filters:=pointcloud and enable_pointcloud:=true. One is kept for backwards compatibility.
The options for pointcloud_texture_stream are: 
RS2_STREAM_ANY - Will not add texture.
RS2_STREAM_COLOR
RS2_STREAM_INFRARED
RS2_STREAM_FISHEYE - available with devices that contains a fisheye sensor. Not D435i.

meghdeepj, 2020-08-05:
Thanks a lot, that solves my issue!

closed 2020-08-05 by meghdeepj