realsenseai/realsense-ros #1318: Realsense D455 device not supported

MrMixit, 2020-08-05:
Hi! 

I just received the new D455 camera and tried it out in Realsense Viewer, where it worked fine, but when launching the ROS wrapper everything crashes. The launch file is just a copy of `rs_camera.launch`, configured with the serial number of the D455. By the error message I understand this camera is not yet supported. If that is the case, do you have any ETA for when it will be? The whole terminal session is pasted below.

Best,
Adam

```
emdkaai@elxc1000145:~$ roslaunch realsense2_camera rs_camera3.launch 
... logging to /home/emdkaai/.ros/log/f3ccb6d4-d6f0-11ea-acf4-14dda9d3d3e6/roslaunch-elxc1000145-4936.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://elxc1000145:35643/

SUMMARY
========

PARAMETERS
 * /D455/realsense2_camera/accel_fps: 250
 * /D455/realsense2_camera/accel_frame_id: D455_accel_frame
 * /D455/realsense2_camera/accel_optical_frame_id: D455_accel_optica...
 * /D455/realsense2_camera/align_depth: False
 * /D455/realsense2_camera/aligned_depth_to_color_frame_id: D455_aligned_dept...
 * /D455/realsense2_camera/aligned_depth_to_fisheye1_frame_id: D455_aligned_dept...
 * /D455/realsense2_camera/aligned_depth_to_fisheye2_frame_id: D455_aligned_dept...
 * /D455/realsense2_camera/aligned_depth_to_fisheye_frame_id: D455_aligned_dept...
 * /D455/realsense2_camera/aligned_depth_to_infra1_frame_id: D455_aligned_dept...
 * /D455/realsense2_camera/aligned_depth_to_infra2_frame_id: D455_aligned_dept...
 * /D455/realsense2_camera/allow_no_texture_points: False
 * /D455/realsense2_camera/base_frame_id: D455_link
 * /D455/realsense2_camera/calib_odom_file: 
 * /D455/realsense2_camera/clip_distance: -2.0
 * /D455/realsense2_camera/color_fps: 30
 * /D455/realsense2_camera/color_frame_id: D455_color_frame
 * /D455/realsense2_camera/color_height: 480
 * /D455/realsense2_camera/color_optical_frame_id: D455_color_optica...
 * /D455/realsense2_camera/color_width: 848
 * /D455/realsense2_camera/depth_fps: 30
 * /D455/realsense2_camera/depth_frame_id: D455_depth_frame
 * /D455/realsense2_camera/depth_height: 480
 * /D455/realsense2_camera/depth_optical_frame_id: D455_depth_optica...
 * /D455/realsense2_camera/depth_width: 640
 * /D455/realsense2_camera/device_type: 
 * /D455/realsense2_camera/enable_accel: True
 * /D455/realsense2_camera/enable_color: True
 * /D455/realsense2_camera/enable_depth: True
 * /D455/realsense2_camera/enable_fisheye1: False
 * /D455/realsense2_camera/enable_fisheye2: False
 * /D455/realsense2_camera/enable_fisheye: False
 * /D455/realsense2_camera/enable_gyro: True
 * /D455/realsense2_camera/enable_infra1: True
 * /D455/realsense2_camera/enable_infra2: True
 * /D455/realsense2_camera/enable_infra: True
 * /D455/realsense2_camera/enable_pointcloud: True
 * /D455/realsense2_camera/enable_pose: False
 * /D455/realsense2_camera/enable_sync: False
 * /D455/realsense2_camera/filters: 
 * /D455/realsense2_camera/fisheye1_frame_id: D455_fisheye1_frame
 * /D455/realsense2_camera/fisheye1_optical_frame_id: D455_fisheye1_opt...
 * /D455/realsense2_camera/fisheye2_frame_id: D455_fisheye2_frame
 * /D455/realsense2_camera/fisheye2_optical_frame_id: D455_fisheye2_opt...
 * /D455/realsense2_camera/fisheye_fps: 30
 * /D455/realsense2_camera/fisheye_frame_id: D455_fisheye_frame
 * /D455/realsense2_camera/fisheye_height: 480
 * /D455/realsense2_camera/fisheye_optical_frame_id: D455_fisheye_opti...
 * /D455/realsense2_camera/fisheye_width: 640
 * /D455/realsense2_camera/gyro_fps: 200
 * /D455/realsense2_camera/gyro_frame_id: D455_gyro_frame
 * /D455/realsense2_camera/gyro_optical_frame_id: D455_gyro_optical...
 * /D455/realsense2_camera/imu_optical_frame_id: D455_imu_optical_...
 * /D455/realsense2_camera/infra1_frame_id: D455_infra1_frame
 * /D455/realsense2_camera/infra1_optical_frame_id: D455_infra1_optic...
 * /D455/realsense2_camera/infra2_frame_id: D455_infra2_frame
 * /D455/realsense2_camera/infra2_optical_frame_id: D455_infra2_optic...
 * /D455/realsense2_camera/infra_fps: 30
 * /D455/realsense2_camera/infra_height: 480
 * /D455/realsense2_camera/infra_width: 640
 * /D455/realsense2_camera/initial_reset: True
 * /D455/realsense2_camera/json_file_path: 
 * /D455/realsense2_camera/linear_accel_cov: 0.01
 * /D455/realsense2_camera/odom_frame_id: D455_odom_frame
 * /D455/realsense2_camera/pointcloud_texture_index: 0
 * /D455/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_INFRARED
 * /D455/realsense2_camera/pose_frame_id: D455_pose_frame
 * /D455/realsense2_camera/pose_optical_frame_id: D455_pose_optical...
 * /D455/realsense2_camera/publish_odom_tf: True
 * /D455/realsense2_camera/publish_tf: True
 * /D455/realsense2_camera/rosbag_filename: 
 * /D455/realsense2_camera/serial_no: 026222250686
 * /D455/realsense2_camera/tf_publish_rate: 1.0
 * /D455/realsense2_camera/topic_odom_in: odom_in
 * /D455/realsense2_camera/unite_imu_method: copy
 * /D455/realsense2_camera/usb_port_id: 
 * /rosdistro: melodic
 * /rosversion: 1.14.6

NODES
  /D455/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [4947]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f3ccb6d4-d6f0-11ea-acf4-14dda9d3d3e6
process[rosout-1]: started with pid [4958]
started core service [/rosout]
process[D455/realsense2_camera_manager-2]: started with pid [4961]
process[D455/realsense2_camera-3]: started with pid [4966]
[ INFO] [1596614097.102845786]: Initializing nodelet with 12 worker threads.
[ INFO] [1596614097.356532155]: RealSense ROS v2.2.15
[ INFO] [1596614097.356579787]: Built with LibRealSense v2.36.0
[ INFO] [1596614097.356599602]: Running with LibRealSense v2.36.0
[ INFO] [1596614097.373617290]:  
[ INFO] [1596614097.401057698]: Device with serial number 026222250686 was found.

[ INFO] [1596614097.401118715]: Device with physical ID /sys/devices/pci0000:00/0000:00:1c.5/0000:0a:00.0/usb6/6-2/6-2.3/6-2.3:1.0/video4linux/video8 was found.
[ INFO] [1596614097.401148033]: Device with name Intel RealSense D455 was found.
[ INFO] [1596614097.401690546]: Device with port number 6-2.3 was found.
[ INFO] [1596614097.401755817]: Resetting device...
[ INFO] [1596614103.496168124]:  
[ INFO] [1596614103.526421530]: Device with serial number 026222250686 was found.

[ INFO] [1596614103.526470681]: Device with physical ID /sys/devices/pci0000:00/0000:00:1c.5/0000:0a:00.0/usb6/6-2/6-2.3/6-2.3:1.0/video4linux/video8 was found.
[ INFO] [1596614103.526499766]: Device with name Intel RealSense D455 was found.
[ INFO] [1596614103.526903841]: Device with port number 6-2.3 was found.
[FATAL] [1596614103.527043841]: Unsupported device! Product ID: 0x0B5C
[D455/realsense2_camera_manager-2] process has died [pid 4961, exit code 1, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/emdkaai/.ros/log/f3ccb6d4-d6f0-11ea-acf4-14dda9d3d3e6/D455-realsense2_camera_manager-2.log].
log file: /home/emdkaai/.ros/log/f3ccb6d4-d6f0-11ea-acf4-14dda9d3d3e6/D455-realsense2_camera_manager-2*.log

```

MrMixit, 2020-08-05:
Nevermind, I added a case in the `realsense_node_factory.cpp` and a corresponding constant in `constants.h` for the PID of the D455 and now everything seems to be working as I expected it to work.

closed 2020-08-05 by MrMixit

doronhi, 2020-08-06:
Thanks for drawing my attention. I added #1321 .