realsenseai/realsense-ros #1320: It support Realsense D455 now?

Zhan-Jia-Rong, 2020-08-06:
I used "Method 1" to install the device,then i modify "device_type" to d455.
But it doesn't work.
[image]
[image]

thanks!

MartyG-RealSense, 2020-08-06:
Hi @Zhan-Jia-Rong   Version 2.2.15 of the RealSense ROS wrapper for matching with SDK 2.36.0 does not contain official support for D455.

[image]

I am not involved in ROS wrapper development.  It is likely though that official D455 support will be added in an upcoming version of the wrapper.

In the meantime, you could try setting the default device to D435i to see if it works as a temporary workaround, since the D455 is similar to the D435 and has an IMU too.

MrMixit, 2020-08-06:
Hey @Zhan-Jia-Rong! I just asked the same question in #1318 yesterday, and very recently they made a new commit that added support for this device!

MartyG-RealSense, 2020-08-06:
@MrMixit  Thanks so much for highlighting this, I hadn't seen it.

Zhan-Jia-Rong, 2020-08-07:
After I installed it using "Method 2", I will try your proposal. 
Thanks for your reply！

closed 2020-08-08 by Zhan-Jia-Rong