realsenseai/realsense-ros #1322: RealSense SDK 2.0 is not detected in realsense-ros/realsense2_camera even it is installed

golnazhabibi3, 2020-08-07:
Hi, I have installed Realsense SDK 2 from the source and real-sense-viewer and hello-realsense from the examples work fine, but when I run catkin_make clean  it gives me the following error:

CMake Warning at realsense/realsense-ros/realsense2_camera/CMakeLists.txt:37 (find_package):
  Could not find a configuration file for package "realsense2" that is
  compatible with requested version "2.37.0".

  The following configuration files were considered but not accepted:

    /usr/local/lib/cmake/realsense2/realsense2Config.cmake, version: 2.36.0
    /usr/lib/x86_64-linux-gnu/cmake/realsense2/realsense2Config.cmake, version: 2.36.0

CMake Error at realsense/realsense-ros/realsense2_camera/CMakeLists.txt:39 (message):
  

  

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases

  

-- Configuring incomplete, errors occurred!
See also "/home/catkin_ws_realsense/build/CMakeFiles/CMakeOutput.log".
See also "/home/catkin_ws_realsense/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Also, I am not sure if it is related but when I run ' modinfo uvcvideo | grep "version:"  '
   to verify the kernel is updated, I do not get the one with realsense string.. could you help me how i can fix this issue in catkin ?
Thanks!

MartyG-RealSense, 2020-08-07:
Hi @golnazhabibi3   The RealSense ROS wrapper should ideally be matched with the version of the librealsense SDK that is listed on the wrapper's **Releases** page for a particular version.

The matching ROS wrapper version for the **2.36.0** SDK is **2.12.15**:

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.15

The matching ROS wrapper version for the **2.37.0** SDK is **2.12.16**:

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.16

vibhubithar, 2020-08-09:
For some reason when  following this guide 
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages
librealsense v 2.36 is installed 

then when trying to install realsense ros wrapper by following this guide https://github.com/IntelRealSense/realsense-ros
cmake is looking for 2.37

which is causing the issue

vibhubithar, 2020-08-09:
I solved this issue by checking out the tag related to git repo for ros wrapper associated with  2.2.15

change this line in the instructions listed on  https://github.com/IntelRealSense/realsense-ros 

from 
git checkout `git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1`
to
git checkout 2.2.15

vibhubithar, 2020-08-09:
now i am facing a different issue realsense-viewer is able to detect the camera and show data but through ROS i am not able to see anything

roslaunch realsense2_camera rs_camera.launch align_depth:=true
... logging to /home/vibhu/.ros/log/8544d29e-d9d6-11ea-835e-20a787008bc2/roslaunch-vibhu-UP-APL01-RPRO-12416.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://vibhu-UP-APL01-RPRO:37499/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
[... 61 lines cut ...]
[ INFO] [1596932600.413781346]: Motion Module was found.
[ INFO] [1596932600.413891325]: num_filters: 0
[ INFO] [1596932600.413929783]: Setting Dynamic reconfig parameters.
[ INFO] [1596932605.788425162]: Done Setting Dynamic reconfig parameters.
[ INFO] [1596932605.791768489]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1596932605.792851833]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1596932605.794969951]: setupPublishers...
[ INFO] [1596932605.801955671]: Expected frequency for depth = 30.00000
[ INFO] [1596932605.895208490]: Expected frequency for color = 30.00000
[ INFO] [1596932605.965968403]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1596932606.036967934]: setupStreams...
[ INFO] [1596932606.089869994]: insert Depth to Stereo Module
[ INFO] [1596932606.090344327]: insert Color to RGB Camera
[ INFO] [1596932606.335955405]: SELECTED BASE:Depth, 0
[ INFO] [1596932606.368411930]: RealSense Node Is Up!
 08/08 17:23:26,476 WARNING [140461058549504] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/08 17:23:26,634 WARNING [140461058549504] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 08/08 17:23:26,695 WARNING [140461058549504] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 08/08 17:29:45,499 WARNING [140461058549504] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/08 17:30:13,155 WARNING [140461058549504] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 08/08 17:30:20,369 WARNING [140461058549504] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

doronhi, 2020-08-09:
The issues of installation comes from the fact that librealsense2 version 2.37.0 is still not available as a debian package. Therefor you could either clone and build it from source, or, as you have chosen, degrade realsense2_camera's version to match the available librealsense2 debian package - 2.36.0. As you are using D435i there will be no difference for you so you have chosen well.
As for not being able to see anything, could you be more specific as I do not notice anything out of the ordinary in the log. Do you not see the topics using `rostopic list`? Do you not see anything using `rostopic hz /camera/color/image_raw`? Or is there any other observation?

MartyG-RealSense, 2020-08-09:
Thanks very much for your assistance with this case @doronhi    :)

vibhubithar, 2020-08-09:
thanks for the reply @doronhi apologies if i am making newbie mistakes but i assume if the camera is publishing on the topics listed then rviz would be able to display the point cloud but i font see anything. There was a global frame error but i was able to fix it by changing one of the parameters.

for example when i run rosrun rqt_reconfigure rqt_reconfigure i dont see any values for the keys as compared to what is shown in the https://github.com/IntelRealSense/realsense-ros example

same goes for the point cloud example

closed 2020-09-03 by MartyG-RealSense