realsenseai/realsense-ros #1324: Raspberry Pi 4 realsense-ros no pointcloud

yiyinglai, 2020-08-09:
System: Linux raspberrypi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux
ROS distro: kinetic
Camera: D415
` 1) Name: Intel RealSense D415, serial number: 950122061295, update serial number: 948123020797, firmware version: 05.12.06.00, USB type: 3.2`
USB cable: official cable
librealsense 2.36.0 built with
`cmake ../ -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DFORCE_RSUSB_BACKEND=ON -DBUILD_WITH_TM2=false -DIMPORT_DEPTH_CAM_FW=false` (firmware is already latest)
realsense-ros 2.2.15

RGB image and point cloud in realsense-viewer:
[image]

When using realsense-ros:
```
pi@raspberrypi:~ $ roslaunch realsense2_camera rs_camera.launch enable_pointcloud:=true
... logging to /home/pi/.ros/log/246e19fc-da81-11ea-98fc-dca632bae7db/roslaunch-raspberrypi-11963.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:36881/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
[... 254 lines cut ...]
[ INFO] [1597006388.835181714]: Done Setting Dynamic reconfig parameters.
[ INFO] [1597006388.839013090]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1597006388.842839929]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1597006388.843021130]: setupPublishers...
[ INFO] [1597006388.858007863]: Expected frequency for depth = 30.00000
[ INFO] [1597006389.432425549]: Expected frequency for color = 30.00000
[ INFO] [1597006389.562867264]: setupStreams...
[ INFO] [1597006389.666845618]: insert Depth to Stereo Module
[ INFO] [1597006389.667064058]: insert Color to RGB Camera
[ INFO] [1597006389.987359525]: SELECTED BASE:Depth, 0
[ INFO] [1597006390.060451373]: RealSense Node Is Up!
 09/08 16:53:10,178 WARNING [2870989840] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 09/08 16:53:10,292 WARNING [2870989840] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ERROR] [1597006390.327135368]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
 09/08 16:53:10,363 WARNING [2870989840] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ERROR] [1597006390.427037053]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
[ERROR] [1597006393.878198306]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

MartyG-RealSense, 2020-08-10:
Hi @yiyinglai   If the **control_transfer returned** warning is being generated continuously, rather than just at the start of launch, then it can indicate a communication problem with the device.  If it is generated only once or twice at start-up and then does not repeat, it is safe to ignore the warning if the launch does not exit.

I looked at your launch instruction:

**roslaunch realsense2_camera rs_camera.launch enable_pointcloud:=true initial_reset:=true**

The term **enable_pointcloud:=true** has been deprecated (not recommended) for a long time, and **filters:=pointcloud** is instead recommended.  This is the format used by the point cloud example in the RealSense ROS wrapper's documentation:

https://github.com/IntelRealSense/realsense-ros#rgbd-point-cloud

Changing the point cloud term may not necesarily make a difference, but could you try it please in order to eliminate the possibility that it is an influence on the problem.  Thank you.

Also, could you expand open the **Global Options** section of the RViz options panel at the top of it to see what your Rviz has set in the **Fixed Frame** option, please?

yiyinglai, 2020-08-10:
Hi @MartyG-RealSense , thanks for you reply.

I tried using the `filters` argument, and here's the result: 
```
pi@raspberrypi:~ $ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to /home/pi/.ros/log/ecfd8066-db2f-11ea-9a1d-dca632bae7db/roslaunch-raspberrypi-2005.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:41595/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 305 lines cut ...]
	min: 0.030s max: 0.036s std dev: 0.00105s window: 28
average rate: 29.861
	min: 0.022s max: 0.060s std dev: 0.00476s window: 57
average rate: 29.793
	min: 0.022s max: 0.060s std dev: 0.00443s window: 87
```
Publishing rate of `/camera/depth/image_rect_raw`:
```
pi@raspberrypi:~ $ rostopic hz /camera/depth/image_rect_raw
subscribed to [/camera/depth/image_rect_raw]
average rate: 30.388
	min: 0.016s max: 0.042s std dev: 0.00430s window: 29
average rate: 30.218
	min: 0.016s max: 0.042s std dev: 0.00344s window: 59
average rate: 30.162
	min: 0.016s max: 0.043s std dev: 0.00325s window: 89
```
Publishing rate of `/camera/depth/color/points`:
```
pi@raspberrypi:~ $ rostopic hz /camera/depth/color/points
subscribed to [/camera/depth/color/points]
no new messages
no new messages
no new messages
```

MartyG-RealSense, 2020-08-10:
Thanks so much for the detailed report!  Are you satisfied with the results please, or do you need further assistance?

yiyinglai, 2020-08-10:
Hi @MartyG-RealSense , the problem persisted so I still need further assistance.

MartyG-RealSense, 2020-08-10:
It looks as though the color and depth images are being published in RViz.   What is the remaining problem please - that there are still **control_transfer returned** warnings?

yiyinglai, 2020-08-10:
The problem is that there is no point cloud data in topic `/camera/depth/color/points`. My apology for the confusion.

MartyG-RealSense, 2020-08-10:
Can you try using this launch term please:

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud**

and setting the Fixed Frame option in RViz to **camera_link** instead of base_link

https://github.com/IntelRealSense/realsense-ros#rgbd-point-cloud

yiyinglai, 2020-08-10:
Sure:
[image]

MartyG-RealSense, 2020-08-10:
Any luck if you **Add** a topic with the button at the bottom of the Davis option's panel?

https://github.com/IntelRealSense/realsense-ros/issues/302#issuecomment-362202818

yiyinglai, 2020-08-10:
The topic for `PointCloud2` in Rviz is `/camera/depth/color/points`, and it is indeed where the point cloud data is supposed to be published to.

The issue is not unable to visualize the point cloud topic in Rviz, but there is no data in point cloud topic. Notice this:
```
pi@raspberrypi:~ $ rostopic hz /camera/depth/color/points
subscribed to [/camera/depth/color/points]
no new messages
no new messages
no new messages
```

MartyG-RealSense, 2020-08-10:
I researched the subject further.  In the past there have been problems related to no data in camera/depth/color/points that are apparently related to the kernel.  There were a number of cases where the data would not be generated with kernel 5.0.  Here are a couple of examples:

https://github.com/IntelRealSense/realsense-ros/issues/1047
https://github.com/IntelRealSense/realsense-ros/issues/1077

Given that kernel 5.4 (which it looks as though you are using) is not yet officially supported by librealsense, I could not rule out a problem being related to it.

yiyinglai, 2020-08-10:
Thanks for pointing the direction to look into. Can you please keep this issue open? I will try with another kernel version and provide updates afterwards.

MartyG-RealSense, 2020-08-10:
I will make a note here about keeping it open.  Good luck!

doronhi, 2020-08-12:
The links @MartyG-RealSense mentions point to a timestamp issue.
Could you check the timestamps of the color and depth images. Use `rostopic echo /camera/color/image_raw/header` and `rostopic echo /camera/depth/image_rect_raw/header`. Try running in 2 separate terminals and see if the timestamps seem to match. Could you post a sample of them (say, 3-4 frames of each)?

yiyinglai, 2020-08-12:
Hi @dorodnic , I run two separate terminals and echoed those two topics, and here is what I got: 
[image]

The terminal on the left corresponds to `rostopic echo /camera/color/image_raw/header` and the one on the right `rostopic echo /camera/depth/image_rect_raw/header`. I tried to specify `-n 4` for each command but I was afraid that switching between terminals will cause extra overhead, so I used the command without `-n 4`. The screenshot was taken while the commands were still running, so the output at the bottom should be the latest messages.

Can you help me interpret this result? Thanks!

MartyG-RealSense, 2020-08-12:
@yiyinglai   I think you meant to tag @doronhi   :).

yiyinglai, 2020-08-12:
@MartyG-RealSense , thanks for pointing it out. @doronhi , @dorodnic I'm so sorry about the mistake.

doronhi, 2020-08-13:
Looking at the timestamps, I noticed that the color stream is published at 15Hz while depth stream is at 30Hz. If it was specified this way it should not have prevented the pointcloud from being created but since this is not what you specified in the launch file (according to the logs you posted previously) I wonder if the overload on your machine is not too great. Could you reduce the frame rate to 15 Hz in both streams and check again?

yiyinglai, 2020-08-13:
Indeed, I looked at the publishing frequencies of both topics if specified at 30 Hz. The publishing frequency of depth remains at 30 Hz while that of color dropped from 30 Hz to 17 Hz gradually.

I tried specifying both publishing frequencies to be 15 Hz, and here's the result:
```
pi@raspberrypi:~ $ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud initial_reset:=true depth_fps:=15 color_fps:=15
... logging to /home/pi/.ros/log/6ba0533c-dd9a-11ea-836d-dca632bae7db/roslaunch-raspberrypi-4310.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:42561/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 536 lines cut ...]
seq: 293
stamp: 
  secs: 1597346656
  nsecs: 816385100
frame_id: "camera_depth_optical_frame"
---
seq: 294
stamp: 
  secs: 1597346656
  nsecs: 883276040
frame_id: "camera_depth_optical_frame"
---
seq: 295
stamp: 
  secs: 1597346656
  nsecs: 953538890
frame_id: "camera_depth_optical_frame"
---
seq: 296
stamp: 
  secs: 1597346657
  nsecs:  17685384
frame_id: "camera_depth_optical_frame"
---
```

yiyinglai, 2020-08-13:
I then checked running `realsense-viewer` and the resolution are 1280X720 and frame rate 30 Hz. Both parameters are higer than 640X480 and 15 Hz when launching rs_camera, so maybe this is not related to the processing power of Raspberry Pi?

I found out there is erroneous timestamp even in `realsense-viewer`: the timestamp of depth stream in this screenshot is negative.
[image]

MartyG-RealSense, 2020-08-14:
@yiyinglai   I believe that there are differences between how librealsense handles the camera and how the RealSense ROS wrapper handles it.  So in ROS cases, it will often be reported that the camera is working fine in librealsense programs such as the RealSense Viewer, but may be encountering problems with streaming in the ROS wrapper.  Also, changes made by the user in librealsense programs such as the Viewer do not affect the camera's behavior in the ROS wrapper, and vice versa.  

So I would advise against comparing the Viewer with the ROS wrapper like-for-like, as they are "apples and oranges" - not directly comparable because of the different ways in which they work.

Your mention of the color FPS dropping from 30 to 17 over time reminds me of a phenomenon that I have heard of before that was related to changes in the environmental lighting causing a drag to be imposed on the camera's FPS.  Dorodnic the RealSense SDK Manager has said that FPS can be affected by auto-exposure of RGB if the scene is dark.  Is your scene well-lit, please?

yiyinglai, 2020-08-14:
@MartyG-RealSense thanks for the explanation.

The scene was not dark. I have another laptop beside the Raspberry Pi, and `realsense-ros` worked well on that laptop, so environmental lighting is unlikely the cause in this case.

doronhi, 2020-08-14:
The issue seems to be related to the timestamp. Could you check if the other timestamps shown in the realsense-viewer behave correctly? Specifically frame-timestamp but take a look at all of them. Maybe you can post 3-4 images showing the progress. We can tell how much really passed according to the progress of "frame counter".

In realsense-viewer: "stereo module"->controls->Global Time enabled: change to false. Is the "Frame timestamp" now seem to advance properly?

MartyG-RealSense, 2020-08-14:
@doronhi  Thanks so much for your reply, we were both answering at the same time.  I had just ran some tests of the RGB mode with my own D415 in the Viewer and compared the metadata to the image kindly provided by @yiyinglai .  They were almost the same, except in two respects.  I was receiving a **Hardware FPS** and a **Backend Timestamp**, whilst these values were both zero in the image of @yiyinglai .  This does suggest a problem with the metadata.

[image]

doronhi, 2020-08-14:
I hope @ev-mp can shed some light here

yiyinglai, 2020-08-15:
@doronhi @MartyG-RealSense I started building everthing again before you commented. This time with kernel 4.19 since I saw some one manged to use `realsense-ros` with kernel 4.19.

I will update the result afterwards.

MartyG-RealSense, 2020-08-15:
@yiyinglai   Thanks, we look forward to the results.

yiyinglai, 2020-08-18:
@doronhi @MartyG-RealSense Hi, thanks for your patience. I tried several times but still wasn't able to get librealsense or realsense-ros working with kernel 4.19 because opencv3 was not working either (I believe librealsense and realsense-ros has some dependencies on opencv).

After switching back to the original setup (I have a backup image, so everthing is exactly the same), I followed @doronhi 's instruciton and disabled `Global Time Enabled`, and here is the result: 
[image]

This time, the `Frame Timestamp` entry are always positive and the readings seems correct. `Hareware FPS` has non-zeros readings now but `Backend Timestamp` are still zeros.

yiyinglai, 2020-08-19:
I managed to install opencv-3.4.2 successfully last night and librealsense with kernel 4.19.
`realsense-viewer` performed similar to when built with kernel 5.4.

`Global Time Enabled`
[image]
`Global Time Disabled`
[image]

Is `Backend Timestamp` the reason I couldn't get point cloud data in ROS?

MartyG-RealSense, 2020-08-19:
Have you checked whether a point cloud is now visible in RViz, please?

I researched your question above carefully.  There was a RealSense user with timestamp problems who fixed it by adding the line below to their launch file - no harm in trying it.

 **<arg name="filters" default="pointcloud"/>**

Other than that I do not have an answer I would feel confident enough in to provide, so I defer back to @doronhi regarding your query about the effect of the backend timestamp on ROS point clouds.

yiyinglai, 2020-08-19:
Hi @MartyG-RealSense , thanks for your comment. The effect of adding `<arg name="filters" default="pointcloud"/>` to the launch file is the same as specifying it in the launch command with `filters:=pointcloud`.

I cannot see the point cloud in Rviz, because there is still no data in topic `/camera/depth/color/points`.

MartyG-RealSense, 2020-08-20:
I did further research and located another case where the backend timestamp was 0 in the Viewer.

https://github.com/IntelRealSense/librealsense/issues/5212

The RealSense team member who replied in that case said that the global timestamp is not affected by the state of the backend timestamp.

Building with RSUSB backend = on (as you indicated that you have done at the start of this case) gives good metadata support, so it is unusual that you are experiencing this problem with a backend timestamp of 0.

The RSUSB installation process is also not dependent on kernel versions or Linux versions and doesn't require patching.  I have heard of problems occurring when patching has been applied after an RSUSB backend = On installation was done (because patching is not needed with this method).  Can you confirm please whether or not you patched the installation after completing the install with RSUSB, please?  If you did patch it, I would recommend doing another build of librealsense with RSUSB but not patching it afterward.

yiyinglai, 2020-08-20:
I confirm that I build `librealsense` with `FORCE_RSUSB_BACKEND=ON` and did not patch the kernels afterwards.

korroktheslavemaster, 2020-08-21:
Hi
I'm facing the same issue. depth and color images are published, but no data on point cloud topic. I can view pointcloud in `realsense-viewer`
System: Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
OS: Raspbian Buster
ROS distro: melodic
Camera: D435i
1) Name: Intel RealSense D435i, firmware version: 05.12.06.00, USB type: 3.2
USB cable: official cable
librealsense 2.37.0 built with ros catkin command (which just calls cmake and make)
`sudo ./src/catkin/bin/catkin_make_isolated --pkg librealsense2 --install -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/melodic `

I am also getting a 0 in `Backend Timestamp`

yiyinglai, 2020-08-24:
Hi @MartyG-RealSense I tried `librealsense 2.36.0` and firmware 5.12.05.
[image]

And also `librealsense 2.35.0` and firmware 5.12.05. In this case, the metadata overlay didn't show up for depth stream.
[image]

MartyG-RealSense, 2020-09-03:
Hi @yiyinglai   Do you require further assistance with this case, please?   Thanks!

yiyinglai, 2020-09-04:
@MartyG-RealSense Yes, the problem still exists.

MartyG-RealSense, 2020-09-04:
Hi @yiyinglai  I went through the case step by step from the beginning and made notes, to try to work out the key points of the case so far.

-  The pointcloud2 topics do not publish data, stating "No new messages".  Color and depth do publish data on their topics though.
-  There are some **control_transfer returned** errors on launch.  Your process-killing exit is not due to a crash though, as you exited manually yourself after a period of up to 12 minutes.  This is presumably why color and depth were able to be successfully published in RViz.
-  You experience the problem of negative / zero timestamps when using the camera in the Viewer.
-  You have tried it with both Kernel 5.4 and kernel 4.19 and have had the same problems with both kernels.

*************

I am wondering if use of the Raspbian OS is one of the main complicating factors.  Typically, librealsense related operations tend to go smoother when a more mainstream Linux flavor such as Ubuntu is used.

justinIRBT, 2020-09-04:
FYI I have the same issue using the realsense_ros driver (with ros2 eloquent branch) on Raspberry Pi 4 using Rasbian Buster OS as discussed in the case here https://github.com/intel/ros2_intel_realsense/issues/147 (as you can see from the comments, we switched from ros2 driver to this driver's ros2 branch based on suggestion from @doronhi ).  It remains unresolved, but I was able to unblock myself by writing my own realsense node based on librealsense2, which is working fine (I used the rs-pointcloud as a starting point.)  I would prefer to be using the official driver and all the configuration options it exposes, so I am still interested in a resolution.

MartyG-RealSense, 2020-09-13:
@yiyinglai  and @justinIRBT    Can you confirm that the problem still exists for you?  Thanks very much.

justinIRBT, 2020-09-13:
Yes, it still exists

MartyG-RealSense, 2020-09-13:
I have been over the entire case from the beginning again but have not been able to find much that has not already been looked at or linked to.  Possibly one of the existing cases that has already been looked at has a useful clue if looked at with a fresh perspective.

I did find one case with similar symptoms to this one (especially in regard to 'no new messages' from the /camera/depth/color/points topic) that has not been linked to in this discussion yet.

https://github.com/IntelRealSense/realsense-ros/issues/1071

yiyinglai, 2020-09-15:
Hi @MartyG-RealSense , I am using the camera with a laptop and probably wouldn't use it with Raspberry Pi 4 afterwards. Thanks for your help. Please feel free to close this issue.

MartyG-RealSense, 2020-09-15:
Okay, thanks very much @yiyinglai

closed 2020-09-15 by MartyG-RealSense

dgrnbrg, 2021-07-05:
I am also encountering this issue, and I believe I can contribute in the debugging of it: I think that the pointcloud filter is doing something similar to the align_depth filter. That filter runs at around 2-3 FPS on a raspberry pi, and as a result, we see the queue backing up & preventing the system from making progress as @doronhi commented above.

dispectra, 2021-07-13:
I am also encountering this issue, that is: it works fine when opened with `realsense-viewer`, but no message published at `/camera/depth/color/points` when launched using `roslaunch realsense2_camera demo_pointcloud.launch`.

I use Raspberry Pi 4B 8GB, ROS Noetic, Ubuntu 20.04. Kernel 5.4.0-1038-raspi. The camera is Intel RealSense D415.

My workaround was decreasing the fps in the launch file. For the argument `depth_fps` and `color_fps`, I set it to 6 (the default was 30) (Why 6? [cf](https://www.intel.com/content/www/us/en/support/articles/000031493/emerging-technologies/intel-realsense-technology.html)). It then suddenly worked, the message published in /camera/depth/color/points 
When checked with `rostopic hz /camera/depth/color/points` the average rate was 5.408 Hz.

tldr; it works when I decrease the fps to 6.

MartyG-RealSense, 2021-07-13:
Thanks very much @dispectra for sharing the details of the workaround method that worked for you!

I would add that if users of the **D455** camera model wish to try your method in future, they should set FPS to 5, as 6 is not supported as an FPS mode on D455.

Thanks again!