realsenseai/realsense-ros #1326: Unable to launch more than 2 d435i

mcdzim, 2020-08-11:
I am working on a system that requires 4 cameras to be running into ROS melodic on Ubuntu 18.04 using either an intel nuc or jetson nano. 

Having tried with both platforms we can reliably launch the realsense-viewer and display 4 live streams (needed to check power from the hub and cables from being a bottleneck) 

When I use the multiple launch option in ros I can only get 2 to launch and if I try launch 4 then a random 2 cameras launch and the others fail often reporting an RS2 power state error 

Here is the launch file and seeing as the issue is across platform think it had to be a part of the realsense-ros wrapper

**Launch file**
```
<launch>
  <arg name="serial_no_camera1"    			default="843112071333"/> 			
  <arg name="serial_no_camera2"    			default="923322070870"/> 			
  <arg name="serial_no_camera3"    			default="919122071947"/> 			
  <arg name="serial_no_camera4"    			default="919122072218"/> 		

  <arg name="camera1"              			default="camera1"/>		
  <arg name="camera2"              			default="camera2"/>		
  <arg name="camera3"              			default="camera3"/>		
  <arg name="camera4"              			default="camera4"/>		
  <arg name="tf_prefix_camera1"         default="$(arg camera1)"/>
  <arg name="tf_prefix_camera2"         default="$(arg camera2)"/>
  <arg name="tf_prefix_camera3"         default="$(arg camera3)"/>
  <arg name="tf_prefix_camera4"         default="$(arg camera4)"/>
  <arg name="initial_reset"             default="true"/>

  <group ns="$(arg camera1)">
    <include file="$(find ziva_bringup)/config/realsense_config.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera1)"/>
      <arg name="tf_prefix"         		value="$(arg tf_prefix_camera1)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
    </include>
  </group>

  <group ns="$(arg camera2)">
    <include file="$(find ziva_bringup)/config/realsense_config.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera2)"/>
      <arg name="tf_prefix"		          value="$(arg tf_prefix_camera2)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
    </include>
  </group> -->

  <!-- <group ns="$(arg camera3)" if="$(eval serial_no_camera3 != '')">
    <include file="$(find ziva_bringup)/config/realsense_config.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera3)"/>
      <arg name="tf_prefix"		          value="$(arg tf_prefix_camera3)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
    </include>
  </group> -->

  <!-- <group ns="$(arg camera4)" if="$(eval serial_no_camera4 != '')">
    <include file="$(find ziva_bringup)/config/realsense_config.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera4)"/>
      <arg name="tf_prefix"		          value="$(arg tf_prefix_camera4)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
    </include>
  </group> -->
```

**Config file** 
```

<launch>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="tf_prefix"           default=""/>
  <arg name="json_file_path"      default=""/>
[... 149 lines cut ...]
    <param name="aligned_depth_to_color_frame_id"    type="str"  value="$(arg aligned_depth_to_color_frame_id)"/>
    <param name="aligned_depth_to_infra1_frame_id"   type="str"  value="$(arg aligned_depth_to_infra1_frame_id)"/>
    <param name="aligned_depth_to_infra2_frame_id"   type="str"  value="$(arg aligned_depth_to_infra2_frame_id)"/>
    <param name="aligned_depth_to_fisheye_frame_id"  type="str"  value="$(arg aligned_depth_to_fisheye_frame_id)"/>
    <param name="aligned_depth_to_fisheye1_frame_id" type="str"  value="$(arg aligned_depth_to_fisheye1_frame_id)"/>
    <param name="aligned_depth_to_fisheye2_frame_id" type="str"  value="$(arg aligned_depth_to_fisheye2_frame_id)"/>

    <param name="publish_tf"               type="bool"   value="$(arg publish_tf)"/>
    <param name="tf_publish_rate"          type="double" value="$(arg tf_publish_rate)"/>

    <param name="odom_frame_id"            type="str"  value="$(arg odom_frame_id)"/>
    <param name="topic_odom_in"            type="str"  value="$(arg topic_odom_in)"/>
    <param name="calib_odom_file"          type="str"    value="$(arg calib_odom_file)"/>
    <param name="publish_odom_tf"          type="bool" value="$(arg publish_odom_tf)"/>
    <param name="filters"                  type="str"    value="$(arg filters)"/>
    <param name="clip_distance"            type="double" value="$(arg clip_distance)"/>
    <param name="linear_accel_cov"         type="double" value="$(arg linear_accel_cov)"/>
    <param name="initial_reset"            type="bool"   value="$(arg initial_reset)"/>
    <param name="unite_imu_method"         type="str"    value="$(arg unite_imu_method)"/>
    <param name="allow_no_texture_points"  type="bool"   value="$(arg allow_no_texture_points)"/>

  </node>
</launch>

```

MartyG-RealSense, 2020-08-12:
Hi @mcdzim   Jetson boards, and to a much lesser extent NUCs, are known as having multicam problems with RealSense that are apparently related to a failed power state.  Version **2.35.2** of librealsense introduced improvements to how multiple cameras are handled, though the changes may not prevent problems that are related to specific models of USB hub.  There are long discussions of the subject at the links below:

https://github.com/IntelRealSense/librealsense/issues/5828
https://github.com/IntelRealSense/librealsense/issues/6379
https://github.com/IntelRealSense/librealsense/issues/6395

One possible workaround suggested within https://github.com/IntelRealSense/librealsense/issues/5828 is to plug the cameras in using a certain order of insertion.

https://github.com/IntelRealSense/librealsense/issues/5828#issuecomment-625671835

If you are launching the cameras with **rs_multiple_devices.launch** and are using serial numbers to identify the two separate cameras, then an alternative approach in ROS that may give more stable device identification may be to identify by USB port ID instead of serial number with the **usb_port_id i**nstruction.

https://github.com/IntelRealSense/realsense-ros#launch-parameters

MartyG-RealSense, 2020-09-09:
Hi @mcdzim  Do you have a further update yet, please?

MartyG-RealSense, 2020-09-18:
Hi @mcdzim   Do you need this case to be kept open for a longer period while you continue to work on issues, please?  Thanks!

MartyG-RealSense, 2020-09-26:
Case closed due to no futher comments received - feel free to re-open the case if you need to resume it at a later date.

closed 2020-09-26 by MartyG-RealSense

mcdzim, 2020-10-15:
Final fix was making sure to use Librealsense 2.36 and realsense-ros 2.2.15 on Ubuntu 18.04 and with this arrangement managed to get all 4 cameras working reliably on an intel Nuc with large externally powered USB 3.2 hub. 

Still unable to make much progress using a Jetson Nano or Xavier NX

MartyG-RealSense, 2020-10-15:
There have been a couple of recent Jetson cases on Nano and Xavier where updating CMake solved problems.  The procedure that the RealSense users in those cases used is in the link below:

https://github.com/IntelRealSense/librealsense/issues/6980#issuecomment-666858977