realsenseai/realsense-ros #1327: D435 rgbd pointcloud no color

kristfrizh, 2020-08-12:
Hello
I'm quite new with this device.
I have problem when i follow the tutorial RGBD pointcloud from here https://github.com/IntelRealSense/realsense-ros.

the problem is the pointcloud has no color in it.

[image]

My environment system is :
Linux ubuntu kernel 5.3..0 - 62 - generic
Linux 18.04 LTS
ros melodic
librealsense Version 2.36.0

Is there anything I miss?
thank you

MartyG-RealSense, 2020-08-12:
Hi @kristfrizh   Does the suggestion in the link below make any diffeence, please?

https://github.com/IntelRealSense/realsense-ros/issues/975#issuecomment-552199690

kristfrizh, 2020-08-12:
I have figured it out 
not only the color transformer must be changed but the style should be changed to points

Thank you @MartyG-RealSense for your fast reply

closed 2020-08-12 by kristfrizh

MartyG-RealSense, 2020-08-12:
You are very welcome.  I am pleased that you achieved a solution!

kristfrizh, 2020-08-12:
For some noted, we should check the kernel version when we run the package (it is for dpkms librealsense). In my case, the non supported kernel make my device suddenly disconnected and the picture is blur

the supported kernel i use is 5.3.
you can read in below https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

BenjaminMichaelGreen, 2020-10-23:
I have been struggling with this issue for a long time. Happens when I run in a virtual machine. 

The solution I just found was to add "export SVGA_VGPU10=0" to my bashrc file and reboot. Suddenly have full colour! Hope this helps.

MartyG-RealSense, 2020-10-23:
Thanks very much for sharing what worked for you with the RealSense ROS community @BenjaminMichaelGreen  !