realsenseai/realsense-ros #1328: Software synchronization between multiple devices.

angshine, 2020-08-13:
I'm using a D435i and a L515 simultaneously and I want to synchronize their messages. However, I found that even the timestamps of the two devices are basically the same (like < 10ms difference), but the acquired images from 2 devices have about 500ms difference in time. What is the recommended way to software synchronize multiple devices?

RealSenseSupport, 2020-08-17:
Hi @angshine 

Timestamps is the best way to sync the two cameras together since from a HW standpoint the two cameras cannot be synced.  The two cameras have different processors and processing time as well as taking into account the images reaching the host platform via USB, the time the image arrives can be different.

RealSenseSupport, 2020-09-11:
Hi @angshine 

Is there anything else needed on this one?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-18 by RealSenseSupport

doronhi, 2020-09-20:
Sorry for my late response.
I would also like to add that for synchronizing 2 cameras using timestamps it is important that the timestamps are given in the "Global" domain. If metadata is not available global timestamps may not be available and the the driver provides "system domain". The ROS wrapper hides the timestamp domain from the user but it may appear as a warning. Use realsense-viewer in order to make sure that "global timestamp" is available.
I don't have an L515 available right now but I'll make a point to test the time difference between a D435i and an L515 as 500ms sounds way too high.