realsenseai/realsense-ros #1329: enable_gyro/accel for SLAM D435i

germal, 2020-08-14:
Hi all , I am trying to implement SLAM following the steps on

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

My environment is 

Operating system: Ubuntu 18.04
Kernel: 4.9.140-tegra
Platform: ARM ( Jetson Nano)
RealSense ROS v2.2.11
LibRealSense v2.31.0
The realsense is built with CUDA support. 
ROS distro: Melodic 

I have calibrated the D435I IMU  with the python tool and t decreased the fps from 30 to 15.
Exploring the rs_camera.launch I found these parameters :+1: 

<arg name="enable_gyro"         default="true"/>
<arg name="enable_accel"        default="true"/>

I supposed that these switch should always be set at true because  refer to the IMU
But I found that setting these parameters to **false** mapping is easier because the camera_link follow more precisely the real movement of the camera.

[image]

Instead switching them to **true** the mapping loose the "syncro" very soon, the depth cloud image start to "fly" in the z direction without any movement of the camera and is not possible to continue the mapping  

[image]

Could you please help me to clarify which is the real meaning of these switch and which it is supposed to be the right behaviour of the system ?

Thanks a lot,
Germal

MartyG-RealSense, 2020-08-15:
Hi @germal   My understanding is that the topics **enable_gyro** and **enable_accel** are normally set to False by default, as they are not accessible on non-IMU RealSense camera models such as D415 and D435.  However, these topics are made accessible with the **opensource_tracking** launch file by setting enable_gyro and enable_accel to True with the arg statements in the launch file.  

So I would speculate (I am not a ROS specialist) that setting these two conditions to False in the launch file would cause the IMU topics not to be published.

Regarding the apparent loss of stability that you experienced when the IMU topics are enabled, the instructions in the D435i SLAM guide make reference to this.  So in a sense, the potnetial for loss of stability can be said to be an expected behavior with this particular SLAM application.  To quote from the guide:

***********

The built-in IMU can only keep track for a very short time.  Moving or turning too quickly will break the sequence of successful point cloud matches and will result in the system losing track.  It could happen that the system will recover immediately if stopped moving but if not, the longer the time passed since the break, the farther away it will drift from the correct position. 

The odds for recovery get very slim, very quickly.  The parameters set in the launch file are most likely not ideal but this is a good starting point for calibrating.

**********

It is possible to perform SLAM without an IMU.  An example of this is **OpenVSLAM**:

https://github.com/xdspacelab/openvslam
https://openvslam.readthedocs.io/en/master/ros_package.html

germal, 2020-08-15:
Hi @MartyG-RealSense , thank you for you reply.
Maybe the madgwick filter node - that is a prerequisite for the project- extract already by default the imu data with the "unite imu" setting in the launch file so with a single topic  ( so no need to set true gyro and accelerometer) ?
Regards
Germal

MartyG-RealSense, 2020-08-15:
I do not know enough about **imu_filter_madgwick** to provide an answer that I could be confident in.  The documentation for it suggests that it is extracting data from the IMU and publishing topics using that data, but I do not know how the madgwick filter is utilised in the D435i SLAM example specifically.  

http://wiki.ros.org/imu_filter_madgwick

@doronhi the RealSense ROS wrapper developer is listed as the author of this guide, so he can provide more accurate advice on this subject than myself.

germal, 2020-08-15:
Hi @MartyG-RealSense ,when I set to false gyro and accelerometer I don't see ayn data published under i/camera/imu  but the proposed SLAM implementation seems to work the same ! At least the mapping part because the localization I haven't still tried.Please @doronhi could you please help to understand ?

Regards

MartyG-RealSense, 2020-08-27:
Re-tagging @doronhi on this question ...

doronhi, 2020-08-30:
Let me quote myself from [issue#951](https://github.com/IntelRealSense/realsense-ros/issues/951#issuecomment-542112046):

> I believe the original launch file in the demo is not good enough and does not pass the IMU information well enough to the RTabMap SLAM engine. I believe that it can be much improved with better knowledge of RTabMap.

Setting enable_gyro and enable_accel to true is needed indeed to start publishing the imu topic. It could be that the Imu is not calibrated but all in all I am not surprised that everything works just as well without it. I believe, but never had the time and knowledge to test it myself, that the integration of the IMU into the SLAM process can be done better then is showed in that example and if done correctly you'll see improvement when enabling the IMU.

germal, 2020-08-30:
Thank you @doronhi, I found regarding this topic an interesting explanation by you here

https://github.com/IntelRealSense/realsense-ros/issues/598#issuecomment-461600197

closed 2020-08-30 by germal