realsenseai/realsense-ros #1330: Two depth images are superimposed  then averaged and convert the averaged depth image to point cloud image problem

dingbaiyin, 2020-08-17:
The goal of the image is to shoot a flat wall, but the image unevenness error value is too large from the point cloud image. I superimpose the two depth images and calculate the average. Finally, use the new depth map to convert the point cloud image. But the converted point cloud image is very different from the original point cloud image. Please take a look at the image. What is the reason that the point cloud image generated after superposition has so many noises and the position has also changed. What is the reason, this problem has been bothering me.
this is original depth map
[image]

Point cloud image of the original depth map
[image]

Point cloud image averaged after superposition
[image]

dingbaiyin, 2020-08-17:
This is my python program code：

import cv2
import numpy as np
import  os

img1 = cv2.imread('/home/pi/Depth/dby-test_138_49_Pig_0_0.00_2020-08-11_09-56-31-925.png',cv2.IMREAD_UNCHANGED)
img2 = cv2.imread('/home/pi/Depth/dby-test_138_48_Pig_0_0.00_2020-08-11_09-56-31-891.png',cv2.IMREAD_UNCHANGED)
img1 = img1.astype(np.float32)
img2 = img2.astype(np.float32)

img=cv2.add(img1,img2)
img=img/2
img = img.astype(np.uint16)
cv2.imwrite('avreage_2_16.png',img)

MartyG-RealSense, 2020-08-17:
Hi @dingbaiyin   There does not seem to be any depth data at all in the original image.  It is reminiscent of the plain black texture that is produced when the camera cannot read any depth detail from a surface.  So the combined image probably is not representative of an image of the wall.

Also, if the point cloud image is the result of data from two png images then I don't think that you would get an accurate 3D point cloud from that.   Data sourced from 2D images would typically have to be "deprojected" to generate a 3D point cloud.  Aligning depth to color would also be highly recommended in order to accurately discern the foreground pixels of the image from the background pixels.

The Python script in the link below that aligns depth and color streams and generates a point cloud may help you to achieve your goal.

https://github.com/IntelRealSense/librealsense/issues/1231#issuecomment-464737908

dingbaiyin, 2020-08-18:
Thank you very much for your answers, so that I can analyze this problem from a deeper perspective. Simply changing from the point cloud map of the wall in the coordinate system, I think the superimposed point cloud map makes more noise, so the spherical center coordinates of the point in the meshlab software are based on the center of the overall point cloud map of the wall The ball heart expresses. So it looks like the coordinate system has changed.
I will try your suggestion, thank you very much.

closed 2020-09-09 by dingbaiyin