realsenseai/realsense-ros #1334: D435i Wrong IMU readings

nflsalex, 2020-08-18:
> RealSense ROS: v2.2.16
> Built with LibRealSense:  v2.36.0
> Firmware: 05.11.06.250

I just bought a new D435i and I set it up with the official imu-calibration python script.
After calibration, it shows different magnitude of acc for different pose when holding still.

as you can see, it has 9.97 and 9.57 as the magnitude of acceleration when camera is facing up and down

[image]
[image]

MartyG-RealSense, 2020-08-18:
hi @nflsalex   Values for position 5 and 6 near to the range that you experienced have been encountered before.  In the link below, a member of the RealSense team suggests that it may be because the camera was not completely orthogonal in steps 2 and 4 of the calibration process (the ones where the camera is held side-on by holding its USB connector).

https://github.com/IntelRealSense/librealsense/issues/5177#issuecomment-550210453

nflsalex, 2020-08-18:
I see. I will try a few more times to see if I can get it right. Just want to make sure I am not doing anything wrong .

Another unrelated question, is the IMU output from this ros package already in camera coordinate? so that I don't need to consider the extrinsic between camera and IMU?

MartyG-RealSense, 2020-08-18:
I will refer your question above to @doronhi the RealSense ROS wrapper developer to ensure that you get a correct answer.

nflsalex, 2020-08-21:
@doronhi  could you help with question mentioned before?

doronhi, 2020-08-26:
The gyro and accel are published with their own frame_ids: camera_gyro_optical_frame, camera_accel_optical_frame. That can be seen with `rostopic echo /camera/gyro/sample` for example.
Using `rosrun tf tf_echo camera_accel_optical_frame camera_depth_optical_frame` you can see the following transformation between the two:
```
- Translation: [-0.006, 0.005, 0.012]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]
```
So, in short and if I understood you correctly, you should consider extrinsics between the camera and the IMU.

closed 2020-08-31 by nflsalex