realsenseai/realsense-ros #1336: eloquent version doesn't work

rjshim, 2020-08-19:
Hi, 

I tested the eloquent branch and below didn't work.
$ ros2 launch realsense2_camera rs.launch.py

There are two parts that have to be fixed. 
1. install location for launch
1) in CMakeLists.txt
install(TARGETS ${PROJECT_NAME}
  DESTINATION lib/${PROJECT_NAME}
)

2. realsense2_camera is not an executable (but a library). Make it an executable.
1) in CMakeListst.txt
rclcpp_components_register_node(${PROJECT_NAME}
  PLUGIN "realsense2_camera::RealSenseNodeFactory"
  EXECUTABLE realsense2_camera_exec)

install(TARGETS ${PROJECT_NAME}
  DESTINATION lib/${PROJECT_NAME}
)
2) in rs.launch.py
        launch_ros.actions.Node(
            package='realsense2_camera', node_executable='realsense2_camera_exec',
            output='screen',
            emulate_tty=True,
            ),

Ryan

RealSenseSupport, 2020-08-20:
Hi @rjshim,

Thank you for your valuable remarks. As you probably have noticed, wrapper for Eloquent is at early stages of development. Therefore, many of the features are still not implemented and it would be more helpful to submit these not implemented parts as PR and not as issues. Would you like to submit  PRs for the two points you have mentioned?
Please let us know if you need additional assistance with this matter.

rjshim, 2020-08-21:
@RealSenseCustomerSupport 

Yeah, I can make some contribution to this repository but there are several points that I want you to note. If you agree, I will fork this repository and get to work on modification.  
1. realsense2_node package is redundant (and can be removed). There is no need for a new package just to make an executable. I will merge the source code with the main function in the package to the realsense_camera package so the executable can be simply run by the rs.launch.py file.
2. realsense_camera_msgs can be (or better be) renamed to realsense2_camera_msgs. The current naming causes unnecessary misunderstanding.

On the other hand, if possible, I hope Intel to support the following points as soon as possible.
1. Install manual for the driver-level library is confusing. Can I get any help with [this](https://github.com/IntelRealSense/realsense-ros/issues/1337)? 
2. As of August 9, 2020, Xacro works in ROS2. realsense_description package can also be added now!
3. Multiple packages developed and maintained by different teams within the company ([link](https://github.com/intel/ros2_intel_realsense)) cause confusion to users.

Ryan

doronhi, 2020-08-24:
@rjshim , thank you so very much for your helpful remarks.
As was mentioned, this branch is under heavy construction right now. As a matter of fact, the next commit may change it altogether.
For that reason, as opposed to contributions to the development branch, I do not encourage PRs right now. I Do encourage helpful remarks of the sort that you just made and I'll take them into consideration.
This branch is published for that reason - to have the useful remarks during the development process but this is definitely not a working version yet.
You can also follow the helpful notes I got so far [here](https://github.com/IntelRealSense/librealsense/issues/5825)
Thanks again and please feel free to comment wherever you find more convenient.

rjshim, 2020-08-25:
@RealSenseCustomerSupport @doronhi 

Thank you for the response. Then, I will wait for the next update as @doronhi said.

As an active user of Realsense products, I strongly agree with the points that are discussed in the link you attached.
I hope to see the next update asap (we bought a bunch of D435i models recently expecting to replace imu sensors with them).

Thanks! 
Ryan

doronhi, 2020-09-13:
> @RealSenseCustomerSupport
>
> Yeah, I can make some contribution to this repository but there are several points that I want you to note. If you agree, I will fork this repository and get to work on [...]

@rjshim , Contrary to my original plan, the version you referred to in your first comment will most probably be the basis for the first eloquent (or foxy) release. I think that regardless to what kind of version it will be, your contribution will be most beneficial either way. Therefor, contrary to my previous remark, I am very much interested in viewing and considering your suggestion and I think a PR is the most convenient way to achieve that.

RealSenseSupport, 2020-09-18:
@rjshim,

As @doronhi has mentioned, can you please submit PRs for the features you have mentioned in this GH issue and close it?

Thank you!

rjshim, 2020-09-23:
@RealSenseSupport  @doronhi 

These are the steps that I suggest you follow and work on this repository together.
1. Make a new branch for devel codes (codes for the software you are still developing)
The first problem is that this repository uses one branch (which is 'eloquent') for released codes and devel codes. A new norm that adopted by many ROS oriented open-source projects ([link1](https://github.com/ros-planning/navigation2), [link2](https://github.com/ros-planning/moveit2)) is that you use the 'main' branch for devel codes and make 'version-devel' branches for released codes. However, this repository also contains ROS1 codes so I suggest you make 'ros2' branch and upload newly updated codes ([ref](https://github.com/ros-planning/srdfdom)).

2. As you said I removed realsense2_node and merged it to realsense2_camera. Also, I changed realsense_camera_msgs to realsense2_camera_msgs. Once you make 'ros2' branch, I will fork it and send a pull request with the modified codes.

Regards,
Ryan

RealSenseSupport, 2020-10-09:
@JaehyunShim Your PR now merged into eloquent branch. Thanks very much for your sharing! And can we close this accordingly?

JaehyunShim, 2020-10-12:
@RealSenseSupport 

Thank you :)

closed 2020-10-12 by JaehyunShim

liushuya7, 2020-11-25:
"No Image" in Rviz2
[image]
I built this package on ROS2 foxy Ubuntu 20.04. However, there is no image showing from the /color/image_raw topic. I tried it on Ubuntu 18.04 with ROS2 Eloquent, it did stream the image topic but lagging heavily. Has anyone successfully run this package on ROS2 foxy Ubuntu 20.04?

liushuya7, 2020-11-27:
> "No Image" in Rviz2
> [image]
> I built this package on ROS2 foxy Ubuntu 20.04. However, there is no image showing from the /color/image_raw topic. I tried it on Ubuntu 18.04 with ROS2 Eloquent, it did [...]

It is solved. The problem is due to "Reliability Policy". Refer to: [ROS2 Image subscriber](https://answers.ros.org/question/361030/ros2-image-subscriber/)