realsenseai/realsense-ros #1337: Runtime error when running the eloquent branch

rjshim, 2020-08-19:
I successfully compiled this repository and when I tested "realsense2_node" executable, I got the errors below.

1)
[ERROR] [1597819925.280547693] [camera]: Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared

2)
[ERROR] [1597819925.292599284] [RealSenseCameraNode]: An exception has been thrown: Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.5/0003:8086:0B3A.0001/HID-SENSOR-200076.3.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied

Can I get any help or advice? 
Thank you in advance,
Ryan

RealSenseSupport, 2020-09-21:
@rjshim We just released one version for eloquent branch. Could you please have a try? https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.0   
Please refer to https://github.com/IntelRealSense/realsense-ros/blob/3.1.0/README.md to install and test. Looking forward to your update. Thanks!

rjshim, 2020-09-21:
I first built my robot system using [this package](https://github.com/intel/ros2_intel_realsense), since I could not spend too much time only for camera package. I will find some time for testing it and let you know soon. 

Thank you very much. 
Ryan

rjshim, 2020-09-23:
I tested it and I didn't see any notable problems. I will close this thread. 

Thank you very much.
Ryan

closed 2020-09-23 by rjshim