realsenseai/realsense-ros #1338: Can't change the framerate of the camera even lower the resolution

ChunYe173, 2020-08-19:
Hi,

I am working with Intel Realsense D435 camera on ubuntu 16.04. I am trying to make the camera to publish image of 60fps with resolution of 424x240 based on the realsense-viewer. I changed the resolution and the frame rate of the color module image in the launch file and it shows in the terminal that the camera is publishing image of 424x240 with 60fps when I run the command _roslaunch realsense2_camera rs_camera.launch_ . But the command _rostopic hz /camera/color/image_raw_ shows that the camera is only publishing image of an average rate of 30fps but not 60fps. Attach is the pictures that I took to further explain the situation above. Please assists. Thanks

[image]

[image]

[image]

[image]

[image]

MartyG-RealSense, 2020-08-19:
Hi @ChunYe173    I replicated your setup in the RealSense Viewer with a D435i.  I set RGB to 60 FPS and RGB8 on a USB2 connection.  Looking at the FPS information during streaming, I observed that when auto-exposure was enabled, FPS was around 33 instead of 60.  When auto-exposure was disabled, the FPS leapt to 59.5.

Under the **Controls** section of the RGB options, if the option **Auto-Exposure Priority** is disabled whilst auto-exposure is enabled then FPS becomes around 59.5.  This is because having Auto-Exposure enabled and Auto-Exposure Priority on the RGB controls disabled forces the camera to try to keep to a constant FPS.

I repeated the tests on a USB 3 connection and the results were identical.

Please try using a **rosrun** command directly after the roslaunch statement to disable auto-exposure and see what difference that makes to the published RGB FPS.

**rosrun dynamic_reconfigure dynparam set realsense2_camera_manager rs435_depth_enable_auto_exposure 0**

ChunYe173, 2020-08-20:
Hi @MartyG-RealSense ,

I have tried the solution that you provide. Its really helpful and the Intel Realsense D435 camera can now publish image of 60fps. Thanks a lot!

closed 2020-08-20 by ChunYe173

MartyG-RealSense, 2020-08-20:
Great news that the method worked for you - thanks for the update!