realsenseai/realsense-ros #1340: D435i ros-wrapper unstable and crash base on jetson nano

sammar80, 2020-08-19:
I appreciate if you could help I am using D435 with with ros and it is not stable on jetson nano I compiled the librealsense2 using CUDA also I followed the recommendation in similar issue
 https://github.com/IntelRealSense/realsense-ros/issues/1171 
and upgraded both the librealsense2 and the realsense-ros wrapper and still facing the same problem 

Ubuntu 18.04
ROS distro Melodic
librealsense2 ver 2.38.1.2169
realsense ros ver 2.2.16

MartyG-RealSense, 2020-08-19:
Hi @sammar80   The librealsense version used should ideally be closely matched with the version recommended for a particular version of the ROS wrapper.  The recommended librealsense SDK version for wrapper 2.2.16 is **2.37.0**.  So results from using 2.38.1.2169 with the 2.2.16 ROS wrapper cannot be guaranteed.

Could you provide further details please, and a roslaunch log if possible, to help us to better understand the instability that you are experiencing with your Jetson Nano?   Thanks!

sammar80, 2020-08-19:
Thank you @MartyG-RealSense  for your fast response I reinstalled the librealsense and realsense-ros wrapper as you mentioned 
I am still getting error please find attached the my launch file and  ros logs 
[realsense-ros-logs.txt](https://github.com/IntelRealSense/realsense-ros/files/5099052/realsense-ros-logs.txt)

and this my launch file 
Note I am using depth images and pointcloud 
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="640"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="15"/>
  <arg name="depth_fps"           default="15"/>
  <arg name="infra_fps"           default="15"/>
  <arg name="color_fps"           default="15"/>
  <arg name="gyro_fps"            default="50"/>
  <arg name="accel_fps"           default="50"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="false"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="false"/>
  <arg name="allow_no_texture_points"   default="false"/>

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
[... 26 lines cut ...]

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
    </include>
  </group>
</launch>

MartyG-RealSense, 2020-08-19:
I see in your log that you are experiencing continuous **control_transfer returned** warnings.  This indicates that there is a serious communication problem with the device.

Since wrapper version **2.2.15**, infra is set to False by default.  So if you need the infra streams then you should set them to True.

Are you able to visualize the streams in RViz despite the continuous warnings?  If you can, this would indicate that they are being published correctly.

sammar80, 2020-08-19:
Actually once I got these errors the images stop showing in rviz  and for the infra I leave it false to minimize the load on the program also sometimes when I run the lunch file the Nvidia shutdown

MartyG-RealSense, 2020-08-20:
If you are using your own custom launch file, do you have the same problems with errors and no images in RViz if you generate a point cloud with the ROS wrapper's RGBD point cloud example that uses this launch statement:

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud**

https://github.com/IntelRealSense/realsense-ros#rgbd-point-cloud

sammar80, 2020-08-20:
Thank you I will check the link and get back to you

sammar80, 2020-08-20:
I the configurations from the link you sent but I am still facing the same errors please find the new ros logs
[realsense-ros-logs-1.txt](https://github.com/IntelRealSense/realsense-ros/files/5103735/realsense-ros-logs-1.txt)

MartyG-RealSense, 2020-08-20:
Thank you very much for your tests.  Do you experience any problems with using the camera with librealsense programs such as the RealSense Viewer?  Or are the problems only in the ROS wrapper?

sammar80, 2020-08-20:
It gives the same errors on realsense-viewer you check the depth images using the High accuracy presets 2D and 3D
[image]
[image]
[realsense-viewer logs.txt](https://github.com/IntelRealSense/realsense-ros/files/5104503/realsense-viewer.logs.txt)

MartyG-RealSense, 2020-09-03:
Hi @sammar80   Do you have an update on your case for us please?   Thanks!

closed 2020-09-13 by MartyG-RealSense