realsenseai/realsense-ros #1341: Distance calculation

rakeshjohn123, 2020-08-19:
How can the distance to the detected object can be obtained as I have subscribed to the rostopic for rgb and depth image.Is there anything similar to depthframe.get_distance when using ros?I know the centre of the detection x&y in 2D from which how to find the depth(distance to the object)could you share some insights using "python" coding.
--

MartyG-RealSense, 2020-08-19:
Hi @rakeshjohn123   An approach that I have typically seen is to get the 16-bit distance value (uint16) in mm from a depth topic - for example, **camera/depth/image_rect_raw** - and convert it to meters using **cv_bridge**.  A long discussion in the link below talks about the subject in terms of Python.

https://github.com/IntelRealSense/realsense-ros/issues/714

Within that discussion, a RealSense user posted their finished Python script on a comment as a text file.

https://github.com/IntelRealSense/realsense-ros/issues/714#issuecomment-561036465

rakeshjohn123, 2020-08-20:
The code mentioned in my_depth_subscriber.txt is converting into array and obtaining the depth. How can this be modified to obtain the distance to a particular pixel in the image with known x & y coordinates.

MartyG-RealSense, 2020-09-03:
Sincere apologies that you did not receive a further comment from me on this case.   Do you still require assistance with it, please?

closed 2020-09-13 by MartyG-RealSense