realsenseai/realsense-ros #1342: Pose estimation (U,V,W)

rakeshjohn123, 2020-08-20:
I am using d435 realsense with ROS wrapper. I know the x & y coordinates in 2D and the distance to the object z. How to find the similar coordinates in 3D?

MartyG-RealSense, 2020-08-20:
Hi @rakeshjohn123  If you are using a D435 that lacks an IMU component like the D435i and D455 have then you may be able to estimate pose in ROS with a technique called **Deep Object Pose (DOPE)**.  Here are a couple of examples that have been tested with D435:

https://github.com/yehengchen/DOPE-ROS-D435
https://github.com/NVlabs/Deep_Object_Pose/issues/10

rakeshjohn123, 2020-08-20:
Thanks for your reply. I would want be more specific as to convert a 2D pixel to 3D camera/world coordinates. However this has to be applicable with ROS wrapper. Is there any other solution for this  apart from your previous comment. I see some implementation using camer_intrinsics using c code.

MartyG-RealSense, 2020-08-20:
I could not find many useful references about converting a 2D pixel coordinate to a 3D world coordinate with ROS in the same way that you can with deprojection in librealsense.  This article may provide some useful guideanc ethough:

https://medium.com/@yasuhirachiba/converting-2d-image-coordinates-to-3d-coordinates-using-ros-intel-realsense-d435-kinect-88621e8e733a

rakeshjohn123, 2020-08-20:
The error I receive is:
def convert_depth_to_phys_coord_using_realsense(x, y, depth, cameraInfo):
_intrinsics = pyrealsense2.intrinsics()
_intrinsics.width = cameraInfo.width
_intrinsics.height = cameraInfo.height
_intrinsics.ppx = cameraInfo.K[2]
_intrinsics.ppy = cameraInfo.K[5]
_intrinsics.fx = cameraInfo.K[0]
_intrinsics.fy = cameraInfo.K[4]
#_intrinsics.model = cameraInfo.distortion_model
_intrinsics.model = pyrealsense2.distortion.none
_intrinsics.coeffs = [i for i in cameraInfo.D]
result = pyrealsense2.rs2_deproject_pixel_to_point(_intrinsics, [x, y], depth)
#result[0]: right, result[1]: down, result[2]: forward
return result[2], -result[0], -result[1]

the error I get:

intrinsics.coeffs = [i for i in cameraInfo.D]
TypeError: (): incompatible function arguments. The following argument types are supported:
1. (arg0: pyrealsense2.pyrealsense2.intrinsics, arg1: List[float[5]]) -> None

Invoked with: width: 0, height: 0, ppx: 0, ppy: 0, fx: 0, fy: 0, model: 0, coeffs: [0, 0, 0, 0, 0], []

MartyG-RealSense, 2020-08-20:
The subject is outside of my programming knowledge, unfortunately as I am on the Support team and are not a developer.  The same is true for your similar question at https://github.com/IntelRealSense/realsense-ros/issues/1341#issuecomment-677368350

The best I can do is highlight your question to @doronhi  the RealSense ROs wrapper developer.   I do apologise.

rakeshjohn123, 2020-08-20:
@ MartyG-RealSense 
That would be great. kindly highlight this to the Realsense Ros wrapper developer.

MartyG-RealSense, 2020-08-20:
The @ name tag I used above should inform him that his name has been mentioned on this case.

doronhi, 2020-08-26:
I am sorry but I can't reproduce your error for the given line. I copied it and it worked great.
I did get that error when enabling the following line:
`_intrinsics.model = cameraInfo.distortion_model`
That's because the wrapper names the distortion model as "plumb_bob" and the name in librealsense is "brown_conrady" or "inverse_brown_conrady".
I'll need to investigate some more to see if there isn't a mixup with the model names in the wrapper but for now, since the parameters are 0,0,0,0,0 anyway, the line that you ended up with (`_intrinsics.model = pyrealsense2.distortion.none`) works for me.
Otherwise you could write something like:
```
            if cameraInfo.distortion_model == 'plumb_bob':
                _intrinsics.model = rs2.distortion.brown_conrady
            elif cameraInfo.distortion_model == 'equidistant':
                _intrinsics.model = rs2.distortion.kannala_brandt4
```

I modified the `show_center_depth.py` from realsense2_camera/scripts to check this out. I post the modified version for you to compare. Maybe I'll modify the one in the repository as well...

```
import rospy
from sensor_msgs.msg import Image as msg_Image
from sensor_msgs.msg import CameraInfo
from cv_bridge import CvBridge, CvBridgeError
import sys
import os
import pyrealsense2 as rs2

class ImageListener:
    def __init__(self, topic):
        self.topic = topic
        self.bridge = CvBridge()
        self.sub = rospy.Subscriber(topic, msg_Image, self.imageDepthCallback)
        self.sub_info = rospy.Subscriber('/camera/depth/camera_info', CameraInfo, self.imageDepthInfoCallback)
        self.intrinsics = None

    def imageDepthCallback(self, data):
        try:
            cv_image = self.bridge.imgmsg_to_cv2(data, data.encoding)
            pix = (data.width/2, data.height/2)
            sys.stdout.write('%s: Depth at center(%d, %d): %f(mm)\r' % (self.topic, pix[0], pix[1], cv_image[pix[1], pix[0]]))
            sys.stdout.flush()
            if self.intrinsics:
                depth = cv_image[pix[1], pix[0]]
                result = rs2.rs2_deproject_pixel_to_point(self.intrinsics, [pix[0], pix[1]], depth)
                print 'result:', result
                # sys.stdout.write('%s: Depth at center(%d, %d): %f(mm)\r' % (self.topic, pix[0], pix[1], cv_image[pix[1], pix[0]]))
                # sys.stdout.flush()

        except CvBridgeError as e:
            print(e)
            return

    def imageDepthInfoCallback(self, cameraInfo):
        try:
            # import pdb; pdb.set_trace()
            if self.intrinsics:
                return
            self.intrinsics = rs2.intrinsics()
            self.intrinsics.width = cameraInfo.width
            self.intrinsics.height = cameraInfo.height
            self.intrinsics.ppx = cameraInfo.K[2]
            self.intrinsics.ppy = cameraInfo.K[5]
            self.intrinsics.fx = cameraInfo.K[0]
            self.intrinsics.fy = cameraInfo.K[4]
            if cameraInfo.distortion_model == 'plumb_bob':
                self.intrinsics.model = rs2.distortion.brown_conrady
            elif cameraInfo.distortion_model == 'equidistant':
                self.intrinsics.model = rs2.distortion.kannala_brandt4
            self.intrinsics.coeffs = [i for i in cameraInfo.D]
        except CvBridgeError as e:
            print(e)
            return

def main():
    topic = '/camera/depth/image_rect_raw'
    listener = ImageListener(topic)
    rospy.spin()

if __name__ == '__main__':
    node_name = os.path.basename(sys.argv[0]).split('.')[0]
    rospy.init_node(node_name)
    main()
```

rakeshjohn123, 2020-08-27:
I will test the above code. Meanwhile I am using this way to manually calculate the 3d coordinates....

import pyrealsense2 as rs
import numpy as np

config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
pipeline = rs.pipeline()
pipe_profile = pipeline.start(config)
frames = pipeline.wait_for_frames()
depth_frame = frames.get_depth_frame()
color_frame = frames.get_color_frame()

# Intrinsics & Extrinsics
depth_intrin = depth_frame.profile.as_video_stream_profile().intrinsics
color_intrin = color_frame.profile.as_video_stream_profile().intrinsics
depth_to_color_extrin = depth_frame.profile.get_extrinsics_to(color_frame.profile)
color_to_depth_extrin = color_frame.profile.get_extrinsics_to(depth_frame.profile)
print("\n Depth intrinsics: " + str(depth_intrin))
print("\n Color intrinsics: " + str(color_intrin))
print("\n Depth to color extrinsics: " + str(depth_to_color_extrin))

# Depth scale - units of the values inside a depth frame, i.e how to convert the value to units of 1 meter
depth_sensor = pipe_profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("\n\t depth_scale: " + str(depth_scale))
depth_image = np.asanyarray(depth_frame.get_data())
depth_pixel = [500, 500] # Random pixel
depth_value = depth_image[200][200]*depth_scale
print("\n\t depth_pixel@" + str(depth_pixel) + " value: " + str(depth_value) + " meter")

# From pixel to 3D point
depth_point = rs.rs2_deproject_pixel_to_point(depth_intrin, depth_pixel, depth_value)
print("\n\t 3D depth_point: " + str(depth_point))

Now we know fx,fy,cx and cy and the rotation and translation values. Now I input these values in the code using ROS wrapper and obtain the same results.

depth_image = bridge.imgmsg_to_cv2(ros_image, "passthrough")
depth_array = np.array(depth_image, dtype=np.int32)
c1 = depth_array[cy,cx] 
cxx = 318.7731628417969
cyy = 238.74026489257812
fx = 384.08673095703125
fy = 384.08673095703125
scale = 0.0010000000475
        u = cx
        v = cy
        x_ = (u - cxx)/fx
        y_ = (v - cyy)/fy
        z1 = float(np.array(c1))
        z = z1/1000
        x = x_ * z
        y = y_ * z 
        print 'closest distance ',text_id, c1, 'mm'

This is specific to the resolution 640x480.  The fx,fx,cx,cy values have to be updated when we use the wrapper with different resolutio.

rakeshjohn123, 2020-08-27:
The x and y coordinates that we obtain what do they mean? If they are in world coordinates, then the camera is 0,0....?

doronhi, 2020-09-02:
x,y,z are given in mm, in the camera's coordinate: X-right, Y-down, Z-forward.
As for the 250 mm distance, with my D435 I usually get around 150mm minimum distance. The wrapper has a maximum range clip option ("clip_distance") but not control over the minimal range.

MartyG-RealSense, 2020-09-09:
Hi @rakeshjohn123   Further commentary about the axes can be found in the link below:

https://github.com/IntelRealSense/librealsense/issues/7279#issuecomment-689031488

Do you require further assistance with this case, please?   Thanks!

MartyG-RealSense, 2020-09-18:
Case closed due to no further comments received.

closed 2020-09-18 by MartyG-RealSense

ravijo, 2021-03-09:
Although this issue is old, however, I am posting my comment here. Just in case, it could be helpful for someone. Please read below:

Given a pixel location (x, y) and its corresponding depth (z), the 3D coordinate in camera space, can be calculated by using camera intrinsic parameters. Luckily, these parameters are provided by the IntelRealSense camera and hence we can directly plug these values to get the 3D point.

[Please see the code here.](https://github.com/ravijo/ros_openpose/blob/b9cbca7f31965af6b5304504e75b9cec1df0f747/include/ros_openpose/cameraReader.hpp#L110-L145)

MartyG-RealSense, 2021-09-20:
Hi @harezhh  Below are a couple of simple example scripts for obtaining camera intrinsics in Python using **as_video_stream_profile().get_intrinsics()**

```
cfg = pipeline.start() # Start pipeline and get the configuration it found
profile = cfg.get_stream(rs.stream.depth) # Fetch stream profile for depth stream
intr = profile.as_video_stream_profile().get_intrinsics() # Downcast to video_stream_profile and fetch intrinsics
print intr.ppx
```

*************************

```
pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_stream(rs.stream.infrared, 1)
cfg.enable_stream(rs.stream.infrared, 2)
cfg.enable_stream(rs.stream.depth)
prof = self.pipe.start(cfg)
ds = prof.get_stream(rs.stream.depth)
intr = ds.as_video_stream_profile().get_intrinsics()
self.fx = intr.fx
```