realsenseai/realsense-ros #1344: Calculation of calibration matrix

UditSinghParihar, 2020-08-21:
Hello Sir,

1. I would like to know how the calibration matrix `K` is calculated which is being published in `/camera/color/camera_info` topic.   
2. I am using this roslaunch command: `roslaunch realsense2_camera rs_camera.launch align_depth:=true`.    
3. Is the calibration matrix produced by the firmware or is it being calculated and calibrated online?  
4. And if it is produced by firmware then is it accurate enough or should we do our own calibration?  

Thanks

MartyG-RealSense, 2020-08-21:
Hi @UditSinghParihar   I hope that the link below, which provides a formula for 'k', will be helpful to you.

https://github.com/IntelRealSense/realsense-ros/issues/709#issuecomment-479950687

The RealSense ROS wrapper's documentation says about align_depth: "If set to true, will publish additional topics with the all the images aligned to the depth image.  The topics are of the form: /camera/aligned_depth_to_color/image_raw etc".

https://github.com/IntelRealSense/realsense-ros/tree/development#aligned-depth-frames

I extensively researched your second question about how alignment in generated in the RealSense ROS wrapper but but not find a good reference, so I will refer that question to the wrapper developer @doronhi

doronhi, 2020-08-26:
The parameters set into matrix K are the intrinsics parameters, derived directly from the firmware.
They are calibrated at the factory. As for them being accurate enough for you or to the possibility of calibration them yourself, I can refer you to [librealsense issue #2329](https://github.com/IntelRealSense/librealsense/issues/2329)

UditSinghParihar, 2020-08-26:
Thanks for mentioning the link for circumstances when I need to calibrate and the calibration toolbox.

closed 2020-08-26 by UditSinghParihar