realsenseai/realsense-ros #1348: Realsense ros don't run with realsense LiDar L515

duclinhfetel, 2020-08-25:
Hi everyone,

Im using realsense L515 for my project. I use realsense-ros package to run with realsense L515 is got the error below:
[image]

Could anyone help me this problem?

```
OS: Ubuntu 18.04
ROS: melodic
librealsense version: v2.38.1
realsense-ros: 2.2.16
```

diogo-sr, 2020-08-28:
I have the exact same problem

```
CPU: Intel® Core™ i7-8565U CPU @ 1.80GHz × 8 
OS: Ubuntu 16.04 
ROS: kinetic
librealsense version: v2.38.1
realsense-ros: 2.2.16

```

However, I have no problems with viewing the depth on the `realsense-viewer`

RealSenseSupport, 2020-08-29:
Hi,

We are able to reproduce this and are investigating. Do you see any *functional* issues per chance? The error is indeed being output but all streams appear to be operating as expected. Are you also experiencing this?

Thanks for catching.

diogo-sr, 2020-08-31:
Hi,

I do not have any information being published on the depth topics.

doronhi, 2020-09-01:
Found the error in librealsense. When requesting "depth" callback also receives "infrared" frames.
While waiting for a fix in librealsense, it can be bypassed, for now, if infrared is actually enabled.
Use the following flag:
`enable_infra=true`

or in the command line: `roslaunch realsense2_camera rs_camera.launch enable_infra:=true`

uncleleer, 2020-09-04:
> Hi,
> 
> We are able to reproduce this and are investigating. Do you see any _functional_ issues per chance? The error is indeed being output but all streams appear to be operating as expected. Are you also experiencing this?
> 
> Thanks for catching.

hello, I have met the same problem while launching rs_rgbd.launch, I have found that neither _hardware align_ nor _software align_ works, rostopic _/camera/depth_registered/points_ has no data. And this is one of the _functional_ problem I think.

doronhi, 2020-09-06:
> hello, I have met the same problem while launching rs_rgbd.launch, I have found that neither _hardware align_ nor _software align_ works, rostopic _/camera/depth_registered/points_ has no data. And [...]

Did you try it out using my suggestion: add the flag `enable_infra:=true`

bae9875, 2020-09-09:
I found that setting filters argument to true fixes this issue.

roslaunch rs_camera.launch filters:=true

bae9875, 2020-09-09:
I have found that setting filters argument to pointcloud fixes this issue.

roslaunch rs_camera.launch filters:=pointcloud

Si-lhouette, 2020-10-11:
add the flag enable_infra:=true didn't  work for me.
I used:
    roslaunch realsense2_camera rs_rgbd.launch enable_infra:=true
It also has error:
    An error has occurred during frame callback: map::at

My Realsense SDK version is 2.39.0, and realsense-ros version is 3.3.1

Appreciate for any help.

doronhi, 2020-10-12:
I thought the problem was fixed in librealsense2 version 2.39.0 but I could be mistaken, I didn't check yet.
However, my example used "rs_camera.launch". rs_rgbd.lauch  is missing the `enable_infra` option. You could add it to the file (in 2 places - copy from the rs_camera.launch file)

Si-lhouette, 2020-10-12:
Thanks！！
I fixed it by adding the following in 'rs_rgbd.launch':
Add above '<arg name="enable_infra1"       default="false"/>':
'<arg name="enable_infra"       default="true"/>'

Add above '<arg name="enable_infra1"            value="$(arg enable_infra1)"/>':
'<arg name="enable_infra"            value="$(arg enable_infra)"/>'

RealSenseSupport, 2020-10-12:
Glad that solved the issue @Si-lhouette.

@duclinhfetel ...will you be needing further help with this?

Thanks

closed 2020-10-22 by duclinhfetel