realsenseai/realsense-ros #1349: ERROR after running command: roslaunch realsense2_camera rs_camera.launch

ChunYe173, 2020-08-25:
Hi,

I got an error after running the command _roslaunch realsense2_camera rs_camera.launch_. I am using the Intel Realsense D435 on Ubuntu 16.04 ROS Kinetic. Attach is the picture of the error. Please kindly assisst. Thank you.

[image]

MartyG-RealSense, 2020-08-25:
Hi @ChunYe173   Which version of librealsense and which version of the RealSense ROS wrapper are you using, please?

ChunYe173, 2020-08-26:
Hi @MartyG-RealSense , I have attached a few pictures of the version of librealsense and RealSense ROS Wrapper

[image]

[image]

MartyG-RealSense, 2020-08-26:
It looks as though you are using librealsense SDK version **2.36.0** and RealSense ROS version **2.2.16**.  If you have 2.36.0 installed, the recommended ROS wrapper version to match with it is **2.2.15**.  This is because 2.2.16 is targeted at SDK version 2.37.0.

ChunYe173, 2020-08-27:
Thanks for the solution. Installing ROS wrapper of version 2.2.15 solve my problem. For now, I can run the command _roslaunch realsense2_camera rs_camera.launch_ successfully as shown in the attach picture. I am trying to disable auto_exposure_priority controls through the RGB section of the controls in the dynamic_reconfigure interface to publish frame rate higher than 30 fps. The dynamic_reconfigure interface will shut down automatically once I click on the rgb_camera section (shown in the picture below) and the terminal running command _rosrun rqt_reconfigure rqt_reconfigure_ will return the error as shown in the picture below. Please kindly assists. Thank you.

[image]

[image]

[image]

MartyG-RealSense, 2020-08-27:
Dynamic Reconfigure options can also be set with a **rosparam** instruction in the launch file if you only need to set the option once at start-up.  In the link below, Doronhi the RealSense ROS wrapper developer highlights an example of changing RGB options on the 400 Series cameras.  

https://github.com/IntelRealSense/realsense-ros/issues/1354#issuecomment-681126057

You should be able to set auto-exposure priority status by substituting the dynamic-reconfigure parameter name for auto-exposure priority into the rosparam code provided in the link.  

I believe that the parameter name for RGB auto-exposure priority is **rs435_color_auto_exposure_priority**:  So the launch file rosparam code would likely look like this:

<rosparam>
  /camera/rgb_camera/rs435_color_auto_exposure_priority: false
  /camera/rgb_camera/rs435_color_auto_exposure_priority: false
</rosparam>

ChunYe173, 2020-08-28:
Thanks @MartyG-RealSense for the solution. It is very helpful. Thanks for your time. I will close this issue

closed 2020-08-28 by ChunYe173

MartyG-RealSense, 2020-08-28:
I'm pleased it was helpful to you - thanks for the update!