realsenseai/realsense-ros #1350: Disable compression and YUY2 to RGB conversion

markozar, 2020-08-25:
Hi,
I'm using a Raspberry to get the camera (D435i) color and depth images and send them to another PC through ethernet gigabit. I'm trying to minimize CPU load and bandwidth utilization. I would like to disable compression and any other processing. I just want to grab the YUY2 image coming from the camera and the depth and send them to the remote PC without any other processing.

I do not know if this is an issue or there is already a way to do it.

Thank you!

MartyG-RealSense, 2020-08-25:
Hi @markozar   If you wanted to minimize processing to its rawest level then you could grab the frames at the camera hardware as they are captured by using the **Low Level API** (usually, the SDK does camera stream processing in the High Level API).  

https://github.com/IntelRealSense/librealsense/blob/master/doc/api_arch.md#low-level-device-api

An SDK tool that uses the Low Level API with callbacks is the **rs-data-collect** program.

https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect

markozar, 2020-08-26:
Thank you for the answer MartyG!

MartyG-RealSense, 2020-09-03:
Okay, thanks so much for the update!   I will close this case now.   Please feel free to create a new case about the Low Level API if you have questions about it later.   Good luck!