realsenseai/realsense-ros #1351: Cmakelist has version difference issues.

farhangnaderi, 2020-08-26:
Cmake is looking for ver 37 but what really exists as SDK is 36.

MartyG-RealSense, 2020-08-26:
Hi @farhangnaderi  Is your problem similar to the one described in the case in the link below, please?

https://github.com/IntelRealSense/realsense-ros/issues/1322

farhangnaderi, 2020-08-26:
@MartyG-RealSense Yes exactly.

MartyG-RealSense, 2020-08-26:
If you have SDK 2.36.0 installed then it should be matched with ROS wrapper version **2.2.15**.  Wrapper version **2.2.16** is targeted at SDK version 2.37.0.

farhangnaderi, 2020-08-26:
I have just GIT CLONEd , is there any specific branch I have to checkout? I
also have installed everything from the source.

MartyG-RealSense, 2020-08-26:
It is recommended to use the **master** branch, as that represents a completed SDK version instead of an in-progress version of the next upcoming version of the SDK (which is on the **development** branch).

closed 2020-08-28 by farhangnaderi