realsenseai/realsense-ros #1354: D435 RGB color variation during image captures

ogil0018, 2020-08-26:
We are trying to use the RGB sensor on the D435 to analyze plant health and need a consistent RGB image.

When taking images of the objects on the conveyor belt the tint of the image shifts over time.

We found using the AWB and AE to be inconsistent as well so we have a fixed value for WB and exposure.

We are using a standard bar led light running at 6500k

[image]
[image]

Any ideas or suggestions on fixing this?

I have also noticed that the AWB seems to set other features then just the white balance, does anybody have more information on these automatic functions that the ISP is doing?

MartyG-RealSense, 2020-08-26:
Hi @ogil0018   I carefully considered your situation and ran tests in the RealSense Viewer.  The image that you kindly provided is symptomatic of AWB being off and WB reduced as low as it would go (2800).  Below is a scene that I replicated with these settings, showing the same blue tint that does not completely affect green foliage.

[image]

If you are getting inconsistent results from AWB and AE, you could try turning off the **AE Priority** option whilst keeping AE enabled.  This should result in a constant FPS speed being enforced.

ogil0018, 2020-08-26:
thanks for your feedback, the issue would be that it changes during image captures without changing the WB.

process:
1. perform AWB
2. disable AE and AWB
3. Set fixed value for WhiteBalance and Exposure
4. start image capture

during the tiem frame of the image capture none of the exposure or WB should be changed but the tint does change for some reason.

the light source is 6500K and we set a value of 5700k since it looks the best at the start.

MartyG-RealSense, 2020-08-26:
When auto-exposure is disabled, the SDK automatically returns to the last manual exposure value set.  So I would suppose that if you set a manual exposure value at the start of the program before enabling auto-exposure for the first time, then that same manual exposure value may be reverted to every time that auto-exposure is disabled.  This is because the manual exposure is not updated when AE is enabled.  

There is a detailed explanation from a RealSense team member in the link below.

https://github.com/IntelRealSense/librealsense/issues/1624#issuecomment-389085387

rakeshjohn123, 2020-08-26:
@MartyG-RealSense 

How can the auto white balance and the exposure be disabled while using ROS wrapper? Is there a possible way apart from changing the values in rqt_reconfigure which is not permanent as it does not retain the set values.

MartyG-RealSense, 2020-08-26:
@rakeshjohn123   I believe that the color exposure and the true / false state of auto-exposure can be configured before roslaunch by using a Python script.

https://github.com/IntelRealSense/realsense-ros/issues/345

doronhi, 2020-08-26:
Every parameter you can change using rqt_reconfigure can be defined in your launch file as well.
Add the following lines to your launch file, just before the lines launching realsense2_camera node:
```
<rosparam>
  /camera/rgb_camera/enable_auto_exposure: false
  /camera/rgb_camera/enable_white_balance: false
</rosparam>
```
Note: setting "enable_auto_exposure" to false generated the following warning for me:
```
 27/08 00:04:05,841 WARNING [140005430609664] (types.cpp:78) hwmon command 0x75( 0 1df 0 27f ) failed (response -6= Invalid parameter)
[ERROR] [1598475845.842005986]: hwmon command 0x75( 0 1df 0 27f ) failed (response -6= Invalid parameter)
```
I didn't find any implications yet. The message appears in the next runs until it is set back to true.

MartyG-RealSense, 2020-08-26:
@doronhi   Thanks very much for the assistance.  I knew that about dynamic reconfigure having rosparam equivalents but didn't know of relevant code to share to access color exposure and white balance specifically.  Appreciate the information.  :)

doronhi, 2020-08-26:
Maybe it is worth mentioning then, that the name of the parameter to put in the launch file is just copied from the one presented in the rqt_reconfigure application. Just look at the app and put "/" symbol between every level.

MartyG-RealSense, 2020-08-26:
@doronhi   The main difficulty I have experienced is not with the parameter names but with knowing which topic to use for a certain camera model.  For example, one reference gave the T265 auto-exposure rosparam as **camera/tracking_module/enable_auto_exposure: false** so I did not feel comfortable with guessing topics for the 400 Series equivalent of the rosparams in case I gave wrong information to the customer.

doronhi, 2020-08-26:
I see now. I can never remember it also. I have to plug a camera in to check :)

ogil0018, 2020-08-27:
We are using python not ROS

MartyG-RealSense, 2020-08-27:
@ogil0018   Understood.  This case was started on the **RealSense ROS** forum though instead of the main RealSense GitHub at https://github.com/IntelRealSense/librealsense/issues/ where non-ROS cases are usually posted, and so that is how ROS got involved in the discussion.  :)

Reviewing your most recent comments:  are you capturing frames constantly, or taking a single snapshot after a certain time period has passed?

ogil0018, 2020-08-27:
@MartyG-RealSense  you are right I am very sorry that I have posted in the wrong location.

I believe the pipe is started once at the start and we acquire images every 5 to 10 min.

we utilize a software trigger based on a hardware pulse the system receives.

MartyG-RealSense, 2020-08-27:
If you have the pipeline open all the time instead of opening for capture, taking the shot and re-closing it til needed again, I wonder if the strong LED bar light is saturating the RGB sensor with light over time.   If this is the case, you could probably observe the image becoming progressively bluer over the course of 15 minutes if you just leave the RGB stream running normally as a test in the RealSense Viewer.

rakeshjohn123, 2020-08-28:
> Every parameter you can change using rqt_reconfigure can be defined in your launch file as well.
> Add the following lines to your launch file, just before the lines launching realsense2_camera node: [...]

I tried this appart from the error you mentioned. I also receive the below mentioned warning during some instances.
[ WARN] [1598602189.034479492]: Hardware Notification:Depth stream start failure,1.5986e+12,Error,Hardware Error

MartyG-RealSense, 2020-08-28:
@ogil0018 and @rakeshjohn123    I am going to be discussing this case with a support manager later today, so I recommend pausing work on it for now until I get back to you after that consultation.

MartyG-RealSense, 2020-08-28:
@ogil0018  After a discussion with my manager, I eliminated one of the possible causes that I was considering.

@rakeshjohn123   As @ogil0018 is not using ROS, it may be easier if you start your own case on this forum so that things do not get confusing by having to switch between ROS and not-ROS discussion.   :)

@ogil0018  Could you provide feedback on my suggestions in the link below please?   Thanks!

https://github.com/IntelRealSense/realsense-ros/issues/1354#issuecomment-682217901

ogil0018, 2020-08-31:
@MartyG-RealSense I will look into this

MartyG-RealSense, 2020-09-09:
Hi @ogil0018   Do you still require assistance with this case, please?   Thanks!

ogil0018, 2020-09-09:
you can close it for now, it will be some time to complete testing since I got pulled to something else right now

MartyG-RealSense, 2020-09-09:
Okay, thanks very much for the update!

closed 2020-09-09 by MartyG-RealSense