realsenseai/realsense-ros #1357: Inconsistent FPS on D435i

r91andersson, 2020-08-27:
The camera I'm using is having very inconsistent fps, sometimes I don't get more than around 15 fps (where it seems to stabalize) even though Im using an external USB3 powered hub. Sometimes it starts directly with 15 FPS, and sometimes it starts at 30FPS but then drops continusly to around 15 FPS, and sometimes it successfully manage to deliver 30FPS continusly.

Here is one example for the frame rate at the beginning when I start the camera with the rs_camera.launch script:
```
average rate: 30.077
	min: 0.029s max: 0.037s std dev: 0.00184s window: 28
average rate: 30.034
	min: 0.029s max: 0.037s std dev: 0.00132s window: 58
average rate: 30.018
	min: 0.029s max: 0.037s std dev: 0.00121s window: 88
average rate: 28.970
	min: 0.029s max: 0.063s std dev: 0.00575s window: 114
average rate: 26.391
	min: 0.029s max: 0.064s std dev: 0.01023s window: 131
average rate: 24.794
	min: 0.029s max: 0.064s std dev: 0.01191s window: 147
average rate: 23.575
	min: 0.029s max: 0.064s std dev: 0.01284s window: 164
average rate: 22.713
	min: 0.029s max: 0.065s std dev: 0.01330s window: 180
average rate: 22.015
```
```
[ INFO] [1598529549.744164093]: Initializing nodelet with 8 worker threads.
[ INFO] [1598529549.964825896]: RealSense ROS v2.2.15
[ INFO] [1598529549.964918669]: Built with LibRealSense v2.36.0
[ INFO] [1598529549.965003089]: Running with LibRealSense v2.36.0
[ INFO] [1598529550.007920748]:  
[ INFO] [1598529550.487569057]: Device with serial number 938422074860 was found.

[ INFO] [1598529550.487789707]: Device with physical ID 2-4.1-56 was found.
[ INFO] [1598529550.487885552]: Device with name Intel RealSense D435I was found.
[ INFO] [1598529550.489604578]: Device with port number 2-4.1 was found.
[ INFO] [1598529550.496518091]: getParameters...
[ INFO] [1598529550.603074876]: setupDevice...
[ INFO] [1598529550.603158144]: JSON file is not provided
[ INFO] [1598529550.603197378]: ROS Node Namespace: camera4
[ INFO] [1598529550.603266053]: Device Name: Intel RealSense D435I
[ INFO] [1598529550.603320488]: Device Serial No: 938422074860
[ INFO] [1598529550.603380491]: Device physical port: 2-4.1-56
[ INFO] [1598529550.603461487]: Device FW version: 05.12.06.00
[ INFO] [1598529550.603519825]: Device Product ID: 0x0B3A
[ INFO] [1598529550.603562739]: Enable PointCloud: Off
[ INFO] [1598529550.603600277]: Align Depth: On
[ INFO] [1598529550.603634295]: Sync Mode: On
[ INFO] [1598529550.603744124]: Device Sensors: 
[ INFO] [1598529550.656947809]: Stereo Module was found.
[ INFO] [1598529550.708580155]: RGB Camera was found.
[ INFO] [1598529550.709282877]: Motion Module was found.
[ INFO] [1598529550.709454373]: num_filters: 0
[ INFO] [1598529550.709524904]: Setting Dynamic reconfig parameters.
[ INFO] [1598529556.056809842]: Done Setting Dynamic reconfig parameters.
[ INFO] [1598529556.067701912]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1598529556.068765931]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1598529556.070630980]: setupPublishers...
[ INFO] [1598529556.078452504]: Expected frequency for depth = 30.00000
[ INFO] [1598529556.086964429]: Expected frequency for color = 30.00000
[ INFO] [1598529556.089683022]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1598529556.093206486]: setupStreams...
[ INFO] [1598529556.146555552]: insert Depth to Stereo Module
[ INFO] [1598529556.146792875]: insert Color to RGB Camera
[ INFO] [1598529556.380462617]: SELECTED BASE:Depth, 0
[ INFO] [1598529556.408906530]: RealSense Node Is Up!

```

My setup:

* Realsense D435i
* Device FW version: 05.12.06.00
* Built with LibRealSense v2.36.0
* RealSense ROS v2.2.15
* ROS Melodic 
* Jetson AGX Xavier 
* Ubuntu 18.04

BR
Robin

MartyG-RealSense, 2020-08-27:
Hi @r91andersson   Can you test please whether depth to color alignment (which is set to True in your log) is responsible by including the instruction below in your roslaunch statement to remporarily turn off alignment, please:

**align_depth:=false**

r91andersson, 2020-08-27:
I just manage to get a 30 FPS consitently by disable the auto_expousure and set it manually to 220 (pretty dark).
Could it be some internal queue that fills up if the exposure is to large and subsequently makes the FPS drop?

r91andersson, 2020-08-27:
> Hi @r91andersson Can you test please whether depth to color alignment (which is set to True in your log) is responsible by including the instruction below in your roslaunch statement to temporarily turn off alignment, please:
> 
> **align_depth:=false**

MartyG-RealSense, 2020-08-27:
The method that you tried is a known method for setting a constant FPS (auto-exposure = disabled and exposure set to a manually chosen value).  That approach can be complicated though, as if the manual exposure is set too low or too high then it can cause FPS to fall or increase instead of staying at the desired level.  You seem to have found a value that works for you though.

The other way to set a constant FPS is to keep auto-exposure enabled and disable an RGB option called **auto-exposure priority**.

r91andersson, 2020-08-27:
I can verify that when I disabled auto-exposure priority while having the auto exposure enabled, the FPS went directly from 15FPS to 30FPS! (even though align_depth:=true) Thanks for help!

closed 2020-08-27 by r91andersson

MartyG-RealSense, 2020-08-27:
You are very welcome - I'm pleased that it worked for you.   :)