realsenseai/realsense-ros #1359: Realsensel D453i in roslaunch realsense2_camera not working !

tt2cloud, 2020-08-29:
Hello,everyone, I used Ubuntu16.04 LTS on PC, ROS kinetic and the device is Intel RealSense D435i.
I followed these installation guide (https://github.com/IntelRealSense/realsense-ros) 

realsense-viewer is working.
[image]
But if I launch 
`roslaunch realsense2_camera rs_camera.launch`
the terminal shows like this.
yun@yun-HP-EliteDesk-800-G4-TWR:~$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/yun/.ros/log/68af5518-e9cc-11ea-8cce-1cbfce291082/roslaunch-yun-HP-EliteDesk-800-G4-TWR-9077.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yun-HP-EliteDesk-800-G4-TWR:40675/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 470 lines cut ...]
7f64764bc000-7f64764e2000 r-xp 00000000 08:14 1179844                    /lib/x86_64-linux-gnu/ld-2.23.so
7f6476686000-7f64766a2000 rw-p 00000000 00:00 0 
7f64766ad000-7f64766ae000 rw-p 00000000 00:00 0 
7f64766ae000-7f64766cc000 r-xp 00000000 08:14 1184688                    /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f64766cc000-7f64766cd000 r--p 0001d000 08:14 1184688                    /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f64766cd000-7f64766ce000 rw-p 0001e000 08:14 1184688                    /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f64766ce000-7f64766e1000 rw-p 00000000 00:00 0 
7f64766e1000-7f64766e2000 r--p 00025000 08:14 1179844                    /lib/x86_64-linux-gnu/ld-2.23.so
7f64766e2000-7f64766e3000 rw-p 00026000 08:14 1179844                    /lib/x86_64-linux-gnu/ld-2.23.so
7f64766e3000-7f64766e4000 rw-p 00000000 00:00 0 
7ffea0665000-7ffea0686000 rwxp 00000000 00:00 0                          [stack]
7ffea0686000-7ffea0687000 rw-p 00000000 00:00 0 
7ffea0787000-7ffea078a000 r--p 00000000 00:00 0                          [vvar]
7ffea078a000-7ffea078c000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
[camera/realsense2_camera_manager-1] process has died [pid 9094, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/yun/.ros/log/68af5518-e9cc-11ea-8cce-1cbfce291082/camera-realsense2_camera_manager-1.log].
log file: /home/yun/.ros/log/68af5518-e9cc-11ea-8cce-1cbfce291082/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/yun/.ros/log/68af5518-e9cc-11ea-8cce-1cbfce291082/camera-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
I have been troubled by this problem for a few days, Could you help me,thanks in advance!!
@RealSenseSupport

MartyG-RealSense, 2020-08-29:
Hi @tt2cloud   It is important to match the librealsense version as closely as possible with the version recommended on the ROS wrapper's Releases page for a particular wrapper version.

Your librealsense version is listed as 2.37.0 and your ROS wrapper version is 2.2.11.  The 2.2.11 ROS wrapper was targed at the 2.31.0 verion of librealsense.  For the recent SDK 2.37.0, the recommended ROS wrapper version to match with it is **2.16.0**.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.16

tt2cloud, 2020-08-30:
> Hi @tt2cloud It is important to match the librealsense version as closely as possible with the version recommended on the ROS wrapper's Releases page for a particular wrapper version.
> 
> Your librealsense version is listed as 2.37.0 and your ROS wrapper version is 2.2.11. The 2.2.11 ROS wrapper was targed at the 2.31.0 version of librealsense. For the recent SDK 2.37.0, the recommended ROS wrapper version to match with it is **2.2.16**.
> 
> https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.16

I will try it later,thanks!!!

MartyG-RealSense, 2020-09-18:
Case closed due to no further comments received.

closed 2020-09-18 by MartyG-RealSense

cxt98, 2021-03-01:
Hi,

I also try to get realsense working with roslaunch on ubuntu 16.04. I can have it run with realsense-viewer and also another example python code using pyrealsense2. I just follow the `sudo apt-get install ros-kinetic-realsense2-camera` to install this library. When I run `roslaunch realsense2_camera rs_camera.launch`, I got the following. Could someone share their experience to figure this problem out? Thanks in advance!

-----------------------------------------------------------------------------------

Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cxt:39453/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 41 lines cut ...]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [32718]
process[camera/realsense2_camera-3]: started with pid [32719]
[ INFO] [1614634429.404945866]: Initializing nodelet with 6 worker threads.
[ INFO] [1614634429.451611861]: RealSense ROS v2.2.22
[ INFO] [1614634429.451634400]: Built with LibRealSense v2.42.0
[ INFO] [1614634429.451643854]: Running with LibRealSense v2.42.0
[ INFO] [1614634429.475527228]:  
[ WARN] [1614634429.482052794]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1614634429.482076936]: The requested device with  is NOT found. Will Try again.
 01/03 16:33:49,481 ERROR [140009424287488] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 01/03 16:33:49,481 ERROR [140009652311808] (sensor.cpp:523) acquire_power failed: failed to set power state
 01/03 16:33:49,482 WARNING [140009652311808] (rs.cpp:306) null pointer passed for argument "device"
[ INFO] [1614634435.495007601]:  
[ WARN] [1614634435.501193130]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1614634435.501208121]: The requested device with  is NOT found. Will Try again.
 01/03 16:33:55,500 ERROR [140009635526400] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 01/03 16:33:55,500 ERROR [140009652311808] (sensor.cpp:523) acquire_power failed: failed to set power state
 01/03 16:33:55,501 WARNING [140009652311808] (rs.cpp:306) null pointer passed for argument "device"
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

MartyG-RealSense, 2021-03-02:
Hi @cxt98   Your launch log looks very wrong, as most of the values are set at -1 instead of the default values that they should actually be configured at.

As a starting point, let's try resetting the camera at launch by adding an **initial_reset** instruction to the end of your roslaunch instruction:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

cxt98, 2021-03-02:
Hi @MartyG-RealSense, thanks a lot for your help! I rerun using the command you suggested and it seems got partially working. I attach the output here:

```
roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/cxt/.ros/log/dc097e12-7b8f-11eb-9100-7085c2a76b1d/roslaunch-cxt-15043.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cxt:32979/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 128 lines cut ...]
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static
```

Is there any reason I got the 'no data error'?

MartyG-RealSense, 2021-03-02:
It seems to be applying the default stream settings of the camera model in the final stages of launch, which may explain how it is able to function despite the values initially being -1.

depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8

It sets both of the above streams to the default speed of 30 FPS.

The **control_transfer returned** errors can be ignored if they only occur at launch, or appear once or twice a minute after launch.  If they generate continuously though then it can indicate a serious communication problem with the device.

In regard to RViz, have you added a topic to it using the **Add** button at the bottom of the RViz side-panel?

You could also test the camera with RViz using the simple pointcloud example in the RealSense ROS wrapper documentation.

https://github.com/IntelRealSense/realsense-ros#point-cloud

cxt98, 2021-03-02:
Okay, I got it run successfully, thanks for your help!

MartyG-RealSense, 2021-03-02:
You are very welcome @cxt98 !

doronhi, 2021-03-03:
Thank you @MartyG-RealSense .
Just as a side note I would like to clarify the behavior regarding the "-1" values in the launch file. Version 2.2.22 introduced a new behavior for the profile selection, as is mentioned in the release notes. It now "Chooses the default profile in case of an invalid request. Notice that the rs_camera.launch file now selects the default profile for each device."
The "launch parameters" section in the README.md file has also been updated to reflect this new behavior.
The feature is meant to deal with the fact that not all devices have the same profiles. It also prevents the need to define default profiles in realsense2_camera, as they are already defined in the underlying librealsense2 library.
Thanks

MartyG-RealSense, 2021-03-03:
Thanks very much @doronhi for the information about the "-1" values!