realsenseai/realsense-ros #1361: where do we have data description?

nflsalex, 2020-08-31:
where can I find data description of the ros output? 

For instance, how do I interpret the 16bit unsigned int that /camera/depth/image_rect_raw topic has?

MartyG-RealSense, 2020-08-31:
Hi @nflsalex  I hope that the discussion of values output from the RealSense ROS wrapper in the link below will help to answer some of your questions.  Please read downward through the comments from the point in the discussion that I have linked to.  Comments from "doronhi" are by the RealSense ROS wrapper developer.

https://github.com/IntelRealSense/realsense-ros/issues/41#issuecomment-473022624

If you have any further questions, please do feel free to ask.

nflsalex, 2020-08-31:
Thanks @MartyG-RealSense I was hoping to read an official document of the data in this ros package but your links also work.

Another question: I am using D435i, does realsense have confidence value for depth? if so, where can I find it in the ros message.

MartyG-RealSense, 2020-08-31:
The RealSense SR300 / SR305 cameras can do confidence value matching but the 400 Series stereo depth cameras do not.  According to Dorodnic the RealSense SDK Manager, an Advanced Mode function called **Second Peak Threshold** can be used as an alternative.

https://github.com/IntelRealSense/librealsense/issues/3185#issuecomment-460540961

You could set an Advanced Mode function with ROS by defining it in a **json** configuration file and loading in the json as part of the ROS launching process.

https://github.com/IntelRealSense/realsense-ros/issues/411

nflsalex, 2020-09-01:
Thanks @MartyG-RealSense !

One last question, do we know whether the depth value is the perpenticular distance from the 3d point to the image plane or the euclidean distance from the 3d point to the optical center? 

basically when I am trying to unproject the depth image to get 3d points? Can I do `(u - cx) / fx * depth` to get `x` in 3d??

MartyG-RealSense, 2020-09-01:
That question is somewhat outside of my knowledge, though someone with better knowledge of the subject of perpendicularity on the 400 Series cameras does provide advice in the link below:

https://github.com/IntelRealSense/librealsense/issues/6800#issuecomment-656797493

The official explanation for measurement of the depth pixel value, taken from the data sheet document for the 400 Series cameras, is provided in the extract below:

[image]

nflsalex, 2020-09-01:
@MartyG-RealSense  Thanks. Your link already answered my question.

closed 2020-09-02 by nflsalex