realsenseai/realsense-ros #1362: rtab mapping with multiple D435i

sapanakhemkar, 2020-08-31:
Hi 
We are planning to use multiple realsense D435i cameras for better FoV coverage for rtabmapping.
I would like to know do we need hardware sync enabled in D435i as described [here](https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration) and [here](https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/)

While launching rtabmap, do I need to set any other parameter apart from "depth_cameras" values to 2?

MartyG-RealSense, 2020-08-31:
Hi @sapanakhemkar   I should first emphasise that multiple-camera RTABMAP with RealSense cameras is not a subject that I have direct experience with, but my research for your case indicates that at least a few have tried it.  It typically seems to involve using custom launch files.

The best of the results that I found during my research was a paper that used a D435 and T265 with RTABMAP.  In that paper, they synced the cameras in ROS by subscribing to certain topics and then publishing modified topics.

http://kastner.ucsd.edu/ryan/wp-content/uploads/sites/5/2014/03/admin/Indiana-Stones-Final-Report.pdf

https://github.com/indianastones/maya-archaeology

In regard to the general subject of multiple cameras in RealSense ROS, Intel have some official guides.

**2 cameras on the same computer**
https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

The bottom of the article provides guidance for adding further cameras.

**3 cameras across 2 computers**
https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

doronhi, 2020-09-06:
I would like to add that hardware sync is used for several devices to take the images at the same time. A use case could be if you wanted to extract stereoscopic information from images taken by different cameras.
The timestamps for all cameras are on the same baseline so assuming you are using the depth provided by each camera, rtabmap can take the timestamps of the images into account when combining the information. I don't think hardware sync is necessary.

sapanakhemkar, 2020-09-07:
Thanks for response. I will try first without hw sync. Hope it will work.

closed 2020-09-07 by sapanakhemkar

tugbakara, 2022-08-18:
> e baseline so a

@sapanakhemkar hi, did you accomplish your task w/o hardware syn? I am planning to use d435i with ZED2 to use rtab map.

tugbakara, 2022-08-18:
> Thanks for response. I will try first without hw sync. Hope it will work.

@sapanakhemkar hi, did you accomplish your task w/o hardware syn? I am planning to use d435i with ZED2 to use rtab map.

tugbakara, 2022-08-18:
> I would like to add that hardware sync is used for several devices to take the images at the same time. A use case could be if you wanted to extract stereoscopic information from images taken by [...]

@doronhi hi, is it true for all othr cameras with realsense's cameras like ZED2-D435i don't they really need hard syn when using rtab map?

Sapana-Khemkar, 2022-08-18:
@tugbakara I did it long back. We used without hw sync up. We could not continue work due to funding issues. So not worked on it since last 1.5 year