realsenseai/realsense-ros #1363: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy

dingbaiyin, 2020-09-03:
As shown in the figure, I added the Controlling the Laser function to the program
The image is acquired when the laster power is set to Max, and the laster power is set to Min after the acquisition is completed. However, in the program, the delay time must be increased after the intelCamera.ControlLasterPowerTo_Max() function and before the intelCamera.ControlLasterPowerTo_Min(), otherwise the former will cause the camera to be not ready to fully capture the image too dark, and the latter will cause the program to report an error "set_xu(. ..). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy". But if I want to increase the image acquisition frequency, due to the increased delay, I cannot collect several images per second. Is there a way to avoid increasing the delay time?
[image]

Linux   4.19.97-v7l+
Intel D430

dingbaiyin, 2020-09-03:
At the same time I want to adjust the“ Mean Intensity Set Point ” value, is there any reference code，thank you.

MartyG-RealSense, 2020-09-03:
Hi @dingbaiyin   It looks as though this is a librealsense case and not a ROS one (this site is the RealSense ROS GitHub).  

In regard to minimum laser power, you may be able to reduce response time by setting minimum laser power as '1' if you are currently setting the minimum to '0'.  This is because setting it to zero turns off the IR Emitter and makes the image become darker.  It will then need to be turned on again when maximum laser power is set.

When capturing the image, you could use **wait_for_frames()** if you are not doing so already.  This tells the SDK to wait until a complete frame is received.

If you are saving a single frame as an image file such as a PNG, the **rs-save-to-disk** SDK C++ example program may be a useful reference if you have not seen it already.  It captures 30 frames to ensure the stability of the image, and then it saves the final frame of the set.

https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk

The part of the program highlighted in the link below shows how it makes use of wait_for_frames().

https://github.com/IntelRealSense/librealsense/blob/master/examples/save-to-disk/rs-save-to-disk.cpp#L30

Alternatively, if you are willing to save the frames in a **bag** file instead of a png, a solution suggested by Dorodnic the RealSense SDK Manager in the link below sounds like what you are trying to achieve with stop-start timing.

https://github.com/IntelRealSense/librealsense/issues/3671

Regarding the mean intensity set point, you could set it using a **json** configuration file:

https://github.com/IntelRealSense/realsense-ros/issues/1199#issuecomment-628918085

dingbaiyin, 2020-09-04:
Thank you for your respond.

dingbaiyin, 2020-09-07:
I read the link you recommended, but I still don't understand how the json file is called with the api function. When calling the json file, all the parameters are automatically updated,
Do I need other operations ？

dingbaiyin, 2020-09-07:
When I used this code(" realsense2_camera rs_camera.launch json_file_path : = /Path；") to read the json file, I found that I need to install ROS Kinetic to call realsense2_camera, but an error was reported during the installation process.
I am using Raspberry Pi 3B+ and the system is Raspberry.
The error is as follows：
[image]
Please help provide a solution ,thank you.

MartyG-RealSense, 2020-09-07:
I believe that you need to include the file name of the json when loading it from the roslaunch instruction, like in this example:

**roslaunch realsense2_camera rs_camera.launch json_file_path:=/path/to/config/DefaultPreset_viewer.json**

As well as loading a json in the roslaunch instruction in ROS, you can include it in the launch file instead.  Include the **absolute path** to the file.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_camera.launch#L5

dingbaiyin, 2020-09-07:
I am Glad to receive your reply so soon.I added the absolute path of the json file, my code is as follows：
  realsense2_camera rs_camera.launch json_file_path: = /home/pi/Desktop/fat1.54/caiji-intel-noweight-client-new/CameraConfig.json; //CameraConfig.json
Is this right?
I did not understand the method of the new solution you provided. I need to save the content in the link in a json file and then how do I call it？
Can you say more in detail

dingbaiyin, 2020-09-07:
CameraConfig.json file content is as follows:
{
    "aux-param-autoexposure-setpoint": "400",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "-0",
    "aux-param-colorcorrection11": "-0",
    "aux-param-colorcorrection12": "-0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "-0",
    "aux-param-colorcorrection6": "-0",
    "aux-param-colorcorrection7": "-0",
    "aux-param-colorcorrection8": "-0",
    "aux-param-colorcorrection9": "-0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "8500",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "156",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "150",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-amplitude-factor": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "24",
    "param-maxscorethreshb": "2047",
    "param-medianthreshold": "500",
    "param-minscorethresha": "1",
    "param-neighborthresh": "7",
    "param-raumine": "1",
    "param-rauminn": "1",
    "param-rauminnssum": "3",
    "param-raumins": "1",
    "param-rauminw": "1",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0499022",
    "param-regioncolorthresholdg": "0.0499022",
    "param-regioncolorthresholdr": "0.0499022",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "10",
    "param-robbinsmonroincrement": "10",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "325",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "480",
    "stream-width": "848"
}

MartyG-RealSense, 2020-09-07:
The structure of your json file path in the roslaunch instruction looks reasonable to me, based on others that I have seen.  

The meaning of the link that I provided was just to illustrate that the json path can be edited into the launch file as an arg instruction instead of loading it from the roslaunch statement:

  **<arg name="json_file_path"      default=""/>**

If you are satisfied to load the file from a path in your roslaunch statement then you do not need to use the launch file arg method.

If your librealsense version is **2.16.1**?  That would be really old, dating back to September 2018.  The matching ROS wrapper version for that SDK is **2.1.0** which also dates back to September 2018.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.1.0

dingbaiyin, 2020-09-08:
Thank you. I will update the librealsense version to 2.38.1 and try it.

dingbaiyin, 2020-09-11:
There is a problem with librealsense2.38.1, and Intel camera cannot be called normally. It is fine to replace it with librealsense2.36.0. I created a new issue about librealsense-ros and closed the issue. The link
 is it.
[https://github.com/IntelRealSense/realsense-ros/issues/1375](url)
Thank you very much.

MartyG-RealSense, 2020-09-18:
Hi @dingbaiyin   Do you require further assistance on this case, please?   Thanks!

dingbaiyin, 2020-09-21:
Thank you for your attention.
I encountered a lot of problems during the installation of ROS using Raspberry Pi 4B, and because other work did not solve the problems, I am trying to solve these problems. After completing the installation, the problems and solutions will be updated here

MartyG-RealSense, 2020-09-21:
Thanks very much - good luck!

dingbaiyin, 2020-09-27:
After installing ros, the memory of the 16G Raspberry Pi system was full, and the current problem of reading json files has been solved by the Intel team. There is no need to continue configuring ROS into the system for now. Thank you very much.

closed 2020-09-27 by dingbaiyin

MartyG-RealSense, 2020-09-27:
Great news that a solution was reached - thanks for the update!