realsenseai/realsense-ros #1365: Some topics not publishing

miajl, 2020-09-04:
I am using a D435i and when I run

roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation 

I am missing some of the topics that are supposed to be there. I particularly want /camera/imu, /camera/infra1/image_rect_raw and /camera/infra2/image_rect_raw

I have verified in realsense viewer that each of the IR cameras work but I was unsure how to test that the imu functions.

Here is the result of rostopic list
[image]

How do I get these topics?

MartyG-RealSense, 2020-09-05:
Hi @miajl  If you are using ROS wrapper **2.2.15** or newer, the infra, fisheye, IMU and pose streams are set to **False** by default.  It is mentioned in the release notes for 2.2.15.  It is understandable if you did not see it though.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.15

miajl, 2020-09-08:
The above worked to get infra1 and infra2. I needed to add enable_gyro:=true and enable_accel:=true to get camera/imu

closed 2020-09-08 by miajl