realsenseai/realsense-ros #1366: catkin_make clean Realsense-ros on Windows 10 PC Issues

mjs513, 2020-09-05:
First let me start off by saying I am a real noob at ROS as well as the D435 camera that I just got :)

I managed to install ROS-noetic for windows 10 and verified that it runs on my windows 10 machine.  Then following the instruction for method 2 catkin_make clean exits with the following errors:

```CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:41 (find_package):
  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "C:/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "C:/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```
I am attaching the output files from CMAKE for reference.  Any help would be appreciated since I have no idea.

[CMakeError.log](https://github.com/IntelRealSense/realsense-ros/files/5178149/CMakeError.log)

[CMakeOutput.log](https://github.com/IntelRealSense/realsense-ros/files/5178150/CMakeOutput.log)

MartyG-RealSense, 2020-09-05:
Hi @mjs513   Let's go through the Windows installation steps as I interpret them from the instructions.  I went through the process below step by step myself to confirm it after I had written the guide.

1.  Have you got **Git** installed on Windows, please?  The **grep** command that you were missing is part of Git.

https://www.atlassian.com/git/tutorials/install-git#windows

2.  Can you confirm please that you have successfully built **Intel RealSense SDK 2.0**  and have been able to test that it is working (for example, by running the **RealSense Viewer** program).

2.  If RealSense SDK 2,0 is working, move on to installing the RealSense ROS wrapper.  First, create a **catkin workspace** on Windows using the commands below:

**mkdir c:\catkin_ws\src
cd c:\catkin_ws\src**

These are  **Windows Command Prompt** terminal window commands.  The Command Prompt interface can be found by typing **command** into the text box at the bottom of the Windows screen, next to the button in the corner where the computer is shut down. 

4.  Clone **dynamic_reconfigure** into your catkin_ws/src/ workspace on Windows.  It can be done like this:

**git clone https://github.com/pal-robotics/ddynamic_reconfigure.git**

5.  Whilst in the folder **c:\catkin_ws\src** in the Command Prompt window, clone the RealSense ROS wrapper into this **src** folder:

**git clone https://github.com/IntelRealSense/realsense-ros.git

At this point I encountered the same problem as you - that **grep** was not recognised.  It seems to be an important command but I could not find any way around the problem on Windows.  I will tag the RealSense ROS developer into this discussion to seek their advice.

@doronhi  Do you have any ideas about how to handle 'grep' during a Windows installation of the ROS wrapper, please?

mjs513, 2020-09-05:
Hi @MartyG-RealSense
Thanks for getting back to me so fast on this.  In answer to your questions:

1. **GIT**:  Am assuming I have git installed since git clone is working but I also just ran a **git --version** command and got back: **git version 2.28.0.windows.1**
2. **_Can you confirm please that you have successfully built Intel RealSense SDK 2.0 and have been able to test that it is working (for example, by running the RealSense Viewer program)**._
  ok not sure what you mean by built SDK 2.0.  I have it downloaded to my "c" drive at C:\Program Files (x86)\Intel RealSense SDK 2.0\.  In tools there is a RealsenseView.exe that runs.  So any help here would be good on how to "BUILD" it.  Wasn't clear on to install the SDK for ROS purposes.  Samples work fine using VisualStudio 2019.
3. Steps 3, 4 and 4:  Did that exactly and thats where I ran into a problem.

Guess in the meantime while we wait for **grep** resolution I can figure out the SDK Build process

Thanks for your help 
Mike

MartyG-RealSense, 2020-09-05:
1.   It sounds as though you have Git installed successfully.

2.  There are various ways to build RealSense SDK 2.0 (librealsense), including the 791 MB **Intel.RealSense.SDK-WIN10-2.37.0.2094.exe** Windows installer from the **Assets** file list on the SDK's **Releases page**.  That is the easiest way to install the RealSense SDK.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.37.0

It is important to match a particular version of the RealSense ROS wrapper with a particular version of the RealSense SDK.  At the time of writing this, the most recent version of the ROS wrapper is **2.2.16**:

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.16

It is recommended that version **2.37.0** specifically of the RealSense SDK is used with this ROS wrapper version.  That is why I linked to 2.37.0 above, even though it is not the latest SDK version now.

Once you have SDK **2.37.0** installed first, then ideally ROS wrapper **2.16.0** should be installed secondly.

While waiting for further advice about the ROS wrapper, if you are a new user of the RealSense camera then you can learn a lot by playing with the **RealSense Viewer** software that the RealSense SDK installs.  A launch icon for it should be placed on your Windows desktop when installation of the SDK is complete.

mjs513, 2020-09-05:
Hi Marty
So I did download and install Intel.RealSense.SDK-WIN10-2.37.0.2094.exe and installed so at least it sounds like I got that part right.  Just verified that with the exe that I used to install the SDK.  So that looks like its installed.  And believe I have 2.16.0 wrapper as well

As for playing with the D435 that is exactly what I was doing and changing parameters to see it effect.  Biggest thing is getting use to looking at it in a 3D view vs. a 2D view.

Ultimate goal to place it on a Rover and use it for mapping and obstacle avoidance.

MartyG-RealSense, 2020-09-05:
It sounds as though you have **SLAM** navigation in mind for your project.  There are some non-ROS options for achieving that.   This includes MIT SPARK Lab's **Kimera** system.

https://github.com/MIT-SPARK/Kimera-VIO

It also has a ROS-enabled version if you decide to carry on down the ROS path:

https://github.com/MIT-SPARK/Kimera-VIO-ROS

If you have non-ROS questions, please feel free to post them on the librealsense GitHub forum:

https://github.com/IntelRealSense/librealsense/issues

MartyG-RealSense, 2020-09-05:
It may also be worth looking at Intel's **ros_object_analytics** ROS package, which can provide both collision avoidance and semantic SLAM.

https://github.com/intel/ros_object_analytics

mjs513, 2020-09-05:
See that the first reference to Kimera is based a relatively new approach (2020), will take a bit of time to read the associated paper but seems to be rather involved process and resource intensive.  

The  ros_object_analytics package is interesting - just have to translate to get it working on ROS for windows.  Think most of the steps are similar to realsense-camera.

What I have been looking at, that got me started was to implement a VFH 3d+, that uses octomap and VFH+: https://github.com/saifullah3396/vfh3d.

MartyG-RealSense, 2020-09-06:
Intel's white-paper document on improving depth sensing quality on drones does state that using depth data with a spatial mapping library can be a way to increase the robustness of depth data against "outliers" that cause wrong depth readings and cites octomap as a good starting point.

https://dev.intelrealsense.com/docs/depth-map-improvements-for-stereo-based-depth-cameras-on-drones#section-vii-limitations-and-outlook

Another approach to collision detection that you could look at is **PX4**, which has been used with flight drones for years as a flight controller.  

https://px4.io/

https://github.com/PX4

Its 'local planner' system also makes use of VFH+ according to the avoidance section of its GitHub.

https://github.com/PX4/avoidance

mjs513, 2020-09-07:
@MartyG-RealSense 
You really are a wealth of information with this stuff.  Thank you.  Made me do a bit more looking.  I came across a few more references along the lines of PX4 implementation:

Intel RealSense Depth Camera on ardupilot: https://ardupilot.org/copter/docs//common-realsense-depth-camera.html

and the second with 3dvfh+ on the PX4:  https://github.com/ArduPilot/OctomapPlanner/issues/9

MartyG-RealSense, 2020-09-07:
You are very welcome.  Yes, **Ardupilot** is another controller that has been used with RealSense-equipped drones.  The Ardupilott website has a tutorial on doing obstacle avoidance with RealSense.

https://ardupilot.org/rover/docs/common-realsense-depth-camera.html

MartyG-RealSense, 2020-09-18:
Hi @mjs513   Do you require further assistance with this case, please?   Thanks!

mjs513, 2020-09-21:
Hi @MartyG-RealSense
Kind of gave up getting it working on Windows.  So I picked up a Raspberry Pi4 and got librealsense built and able to run realsense viewer without an issue following the instructions from here: https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

But now when i try to install realsense2-camera using  sudo apt-get install ros-$ROS_VER-realsense2-camera i get the error: E: Unable to locate package ros-melodic-realsense2-camera.

MartyG-RealSense, 2020-09-21:
Installing the ROS wrapper on Raspbian can be problematic.  It is likely to be easier if using Ubuntu Server as the OS on the Pi 4.  

https://support.intelrealsense.com/hc/en-us/community/posts/360050092674

If you prefer to use Raspbian, **eleccelerator** offer guidance about installing the ROS wrapper on Pi 4 and Raspbian.

https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense#Installing_ROS_Package:_realsense-ros

mjs513, 2020-09-21:
Thanks @MartyG-RealSense will probably convert to unbuntu to see if I can get it working.

mjs513, 2020-09-23:
Evening @MartyG-RealSense 
Tried ubuntu 20.04 on the Pi4 (couldn't get 18.04 installed) but no luck.  I did get realsense-viewer working but the performance was terrible and 3d would not work.  So spent the day putting back in BUster and using your reference installing realsense ros on the pi4. Still executing the last step but looks like it will work.  May have to redo librealsense.

After it finishes will roslauch realsense camera work? or do I have to do something else

mjs513, 2020-09-23:
Ok - knew i couldn't get away with it working out of the box.  Get through everything and ran roslauch realsense camera.  It started but I am getting this error:

[ERROR] [1600826462.700901636]: An error has occurred during frame callback: Time is out of dual 32-bit range

any ideas.

MartyG-RealSense, 2020-09-23:
Apologies for the delay in responding further.  The **Time is out of dual 32-bit range** error is one that the RealSense ROS wrapper developer has researched in the past and concluded that the cause was likely something that is outside of the ROS wrapper.  So a solution was not found.  Below is a link to a case where this error also occurs with a Pi and Raspbian:

https://github.com/IntelRealSense/realsense-ros/issues/1178

The experience I have with RealSense users' cases involving installing wrappers such as ROS and Python with Pi 4 and Raspian Buster is that it can be problematic, and using **Ubuntu Server** as the Pi's OS may make things more straightforward.

mjs513, 2020-09-23:
Sorry for the  delay in response was working on installing Ubuntu 18.05 LTS plus following the instructions on github.  No joy.  When I install realsense SDK it installs 2.35.1 and only the 2d stereo view works.  Also if i check for Ros dep nothing shows up for /camera.  Pretty much a bust on the PI4

Going to go ahead and close this.

MartyG-RealSense, 2020-09-24:
Sorry to hear that you were not successful.  I researched the **Time is out of dual 32-bit range** error again and did find a recent case where using the roslaunch term **enable_sync:=true** worked for one respondent.

https://answers.ros.org/question/350467/intel-realsense-time-is-out-of-dual-32-bit-range/

doronhi, 2020-09-24:
I remembered @MartyG-RealSense referencing this issue once, with 1 link to my investigating and another to a possible solution. Check also this link: https://github.com/IntelRealSense/realsense-ros/issues/1284#issuecomment-658146832

mjs513, 2020-09-24:
Thanks guys for the response but i decided to try another approach in the mean time - but may go back and try the approach that gave me the time synch issue.  Here is my latest (dont know if you want me to open up another PR):
1.  Installed the SDK (2.38) using the Linux Installation instructions: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md.  However while I do get the streams for both modes it does throw errors to the console:
a.  On startup of realsense-viewer i get the following:
`24/09 15:07:09,660 WARNING [281472609100224] (backend-v4l2.cpp:1269) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!`
b. When i turn on the RGB camera stream works
c,  Turning off RGB and putting it into stereo mode in 2D works without further erros.
d. Putting into 3D mode with Stereo I get the following:
```
  24/09 15:12:07,907 ERROR [281472158134720] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
 24/09 15:12:07,908 ERROR [281472158134720] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Error occured during execution of the processing block! See the log for more info
 24/09 15:12:07,956 ERROR [281472158134720] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
```
Now if I install realsense-ros using Method 1 for Ubuntu and start up with:
` roslaunch realsense2_camera rs_camera.launch`
I get a series of errors repeated:
```
 24/09 11:47:04,282 WARNING [281473131196800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 24/09 11:47:04,441 WARNING [281473131196800] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```
Now if i try looking at whats published only thing i see published is camera-info and raw-image.  I was looking at this PR:
https://github.com/IntelRealSense/realsense-ros/issues/1271

But it doesn't look like the resolved the issue with depth or point cloud?

Was going to try building the wrapper as opposed to apt-get install.

mjs513, 2020-09-24:
Just by way of update I fixed the first issue with the errors when running viewer:   Had to edit config.txt to set kms for opengl and i increased the GPU mem to 128.  But I still get the  Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!

MartyG-RealSense, 2020-09-25:
I cannot see much wrong in the log, other than the **control_transfer returned** errors at the bottom.  Are they continuously generating please (which would indicate  a serious communication problem with the device if they were), or are there just that initial several after launch and then no more? 

Also, are the depth and color topics being published and viewable in RViz?

mjs513, 2020-09-25:
Looks like only getting it on start of the node - not continuous

Using hz this is what I am seeing
```
ubuntu@ubuntu:~$ rostopic hz /camera/color/image_raw
subscribed to [/camera/color/image_raw]
average rate: 16.908
	min: 0.048s max: 0.065s std dev: 0.00502s window: 16
average rate: 16.765
	min: 0.048s max: 0.065s std dev: 0.00424s window: 33
average rate: 16.693
	min: 0.048s max: 0.066s std dev: 0.00402s window: 50
average rate: 16.676
	min: 0.048s max: 0.071s std dev: 0.00431s window: 66
^Xaverage rate: 16.643
	min: 0.048s max: 0.071s std dev: 0.00433s window: 83
average rate: 16.642
	min: 0.048s max: 0.071s std dev: 0.00422s window: 100
average rate: 16.637
	min: 0.044s max: 0.078s std dev: 0.00486s window: 116
^Caverage rate: 16.624
	min: 0.044s max: 0.078s std dev: 0.00485s window: 118
ubuntu@ubuntu:~$ rostopic hz /camera/extrinsics/depth_to_color
subscribed to [/camera/extrinsics/depth_to_color]
no new messages
no new messages
no new messages
^Cno new messages
ubuntu@ubuntu:~$ rostopic hz /camera/infra1/image_rect_raw
WARNING: topic [/camera/infra1/image_rect_raw] does not appear to be published yet
^CTraceback (most recent call last):
  File "/opt/ros/melodic/bin/rostopic", line 35, in <module>
    rostopic.rostopicmain()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py", line 2132, in rostopicmain
    _rostopic_cmd_hz(argv)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1524, in _rostopic_cmd_hz
    use_wtime=options.use_wtime, tcp_nodelay=options.tcp_nodelay)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py", line 315, in _rostopic_hz
    rospy.Subscriber(real_topic, rospy.AnyMsg, rt.callback_hz, callback_args=topic, tcp_nodelay=tcp_nodelay)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 563, in __init__
    super(Subscriber, self).__init__(name, data_class, Registration.SUB)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 131, in __init__
    raise ValueError("topic name is not a non-empty string")
ValueError: topic name is not a non-empty string
ubuntu@ubuntu:~$ rostopic hz /camera/depth/image_rect_raw
subscribed to [/camera/depth/image_rect_raw]
average rate: 29.937
	min: 0.029s max: 0.039s std dev: 0.00282s window: 29
average rate: 30.013
	min: 0.025s max: 0.039s std dev: 0.00288s window: 59
average rate: 30.010
	min: 0.025s max: 0.039s std dev: 0.00298s window: 89
average rate: 30.000
	min: 0.025s max: 0.039s std dev: 0.00298s window: 119
^Caverage rate: 30.016
	min: 0.025s max: 0.040s std dev: 0.00293s window: 142
ubuntu@ubuntu:~$ rostopic hz /camera/extrinsics/depth_to_color
subscribed to [/camera/extrinsics/depth_to_color]
no new messages
no new messages
no new messages
no new messages
^Cno new messages
ubuntu@ubuntu:~$ rostopic hz /camera/color/image_raw
subscribed to [/camera/color/image_raw]
average rate: 16.757
	min: 0.052s max: 0.064s std dev: 0.00365s window: 16
average rate: 16.686
	min: 0.050s max: 0.069s std dev: 0.00405s window: 32
average rate: 16.645
	min: 0.050s max: 0.069s std dev: 0.00401s window: 49
average rate: 16.631
	min: 0.050s max: 0.069s std dev: 0.00398s window: 66
average rate: 16.643
	min: 0.050s max: 0.071s std dev: 0.00417s window: 82
^Caverage rate: 16.622
	min: 0.050s max: 0.071s std dev: 0.00405s window: 92
ubuntu@ubuntu:~$ rostopic hz /camera/depth/camera_info
subscribed to [/camera/depth/camera_info]
average rate: 29.968
	min: 0.030s max: 0.036s std dev: 0.00153s window: 28
average rate: 29.956
	min: 0.030s max: 0.039s std dev: 0.00174s window: 58
average rate: 29.913
	min: 0.028s max: 0.039s std dev: 0.00202s window: 88
^Caverage rate: 29.942
	min: 0.028s max: 0.039s std dev: 0.00213s window: 111
ubuntu@ubuntu:~$ 
```
Maybe I am doing something wrong with RVIZ

mjs513, 2020-09-25:
[image]

MartyG-RealSense, 2020-09-25:
Does it display if you change Fixed Frame from 'map' to **camera_link**

mjs513, 2020-09-25:
No image received.  Getting that message for camera status warning.  

Some debug.  Used realsense-viewer to make sure had a good connection before doing roslaunch/rviz.  After closing rviz and roslauch windows did a realsense-viewer again and shows no connection.  

Seems like it looses the connection along the way.

MartyG-RealSense, 2020-09-26:
It would not matter that you are using ROS1, as the RealSense ROS wrapper is fully compatible with ROS1.

After the camera becomes non-detectable, is the camera detection restored if you unplug the camera and plug it back in?

mjs513, 2020-09-27:
Evening @MartyG-RealSense 
Spent the day getting it working with Raspian-Buster, which it is now working but more on that later) and reread a few things on how rviz works.  So went back to the Ubuntu install and repeated the test with RVIZ and it seems to be working,  Just my ignorance - sorry, but did learn a lot;  Here is a screen shot
[image]

Pretty much seeing the same thing when I run it under Raspian.

Question now is, is there are good examples for using the camera on a robot?

Thanks for all your help and you patience.

Mike

PS> enable_sync:=true works to get rid of the time issue when running under Buster..

MartyG-RealSense, 2020-09-27:
Thanks so much for the update @mjs513  - it is great to know that there may be a reliable solution for the **Time is out of dual 32-bit range** error now.

mjs513, 2020-09-27:
@MartyG-RealSense - Proably should document the steps somewhere :) One more question if I may.

How can I see the 2D depth Map with RVIZ?  Not sure this is possible.    Are there any other Demo's that can be run?

MartyG-RealSense, 2020-09-27:
There was a case where a RealSense user published a depth map in RViz, though the process by which it was achieved was not described.

https://github.com/IntelRealSense/realsense-ros/issues/590

There is another case though where a method used to generate a depth map with ROS is described in the opening paragraph.

https://github.com/IntelRealSense/realsense-ros/issues/436

The depth map image generated in this particular case looks like this:

[image]

mjs513, 2020-09-28:
Morning @MartyG-RealSense 
Thanks for the references.  Spent the day yesterday starting to collect references and techniques for using the D435 for SLAM and Obstacle Avoidance.  Now going to take while to go through them all:  Going to start with this one and then move on to the next: https://github.com/saifullah3396/vfh3d

Not sure you can answer me on this one since its a catkin question.  Since I built everything from src's for libsense I already have the src files plus a devel plus a build directory in my catkin_ws folder.  To build another package do i just leave the src alon directory alone and then delete the devel & build folders so I can do the next setup?  This is what is in the link above:
```
cd ~/catkin_ws
catkin_make
source ~/catkin_ws/devel/setup.bash
```
Just confused on this point?

MartyG-RealSense, 2020-09-28:
@mjs513   I will refer this question about catkin to @doronhi  who will be able to provide a better answer than I can.  Thanks for your patience!

mjs513, 2020-09-28:
@MartyG-RealSense 
Again - thanks for all your help.  Now the real fun begins :)

doronhi, 2020-09-29:
Sorry, too abstract for me.

mjs513, 2020-09-29:
@MartyG-RealSense , @doronhi 
Thanks again for all your help and patience in getting this working for me.

Going to go ahead and close the issue since its kind of been resolved.  Maybe I will go back to Windows at some point and give it a try :).

closed 2020-09-29 by mjs513

MartyG-RealSense, 2020-09-29:
Thanks very much for the update @mjs513  !