realsenseai/realsense-ros #1367: Sub paragraph mistype for 'unite_imu_method' in README.md

bigbellmercy, 2020-09-08:
In '[README.md](https://github.com/IntelRealSense/realsense-ros/blob/development/README.md)', the two paragraphs for 'linear_interpolation' and 'copy' should be indented under 'unite_imu_method' paragraph because they are options for it.

MartyG-RealSense, 2020-09-08:
Thanks very much for highlighting that, @bigbellmercy - I will tag the RealSense ROS wrapper developer @doronhi into this discussion to draw attention to your report.

bigbellmercy, 2020-09-08:
Thanks! Another option ‘none’ may be inserted on that.

MartyG-RealSense, 2020-09-18:
I will add a Documentation label to this case to ensure that it is kept open.

doronhi, 2020-09-20:
Thanks for bringing it to my attention and for the reminders. Done.

MartyG-RealSense, 2020-09-29:
Thanks so much @doronhi   I will close this case now as you are aware of the case and have acted on it.  Thanks again!

closed 2020-09-29 by MartyG-RealSense