realsenseai/realsense-ros #1368: D435i issues  IMU calibration and topics missing

FaboNo, 2020-09-08:
I am running Ubuntu 18.04 and ROS MELODIC

D435i version: 
```
rs-fw-update -l
connected devices:
1) Name: Intel RealSense D435I, serial number: 021222071136, update serial number: 020323050717, firmware version: 05.12.07.100, USB type: 3.2
```

I installed all the packages following the instructions provided here: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

My first problem is coming when I started realsense-viewer and activated motion, looks like the IMU is not calibrated but I though it was "factory calibrated"? So does it mean that I have to calibrate it myself?
```
fabrice@fabrice-desktop:~$ realsense-viewer 
 08/09 10:19:49,180 WARNING [140415000639232] (backend-v4l2.cpp:1269) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 08/09 10:20:06,106 WARNING [140415205071808] (ds5-motion.cpp:480) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 08/09 10:20:06,275 WARNING [140414117984000] (sensor.cpp:970) HID timestamp not found, switching to Host timestamps.
```
Second issue is coming when I launch

```
 fabrice@fabrice-desktop:~/catkin_rs435_ws$ roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true
... logging to /home/fabrice/.ros/log/d5309f1c-f1b0-11ea-bd3f-88b1116f16c5/roslaunch-fabrice-desktop-9264.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fabrice-desktop:33089/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
[... 128 lines cut ...]
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static
```
Can you help me to understand where is the problems?

Thank you

MartyG-RealSense, 2020-09-08:
Hi @FaboNo  Since ROS wrapper version **2.2.15**, infra, fisheye, IMU and pose streams are set to false by default.  You can enable them in your roslaunch statement or in the launch file.  For example, to enable infra1 and infra2 in roslaunch:

**roslaunch realsense2_camera rs_camera.launch enable_infra1:=true enable_infra2:=true**

Regarding the IMU calibration: originally, the D435i IMU sensor was not calibrated in the factory, though the other sensors in the camera are factory-calibrated.  I haven't heard information that this has changed.  A calibration of the IMU can be performed using a Python-based tool.  

https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera

Use of this tool can correct an issue where the accel values do not settle near zero even when the camera is stationary.  The software calibration is needed because the IMU component in the camera does not have its own internal calibration.

Once a calibration has been successfully completed and written to the camera hardware, an option becomes available in the RealSense Viewer to perform motion correction on the IMU data.  This 'fixes' the raw IMU data to take account of known inaccuracies in the IMU component so that the 'fixed' results are closer to what they should be.

It looks as though your librealsense installation has significant problems related to the kernel though.   This was indicated forst by the error ** Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!** and later by the long list of kernel related **xioctl(UVCIOC_CTRL_QUERY) failed** errors in your log.

Which computer / computing device do you have Ubuntu installed on, please?

FaboNo, 2020-09-08:
@MartyG-RealSense Thank you for your answer

Actually I did enter 

`roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true`

but i did not get all the expected topics.

I installed the librealsense on an Intel NUC. What I will do is deinstall and reinstall and calibrate the IMU. Then I will see what's happen and get back to you in case I run into problems again

Thank you

FaboNo, 2020-09-08:
one more information because after reinstalling everything and start `realsense-viewer` I still get the following message:
`08/09 12:30:33,495 WARNING [140351643637504] (backend-v4l2.cpp:1269) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!`

My kernel is `5.4.0-47-generic`

Should be compatible with the SDK?

MartyG-RealSense, 2020-09-08:
@FaboNo  Kernel 5.4 is not yet officially validated for use with librealsense (the most recent officially validated kernel is 5.3), though many RealSense users have been able to unofficially use 5.4 with librealsense.  

Assuming that your Intel NUC has an Intel processor chip, it should be able to work with the distribution package method of librealsense installation.

A way to build librealsense without kernel conflicts is to use a method called **RSUSB backend**.  It is useful for situations where the normal installation methods are not working.  It requires an internet connection to perform the installation but is not dependent on Linux versions or kernel versions and does not require patching.  the instructions in the link below are for SDK 2.36.0 - just substitute the 2.36.0 parts of the instructions for 2.37.0.

https://github.com/IntelRealSense/librealsense/issues/6845#issuecomment-659540316

FaboNo, 2020-09-08:
@MartyG-RealSense so you mean I have to recompile from source the 2.37 release?

MartyG-RealSense, 2020-09-08:
Yes, the RSUSB method uses the source code folder from the Releases page instead of the DKMS packages used in the Distribution method.  The zipped source code folder for 2.37.0 should be downloaded from the **Assets** list of that version and extracted.

FaboNo, 2020-09-08:
@MartyG-RealSense I successfully compiled the library. But I am a bit puzzled with the next steps (sorry about that). Do I need to do `sudo make install`?  Deinstall all the dkms one? ....

MartyG-RealSense, 2020-09-08:
No problem at all.  Having multiple librealsense installations made by different methods can sometimes cause conflicts, so uninstall the DKMS one if you encounter problems.

As far as I know there is not a further step after the CMake instruction completes with the RSUSB method.  Are you able to use applications such as RealSense Viewer in the new installation of librealsense, please?

FaboNo, 2020-09-08:
@MartyG-RealSense I did `sudo make install' to install the libraries on /usr/local 

when I entered
`fabrice@fabrice-desktop:~/src/librealsense-2.37.0/build$ tools/realsense-viewer/realsense-viewer `

I got the following messages, looks like I got even more problems because this time it did not find the gyros...
 ```
08/09 14:58:55,351 WARNING [139978201880320] (backend-v4l2.cpp:1270) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 08/09 14:59:11,385 WARNING [139978493462400] (types.cpp:78) hwmon command 0x4f( 0 208 0 0 ) failed (response -21= No data to return)
 08/09 14:59:11,385 WARNING [139978493462400] (ds5-motion.cpp:480) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 08/09 14:59:11,393 ERROR [139978493462400] (backend-hid.cpp:1026) sensor gyro_3d not found!
 08/09 14:59:11,393 WARNING [139977765676800] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.5/0003:8086:0B3A.000A/HID-SENSOR-200073.3.auto/iio:device1/buffer/enable

```

MartyG-RealSense, 2020-09-08:
Could you try uninstalling all of the RealSense DKMS packages from the previous installation with the command below please to see if that makes a difference?

**dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge**

FaboNo, 2020-09-08:
@MartyG-RealSense  yes I did it before and reinstall the package for the udev rules only.

MartyG-RealSense, 2020-09-08:
If the source code wa being compiled using the normal (non-RSUSB) method then the instructions state that the install statement that would be used to make sure that it is a clean installation is:

**sudo make uninstall && make clean && make && sudo make install**

FaboNo, 2020-09-08:
@MartyG-RealSense actually I just did `mkdir build,` `cd build` `cmake` .. and then `make`
Do I need to setup any parameter with make?

MartyG-RealSense, 2020-09-08:
I understand that the role of the uninstall and make clean commands is to do clear-up operations such as clearing the CMake cache before doing the install process.  Entering the commands exactly as listed in the instructions should be sufficient:

sudo make uninstall && make clean && make && sudo make install

You can use **make -j8** instead of **make** to speed up the build process by taking advantage pf multiple cores in the processor chip, though this may make the build process more prone to failure on low-end computer specifications.

FaboNo, 2020-09-09:
@MartyG-RealSense  I finally gave up on Melodic and did the installation on Kinetic and it works.
However after achieving the imu calibration I ended up with the following message:
```
[-0.00424225 -0.00318948 -0.00178359]
[1000 1000 1000 1000 1000 1000]
using 6000 measurements.
[[ 1.02273049e+00  1.38555019e-02 -8.31411095e-03]
 [-2.29580697e-03  1.01606005e+00  1.86339471e-02]
 [-7.21045918e-05  1.07868300e-03  1.01519079e+00]
 [ 4.39851838e-02  5.38233324e-01  3.48860872e-01]]
residuals: [  6.37946886   7.88940666 157.87314031]
rank: 4
singular: [437.13818699 430.25124687 427.77323301  76.94863166]
norm (raw data  ): 9.646208
norm (fixed data): 9.805130 A good calibration will be near 9.806650

```
With the camera fixed on the tripod, no vibration on the table, the topic /camera/imu (with <arg name="unite_imu_method"  default="linear_interpolation"/> in rs_camera.launch file)        gave me these values - and the values for the linear acceleration are REALLY BAD
```
angular_velocity: 
  x: -0.00167128804605
  y: -0.00168966234196
  z: -0.001714199665
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: -0.213709972834
  y: -9.75781635933
  z: -0.674374943543
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
```
Indeed x and z should be something like 0.00xxx  but definitively not 0.21 or 0.67, is there any mean to improve the calibration?

Thank you

MartyG-RealSense, 2020-09-09:
You can repeat the IMU calibration as many times as you feel necessary until you are satisfied with the results.  The results can be affected by how level the camera is when performing the steps where you are holding it side-on (holding it by the USB cable connector or by the other end with the USB cable resting on the flat surface).   It is difficult to position the camera completely level in these two particular orientations, so you may never acheve ideal results.

[image]

[image]

Once you have written a calibration to the camera, a new option in the RealSense Viewer becomes accessible to allow motion correction of the data.  The **rs-motion** SDK sample program can also balance gyro and accel data using a **Complementary Filter,** so this may be worth looking at if you aim to use IMU data in your own librealsense application.

https://dev.intelrealsense.com/docs/rs-motion

MartyG-RealSense, 2020-09-18:
Hi @FaboNo   Do you require further assistance with this case, please?   Thanks!

FaboNo, 2020-09-23:
@MartyG-RealSense I am sorry for my late response. I have just a last question, If i want to use Kalibr can you tell me what is the best camera model to use: is it pinhole-equi or pinhole-radtan ?
Thank you very much

MartyG-RealSense, 2020-09-23:
I have not got personal experience with using Kalibr so I cannot offer guidance about that, but a RealSense user did recently uae both pinhole-equi and pinhole-radian with the D435i IMU and produced charts.

https://github.com/IntelRealSense/librealsense/issues/7103

FaboNo, 2020-09-28:
@MartyG-RealSense thanks again. I did few calibrations because I need to get the distortion prameters nd they are not provided by /camera/infra1/camera_info  but what bother me is that the intrinsics I got are not similar to the ones provided by camera_info. In my case fx, fy = 381.82/381/82 (/camera/infra1/camera_info) but the calibration values are more between 386/386 -- 394/394 values.
My question is : does it matter? and each camera is really calibrated or not?
Thank you very much for your answer

MartyG-RealSense, 2020-09-28:
You can check the intrinsics and extrinsics of the camera with ** **rs-enumerate-devices -c** to list the camera's full calibration information for all of the sensors that it supports on that particular model.

If you calibrated the camera and wrote the calibration to the camera then yes, you can consider it calibrated.  Whether it is a **good** calibration will depend on the values that you wrote to the camera.  You can of course try the calibration again instead of writing it to the camera if you are not satisfied with the results at the end of the calibration process.

It may therefore be worth comparing the results from **rs-enumerate-devices** to those that were provided at the end of the calibration process to see how well they match up.

FaboNo, 2020-09-28:
@MartyG-RealSense ah yes quite comprehensive source of information! Thank you

However is it normal to get a distortion array with 0 values?

```
Intrinsic of "Infrared 2" / 640x480 / {Y8}
  Width:      	640
  Height:     	480
  PPX:        	319.074340820312
  PPY:        	237.638229370117
  Fx:         	381.82470703125
  Fy:         	381.82470703125
  Distortion: 	Brown Conrady
  Coeffs:     	0  	0  	0  	0  	0  
  FOV (deg):  	79.93 x 64.3
```

Does it mean that the image is already rectified?

MartyG-RealSense, 2020-09-28:
The 400 Series cameras deliberately set all the coefficients to zero, so that is normal.

https://github.com/IntelRealSense/librealsense/issues/1430

FaboNo, 2020-10-08:
@MartyG-RealSense  No I have all the information I required thanks for your support

closed 2020-10-08 by FaboNo

MartyG-RealSense, 2020-10-08:
Thanks for the update!

FaboNo, 2020-10-08:
@MartyG-RealSense  oups sorry as I am working with other people, I have another (last) question. Can you confirm that the infra1 and infra2 images are already rectified?

MartyG-RealSense, 2020-10-08:
The Y8 infrared channel is rectified.  Y16 in librealsense is unrectified.

FaboNo, 2020-10-08:
@MartyG-RealSense  it is interesting because as I am working with other researchers on VIO based algorithm, and they prefer to work on unrectified images as a precise calibration is required. Is it possible to setup in rs_camera.launch Y16 (meaning 25Hz max and 640x400) ? Is it supported in the ros wrapper?
Do I need to open a new thread because I closed this one?

MartyG-RealSense, 2020-10-08:
It is fine to continue on this discussion after it is closed if you wish to.

Y16 is a grayscale format channel and is unrectified because it is intended to be used for the purpose of RealSense camera calibration.   

I am not aware of a means to change a stream's format in RealSense ROS though.

FaboNo, 2020-10-08:
@MartyG-RealSense I see that I can change the parameters (Y16 and image size) with realsense-viewver but I do not know if I can save them (well I do not know how to). Even though it is possible, I do not know how the ros wrapper will react...

MartyG-RealSense, 2020-10-08:
Yes, you can set Y16 in librealsense @FaboNo  but librealsense and the RealSense ROS wrapper operate independently.  So setting an option in a librealsense-based program such as the Viewer does not affect the camera's behavior in ROS and vice versa.

FaboNo, 2020-10-09:
@MartyG-RealSense Yes this is what I was afraid of (well in some sense :) ) So regarding ros wrapper we are stick to Y8 and rectified images then. Thanks again for your support.

MartyG-RealSense, 2024-06-11:
Hi @Hanning-Liu Thanks for your question.  I will reply at https://github.com/IntelRealSense/realsense-ros/issues/3121