realsenseai/realsense-ros #1370: L515 |  time synchronization issue between IMU and image streams

lvj5077, 2020-09-08:
Hi,
We are experiencing time synchronization issue between IMU and image streams.
The timestamps among images are synchronized while there are always some seconds delay between IMU and image's timestamp. Also, the temporal offset is not a constant and changes every time.

In addition, when I turn off the "global timestamp" function, the IMU timestamp is switched to the internal clock while the image streams are still using the timestamp the same as the ROS.

bigbellmercy, 2020-09-09:
This is another similar report for IMU time sync problem of L515.

### Problem:
The timestamps of gyro and accel increase of itself continuously. The rate is about 2 ~ 4 ms per second.

### Environment:
Ubuntu 18.04 kernel **5.4.0-47**-generic
libRealSense v2.37.0
realsense-ros wrapper v2.2.16
L515 fw v1.5.0

### Reproducing

1. Install `librealsense2` debian package using `apt-get` as [Installing the packages](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages)
2. Install `realsense-ros` by `apt-get`
3. Edit `rs_camera.launch` in the `launch` folder of `realsense-ros`:
- Change `enable_gyro` to `true`
- Change `enable_accel` to `true`
- Change `accel_fps` to one of 100/200/400 Hz (because default 250 Hz is not supported for L515)
4. Connect L515 and Run `roslaunch realsense2_camera rs_camera.launch`
5. Check delay time:
`rostopic delay /camera/gyro/sample`
`rostopic delay /camera/accel/sample`

### Comment
1. According to [Linux Distribution](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#using-pre-build-packages), my Ubuntu kernel v5.4.0 is not listed as following:
> The Realsense DKMS kernel drivers package (librealsense2-dkms) supports Ubuntu LTS kernels 4.4, 4.8, 4.10, 4.13, 4.15, 4.18* 5.0* and 5.3*. 
2. To try to bypass the problem, if the dynamic parameter `global_time_enabled` is set to `true` for IMU(gyro and accel) and detph image, the timestamp value is doubled (twice bigger than current system time)
3. The issue [1370](https://github.com/IntelRealSense/realsense-ros/issues/1273) is similar but the solution is not apparent.

lvj5077, 2020-09-11:
Hi, @RealSenseCustomerSupport
I have upgraded to the latest FW 01.05.00.00, driver v2.38.1 and ROS  v2.2.17
on Ubuntu 18.04.5 kernel 5.4.0-47-generic
But I still have the time synchronization issue.
It happens when I am using rs-viewer and ROS. 

[image]
[image]

In the attached images, it seems the clock of IMU is always several seconds back from images, no matter what time(system or hardware) is in use.

@bigbellmercy mentioned that the timestamp value is doubled when ''global_time_enabled'' is on.
But the the doubled time still doesn't match.

gonultasbu, 2020-09-25:
@bigbellmercy 

what is the expected delay time after typing:

`rostopic delay /camera/gyro/sample`
or
`rostopic delay /camera/accel/sample`

For my case, the average, min or max delay values do not change significantly.

I am trying to replicate and potentially solve this issue for a robotics application myself but haven't been able to do so. I think there is something I am missing?

bigbellmercy, 2020-09-25:
@gonultasbu, in my case the delay time increases endlessly.
As I commented on my report above, how about check the kernel version? I am using unsupported kernel version. So I will try to down grade to older kernel version.

gonultasbu, 2020-09-25:
@bigbellmercy I will do it ASAP, the laptop is attached to a field testing robot which I currently don't have access to.

gonultasbu, 2020-09-25:
@bigbellmercy the kernel version is 5.3*

gonultasbu, 2020-09-25:
@bigbellmercy is the behaviour you're referring to as follows:

[image]

the max and avg delays being increased?

bigbellmercy, 2020-09-29:
@doronhi Is it a bug?

doronhi, 2020-10-02:
I'm looking into it.
Here is what I noticed so far:
accel is about 60 miliseconds behind the depth. I couldn't see an increasing time difference. @bigbellmercy , How did you measure about 2 ~ 4 ms per second? Using what method, so I could replicate?
Enabling global_time_stamp on any topic causes "Time is out of dual 32-bit range" exception but not doubling or changing the timestamp. How did you see that? using rostopic echo? I used Ubuntu 16.04 to test.

lvj5077, 2020-10-02:
> I'm looking into it.
> Here is what I noticed so far:
> accel is about 60 miliseconds behind the depth. I couldn't see an increasing time difference. @bigbellmercy , How did you measure about 2 ~ 4 ms [...]

@doronhi  "I couldn't see an increasing time difference" Yes. But the time difference is not a constant and changes every time.
"using rostopic echo" Using rostopic echo can detect the problem, and I also recorded the timestamps as text files. 

I have tested under Ubuntu 16, 18, windows10 and Mac Catalina. (Under Mac, launching IMU node will result in a crash)

bigbellmercy, 2020-10-03:
> I'm looking into it.
>
> Here is what I noticed so far:
>
> accel is about 60 miliseconds behind the depth. I couldn't see an increasing time difference. @bigbellmercy , How did you measure about 2 ~ 4 ms [...]

@doronhi I measured the time delays by 'rostopic delay' command.

doronhi, 2020-10-05:
I couldn't see the problem yet:
As for `rostopic delay`, I couldn't find any documentation as to what it measures. Delay between what? Publishing a topic and receiving it? If so, I don't think it has anything to do with the camera, librealsense2 or realsense2_camera wrapper.

As for the time difference not being constant - Using `rostopic echo` for checking that is tricky. Since the accel and depth topics publish in very different rates and since they do not measure at the same point in time, the difference between 2 samples will be different every time you check.
To see if there is a time difference I used `rosbag record /camera/depth/image_rect_raw /camera/accel/sample --duration=10`.
Then I used python to plot the accel and depth hardware-timestamps (the time given in msg.header.stamp) vs. the system-timestamp (the time of the message being recorded by rosbag record):
[image]
**Zooming In:**
[image]

In the graphs I see some fluctuations that could very possibly result from system delays in passing the messages. Also I see that for every system-time I receive a more recent depth message then accel message. I would expect it to be the other way around, for the depth image to have a longer processing time between taking and sending. I will check this out, but in any case I see a pretty much constant time difference between those 2 topics.
The fact that there is a different delay between taking the sample (imu or image) and receiving the sample is not so important and probably can not be avoided without delaying the quicker message. The question is if they are synced or not. That could be answered by a different method: moving the camera in one direction and then suddenly change direction. Then locating the movement-change event in both steams, depth and accel, and compare those timestamps. Or better yet, using a surface tracker on the depth image and compare the whole movement graphs of both streams.

lvj5077, 2020-10-09:
@doronhi “The question is if they are synced or not. That could be answered by a different method: moving the camera in one direction and then suddenly change direction. Then locating the movement-change event in both steams, depth and accel, and compare those timestamps. Or better yet, using a surface tracker on the depth image and compare the whole movement graphs of both streams.”

I am asking about the **synchronization issue** at the very beginning as the title of this ticket.
In our cases, even a temporal difference in millisecond would have a huge impact on the system while there's no easy fix.

doronhi, 2020-10-11:
@lvj5077 , I want to see if I understand you correctly:
Suppose that at time t=0 2 samples were taken - 1 sample for the imu and another for the image.
Now Suppose it took the image 0.5 second to reach your application while it took only 0.1 second for the imu message. That is, At time t=0.1 you'll receive an imu message with timestamp t=0 and at time t=0.5 you'll receive an image message with the same timestamp, t=0.
I I understand you correctly, this situation is not ok for you and you expect synchronization to hold the imu messages back and deliver them to you at the same time as the images. Am I correct?

lvj5077, 2020-10-12:
@doronhi You understood my problem correctly. 

"Now Suppose it took the image 0.5 second to reach your application while it took only 0.1 second for the imu message. That is, At time t=0.1 you'll receive an imu message with timestamp t=0 and at time t=0.5 you'll receive an image message with the same timestamp, t=0." 
But I am not sure if this mechanism results in the delay. (I don't have synchronization problem with other RealSense products, such as D435 D415 and other 200 series)

Right now, the IMU messages are always few seconds timestamped ahead. I want them them delivered at the same time as the images.

bigbellmercy, 2020-10-13:
> @bigbellmercy is the behaviour you're referring to as follows:
> 
> 
> 
> [image]
> 
> 
> 
> the max and avg delays being increased?

@gonultasbu, Yes, your screen shot is the same to mine. The average delay time increases on and on. Have you solved it?

gonultasbu, 2020-10-13:
@bigbellmercy I don't know how to interpret this average delay, is it a problem that needs to be solved or is it an expected behavior?

bigbellmercy, 2020-10-13:
@gonultasbu I think it is a problem because the delay time becomes 400 ms bigger after 100 sec passed since its increasing rate is about 4 m/s per second.

gonultasbu, 2020-10-13:
@bigbellmercy what does "delay time" mean here though? The delay between which metrics/quantities?

bigbellmercy, 2020-10-26:
@gonultasbu, when I search for it. It's just as below in ROS site:

> Display the delay for topic which has header.
> $ rostopic delay /topic_name

And other command was found in ROS site for checking delay:

> Display time in messages as offset from current time (e.g. to calculate lag/latency).
> $ rostopic echo --offset /topic_name

taogashi, 2021-01-27:
Same issue found.
realsense-ros commit: **c5403a28af7a82a30e9**
librealsense: **2.41.0**
realsense d455 firmware: **5.12.10**
os: **ubuntu 18.04.5 LTS**
ros: **melodic**
[image]

In two terminal, run
```
rostopic echo /camera/imu/header
```
and
```
rostopic echo /camera/infra1/image_rect_raw/header
```
respectively, then I got:
[image]

doronhi, 2021-02-11:
It seems, in my L515 that the device clock goes a little slower than the pc's clock. That accounts for the increasing delay of 3-4 mSec per sec. Turning `global_time_enabled` to `true` should handle that issue.
Regarding the huge delays incidents, (average delay: 96352 ...) I think it was caused by mixing timestamp domains.
I'm not sure I reproduced correctly but if so, #1684 should handle those 2 issues.

PR #1684 (2021-02-17): fix timestamp domain issues.
Add offset to ros_time only if the device uses hardware-clock. Otherwise use device time - either system_time or global_time.
This prevents situations when the base time was taken with one timestamp domain and then applied to frames with different timestamp domain.
This aims to fix #1370 and #1665
closed 2021-02-17 by doronhi