realsenseai/realsense-ros #1375: Raspberry Pi4B install librealsense-ros error

dingbaiyin, 2020-09-11:
I encountered an error when installing realsense-ros on the Raspberry Pi 4B, which cannot be resolved at present.
The installation reference link is as follows：
http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi
Problems encountered when running the 3.22（Resolving Dependencies with rosdep） command
As shown in the picture 1：
[image]
Picture 2 is the error report details：
[image]
Please provide some suggestions to solve the installation librealsense-ros problem,thank you very much .

dingbaiyin, 2020-09-11:
There is a problem with librealsense2.38.1, and Intel camera cannot be called normally. It is fine to replace it with librealsense2.36.0. I created a new issue about librealsense-ros and closed the issue。
Thank you very much.

MartyG-RealSense, 2020-09-12:
Hi @dingbaiyin   The **eleccelerator** guide for installing Librealsense and ROS on Pi 4 and Raspbisn Buster lists some of the errors that users of the **ROSberry** ROS installation guide might encounter and ways to fix them.

https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense

The guide then provides guidance about installing librealsense and realsense-ros:

**librealsense**
**https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense#Installing_librealsense**

**realsense-ros**
https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense#Installing_ROS_Package:_realsense-ros

MartyG-RealSense, 2020-09-21:
Hi @dingbaiyin   Do you require further assistance with this case, please?  Thanks!

dingbaiyin, 2020-09-28:
No,thanks.

closed 2020-09-28 by dingbaiyin

MartyG-RealSense, 2020-09-29:
@dingbaiyin   Thanks for the update!