realsenseai/realsense-ros #1376: Calculate 3D Coordinates

harshmittal2210, 2020-09-11:
Hello I have saved bagfile from realsense-viewer from D435i. I am having following ros topics in the bag:

/device_0/info
/device_0/sensor_0/Depth_0/image/data
/device_0/sensor_0/Depth_0/image/metadata
/device_0/sensor_0/Depth_0/info
/device_0/sensor_0/Depth_0/info/camera_info
/device_0/sensor_0/Depth_0/tf/0
/device_0/sensor_0/info
/device_0/sensor_0/option/Asic_Temperature/description
/device_0/sensor_0/option/Asic_Temperature/value
/device_0/sensor_0/option/Depth_Units/description
/device_0/sensor_0/option/Depth_Units/value
/device_0/sensor_0/option/Emitter_Enabled/description
/device_0/sensor_0/option/Emitter_Enabled/value
/device_0/sensor_0/option/Emitter_On_Off/description
/device_0/sensor_0/option/Emitter_On_Off/value
/device_0/sensor_0/option/Enable_Auto_Exposure/description
/device_0/sensor_0/option/Enable_Auto_Exposure/value
/device_0/sensor_0/option/Error_Polling_Enabled/description
/device_0/sensor_0/option/Error_Polling_Enabled/value
/device_0/sensor_0/option/Exposure/description
/device_0/sensor_0/option/Exposure/value
/device_0/sensor_0/option/Frames_Queue_Size/description
/device_0/sensor_0/option/Frames_Queue_Size/value
/device_0/sensor_0/option/Gain/description
/device_0/sensor_0/option/Gain/value
/device_0/sensor_0/option/Global_Time_Enabled/description
/device_0/sensor_0/option/Global_Time_Enabled/value
/device_0/sensor_0/option/Inter_Cam_Sync_Mode/description
/device_0/sensor_0/option/Inter_Cam_Sync_Mode/value
/device_0/sensor_0/option/Laser_Power/description
/device_0/sensor_0/option/Laser_Power/value
/device_0/sensor_0/option/Output_Trigger_Enabled/description
/device_0/sensor_0/option/Output_Trigger_Enabled/value
/device_0/sensor_0/option/Projector_Temperature/description
/device_0/sensor_0/option/Projector_Temperature/value
/device_0/sensor_0/option/Stereo_Baseline/description
/device_0/sensor_0/option/Stereo_Baseline/value
/device_0/sensor_0/option/Visual_Preset/description
/device_0/sensor_0/option/Visual_Preset/value
/device_0/sensor_0/post_processing
/device_0/sensor_1/info
/device_0/sensor_1/option/Auto_Exposure_Priority/description
/device_0/sensor_1/option/Auto_Exposure_Priority/value
/device_0/sensor_1/option/Backlight_Compensation/description
/device_0/sensor_1/option/Backlight_Compensation/value
/device_0/sensor_1/option/Brightness/description
/device_0/sensor_1/option/Brightness/value
/device_0/sensor_1/option/Contrast/description
/device_0/sensor_1/option/Contrast/value
/device_0/sensor_1/option/Enable_Auto_Exposure/description
/device_0/sensor_1/option/Enable_Auto_Exposure/value
/device_0/sensor_1/option/Enable_Auto_White_Balance/description
/device_0/sensor_1/option/Enable_Auto_White_Balance/value
/device_0/sensor_1/option/Exposure/description
/device_0/sensor_1/option/Exposure/value
/device_0/sensor_1/option/Frames_Queue_Size/description
/device_0/sensor_1/option/Frames_Queue_Size/value
/device_0/sensor_1/option/Gain/description
/device_0/sensor_1/option/Gain/value
/device_0/sensor_1/option/Gamma/description
/device_0/sensor_1/option/Gamma/value
/device_0/sensor_1/option/Global_Time_Enabled/description
/device_0/sensor_1/option/Global_Time_Enabled/value
/device_0/sensor_1/option/Hue/description
/device_0/sensor_1/option/Hue/value
/device_0/sensor_1/option/Power_Line_Frequency/description
/device_0/sensor_1/option/Power_Line_Frequency/value
/device_0/sensor_1/option/Saturation/description
/device_0/sensor_1/option/Saturation/value
/device_0/sensor_1/option/Sharpness/description
/device_0/sensor_1/option/Sharpness/value
/device_0/sensor_1/option/White_Balance/description
/device_0/sensor_1/option/White_Balance/value
/device_0/sensor_1/post_processing
/device_0/sensor_2/info
/device_0/sensor_2/option/Enable_Motion_Correction/description
/device_0/sensor_2/option/Enable_Motion_Correction/value
/device_0/sensor_2/option/Frames_Queue_Size/description
/device_0/sensor_2/option/Frames_Queue_Size/value
/device_0/sensor_2/option/Global_Time_Enabled/description
/device_0/sensor_2/option/Global_Time_Enabled/value

The topic: /device_0/sensor_0/Depth_0/image/data provides 8bit 2D depth image. How should use it to convert into 3D point cloud in c++ using ros.

MartyG-RealSense, 2020-09-12:
Hi @harshmittal2210    How I would recommend approaching this is to record the point cloud into the bag in ROS instead of using the RealSense Viewer to create the bag, so you can then play the ROS-recorded bag back and display it as a point cloud in RViz.  Doronhi the RealSense ROs wrapper developer explains a method for dong so in the link below:

https://github.com/IntelRealSense/realsense-ros/issues/1076#issuecomment-586702026

harshmittal2210, 2020-09-12:
Hi @MartyG-RealSense  thanks for the reply. It would be great if you can also let me know which topic can provide me directly 3D coordinates in form of (x,y,z). As recording all the topics creates 1.4gb for approx 20sec of recording.

MartyG-RealSense, 2020-09-12:
Bag recording in librealsense creates similarly large GB-sized files, so it may be difficult to avoid.  You may be able to reduce the file size by reducing resolution.  For example, in a test in the RealSense Viewer that I just conducted, a 10 second recording of depth at 848x480 resulted in a 281 mb bag file, whilst a 10 second recording at 640x480 produced a 137 mb file.

If you are referring to a live point cloud from the camera when you say "directly", the RealSense ROS wrapper documentation has a tutorial for generating a point cloud in RViz using an aligned depth topic:

https://github.com/IntelRealSense/realsense-ros#rgbd-point-cloud

Please look in **pointcloud2** for the coordinate information that you seek.

harshmittal2210, 2020-09-14:
Yes, that is true that while recording with RealSense Viewer the file size is decreased, but also it does not store all the ros topics. /device_0/sensor_0/Depth_0/image/data has a depth image but it does not help in getting the world coordinate of that point. 

Moreover, if I record using
` roslaunch realsense2_camera rs_camera.launch
`
`rosbag record -a`
then the topics are different from the one recorded using realsense viewer. Also, the bagfile recorded shows header error when viewed over Realsense-viewer.

MartyG-RealSense, 2020-09-14:
The RealSense SDK's bag format is mostly based on the 'rosbag' structure.  Depending on what data was recorded into the bag, there may be issues when trying to play in the RealSense SDK a bag recorded in ROS outside of the RealSense SDK, with a general ROS bag record-playback command such as **rosbag record**.

Dorodnic the RealSense SDK Manager comments on the subject in a discussion in the link below:

https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

harshmittal2210, 2020-09-14:
Okay got it, thank you!

harshmittal2210, 2020-09-17:
Hi @MartyG-RealSense In the following example below mentioned topics are used to calculate the depth
https://github.com/IntelRealSense/librealsense/issues/2215#issue-348685746

Topic:
/device_0/sensor_0/Depth_0/info/camera_info
/device_0/sensor_0/option/Depth Units/value
/device_0/sensor_0/Depth_0/image/data

There is no CMakelist.txt provided along with and also the calculations seems to be a bit complicated. It would be great if you can help me in either providing a cmake file or some other way so that I can also implement the same.

MartyG-RealSense, 2020-09-17:
When you refer to the above topics being used to calculate the depth, do you mean that they are used to do this in the code of the **RealSenseHelper.zip** program that is at the bottom of the comment that you linked to, please?

If you are still interested in generating a real-time point cloud with **pointcloud2** then you may find useful a discussion in which converting pointcloud2 into xyzi coordinates with **PCL** is referenced.

https://github.com/IntelRealSense/realsense-ros/issues/1386

harshmittal2210, 2020-09-18:
Hi @MartyG-RealSense Yes I am referring to the code present in RealSenseHelper.zip only.

Sure I'll look into the discussion. Thanks

MartyG-RealSense, 2020-09-18:
In regard to alternate ways to the zip-file program for calculating depth using ROS topics, the discussions in the links below may be useful:

https://github.com/IntelRealSense/realsense-ros/issues/1341
https://github.com/IntelRealSense/realsense-ros/issues/714

MartyG-RealSense, 2020-09-26:
Hi @harshmittal2210   Do you still require assistance with this case, please?  Thanks!

harshmittal2210, 2020-09-27:
Hi, @MartyG-RealSense Thank you for the help.

closed 2020-09-27 by harshmittal2210