realsenseai/realsense-ros #1378: crash with L515

RapIt, 2020-09-11:
Hi,

I am not able to run the basic `rs_camera.launch` with the Realsense L515. The node does find the device but crashes when looking for available sensors. All streams are working fine in the realsense-viewer. I just followed the source installation instructions and everything built without any errors under Ubuntu 18.04 with ros-meldoic.

```
process[camera/realsense2_camera-3]: started with pid [5720]
[ INFO] - /camera/realsense2_camera_manager: [1599827845.256427731] Initializing nodelet with 12 worker threads.
[ INFO] - /camera/realsense2_camera_manager: [1599827845.499869949] RealSense ROS v2.2.17
[ INFO] - /camera/realsense2_camera_manager: [1599827845.499971318] Built with LibRealSense v2.38.1
[ INFO] - /camera/realsense2_camera_manager: [1599827845.500032355] Running with LibRealSense v2.38.1
[ INFO] - /camera/realsense2_camera_manager: [1599827845.519634338]
[ INFO] - /camera/realsense2_camera_manager: [1599827845.589633581] Device with serial number 00000000f0191018 was found.

[ INFO] - /camera/realsense2_camera_manager: [1599827845.589670019] Device with physical ID /sys/devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:02.0/0000:38:00.0/usb4/4-2/4-2.3/4-2.3.1/4-2.3.1:1.0/video4linux/video0 was found.
[ INFO] - /camera/realsense2_camera_manager: [1599827845.589681267] Device with name Intel RealSense L515 was found.
[ INFO] - /camera/realsense2_camera_manager: [1599827845.589974420] Device with port number 4-2.3.1 was found.
[ INFO] - /camera/realsense2_camera_manager: [1599827845.591080574] getParameters...
[ INFO] - /camera/realsense2_camera_manager: [1599827845.617930421] setupDevice...
[ INFO] - /camera/realsense2_camera_manager: [1599827845.617946328] JSON file is not provided
[ INFO] - /camera/realsense2_camera_manager: [1599827845.617960071] ROS Node Namespace: camera
[ INFO] - /camera/realsense2_camera_manager: [1599827845.617970265] Device Name: Intel RealSense L515
[ INFO] - /camera/realsense2_camera_manager: [1599827845.617978682] Device Serial No: 00000000f0191018
[ INFO] - /camera/realsense2_camera_manager: [1599827845.617988445] Device physical port: /sys/devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:02.0/0000:38:00.0/usb4/4-2/4-2.3/4-2.3.1/4-2.3.1:1.0/video4linux/video0
[ INFO] - /camera/realsense2_camera_manager: [1599827845.617997576] Device FW version: 01.05.00.00
[ INFO] - /camera/realsense2_camera_manager: [1599827845.618008414] Device Product ID: 0x0B64
[ INFO] - /camera/realsense2_camera_manager: [1599827845.618016925] Enable PointCloud: Off
[ INFO] - /camera/realsense2_camera_manager: [1599827845.618026498] Align Depth: Off
[ INFO] - /camera/realsense2_camera_manager: [1599827845.618034688] Sync Mode: Off
[ INFO] - /camera/realsense2_camera_manager: [1599827845.618055780] Device Sensors:
 11/09 14:37:25,618 WARNING [140211005691648] (backend-v4l2.cpp:1269) Pixel format 20204746-0000-0010-8000-00aa003 likely requires patch for fourcc code FG  !
[camera/realsense2_camera_manager-2] process has died [pid 5719, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=~/.ros/log/8b4480a8-f42b-11ea-8c80-c03ebaa1d8f5/camera-realsense2_camera_manager-2.log].
log file: ~/.ros/log/8b4480a8-f42b-11ea-8c80-c03ebaa1d8f5/camera-realsense2_camera_manager-2*.log
```

The master log file has some of the following errors:
```
[rosmaster.master][INFO] 2020-09-11 14:37:27,218: publisherUpdate[/camera/realsense2_camera_manager/bond] -> http://nuxor:38865/ ['http://nuxor:41117/']
[rosmaster.master][INFO] 2020-09-11 14:37:27,218: publisherUpdate[/camera/realsense2_camera_manager/bond] -> http://nuxor:38865/ ['http://nuxor:41117/']: sec=0.00, exception=[Errno 111] Connection refused
[rosmaster.threadpool][ERROR] 2020-09-11 14:37:27,218: Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1311, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1459, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1078, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 894, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 856, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 833, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
    raise err
error: [Errno 111] Connection refused

[rosmaster.master][INFO] 2020-09-11 14:37:27,317: -PUB [/rosout_agg] /rosout http://nuxor:42167/
```

I saw the Pixel format problem in other logs without crashing so I guess the crash does not necessarily have to do something with the warning.

Does anyone have an idea what went wrong?

RealSenseSupport, 2020-10-12:
Hi @RapIt 

What hardware platform / CPU are you running this on? And when you say source installation instructions which specifically? And did you build from source librealsense, realsense-ros, or both?

Also could you try launching with:

roslaunch realsense2_camera rs_rgbd.launch enable_infra:=true

to rule out if you're running into a variant of this known issue:

https://github.com/IntelRealSense/realsense-ros/issues/1348

Thanks

RapIt, 2020-11-26:
Hi, sorry for the late reply. I made a source installation for realsense-ros and installed librealsense from the available ubuntu binaries as suggested in "Method 2" in the readme. Ran on Intel i7-10850H.

But nevermind, I recently upgraded everything to the latest version 2.40 and the issue no longer occurs. So I will close the issue. Thank you!

closed 2020-11-26 by RapIt