realsenseai/realsense-ros #1379: Emitter Turns On Automatically

fiorano10, 2020-09-11:
We have a multi-cam setup all, D435i's. We disable the emitter using dynamic_reconfigure when we launch the node, but for some unknown reason, it turns itself on under low light. 
Currently we have the parameter in a yaml file as such, `emitter_enabled: False` should this be 0 or 1?
dynamic param state:
```
{'laser_power': 150.0, 'groups': {'laser_power': 150.0, 'parent': 0, 'global_time_enabled': True, 'inter_cam_sync_mode': 1.0, 'gain': 16, 'groups': {}, 'id': 0, 'output_trigger_enabled': False, 'exposure': 8500, 'name': 'Default', 'parameters': {}, 'emitter_enabled': 0, 'enable_auto_exposure': True, 'error_polling_enabled': True, 'state': True, 'visual_preset': 0, 'emitter_on_off': False, 'type': '', 'frames_queue_size': 16}, 'output_trigger_enabled': False, 'emitter_enabled': 0, 'enable_auto_exposure': True, 'global_time_enabled': True, 'inter_cam_sync_mode': 1.0, 'visual_preset': 0, 'error_polling_enabled': True, 'gain': 16, 'frames_queue_size': 16, 'emitter_on_off': False, 'exposure': 8500}
```
[image]
[image]

### Your environment
*Camera: D435i
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
*Librealsense: LibRealSense v2.33.1
*realsense-ros: RealSense ROS v2.2.9

MartyG-RealSense, 2020-09-12:
Hi @fiorano10   You can disable the IR emitter in the launch file with a **rosparam** instruction instead of doing it during runtime with dynamic_reconfigure:

**<rosparam>
/camera/stereo_module/emitter_enabled: false
</rosparam>**

fiorano10, 2020-09-14:
Does it matter how we disable the emitter? Why does it come on when disabled during after being launched?

MartyG-RealSense, 2020-09-14:
If you do not need to change an option during runtime then I suggest setting it to false in the roslaunch statement (or preferably within the roslaunch file with rosparams if you have a lot of settings to apply).  

There was apparently a firmware issue that caused the emitter mode to be shifted by 1, and a fix was put in firmware **5.12.5**.  Are you using a firmware older than 5.12.5 in the camera, please?

https://github.com/IntelRealSense/librealsense/issues/6191

MartyG-RealSense, 2020-09-21:
Hi @fiorano10   Do you require further assistance with this case, please?  Thanks!

doronhi, 2020-09-23:
@fiorano10, I am not aware of a phenomena like you describe, that the emitter turns ON automatically.
I guess the issue occurs whether there is 1 D435i attached or multiple devices.
Can you reproduce this automatic enabling of the emitter using realsense-viewer with 1 camera attached?
What is the camera's firmware version?
Can you reproduce using latest release of librealsense2?
If so, can you take a print-screen, before and after, while the "show frame metadata" icon is on?

fiorano10, 2020-09-23:
@doronhi This occurs when multiple D435is are attached (up to 4). I can try running 1 D435i with realsense-viewer to reproduce the issue. Camera firmware was 5.12.1, but can be seen on 5.12.5 also. I have not updated the librealsense version, still running 2.33.1. Is there a way to log emitter status using ROS? Does it matter how/when the emitter is disabled, dynamic-reconfigure vs rosparam?

doronhi, 2020-09-24:
It should not matter how you set the parameter and based on @MartyG-RealSense 's remarks I assume you set the correctly. Still, there was an issue once that I can't find with the following description:

> We're using the emitter on/off feature and using the frame metadata to detect which frames have emitter on. We have discovered that enabling the option when the ROS node is starting up causes the frame metadata to get into a bad off-by-one state, where it reports:
> 1. Frames with emitter on are reported as emitter off
> 2. Frames with emitter off are reported as emitter on

> The line "_sensors[DEPTH].set_option(RS2_OPTION_EMITTER_ON_OFF, 1)" is inserted at the end of setupDevice (https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/base_realsense_node.cpp#L610)

> However, if we don't do this in setupDevice, and instead do a hacky dynamic_reconfigure call using a script (
rosrun dynamic_reconfigure dynparam set /camera0/stereo_module emitter_on_off 1), the metadata behaves as designed.

That issue was fixed but it's not unheard of that setting the parameter after start streaming has a different, unintentional, effect.

There is currently no way to log emitter status using ROS.
You could do the following:
Add the following line inside the [publishFrame() function](https://github.com/IntelRealSense/realsense-ros/blob/de64e737c6a9de4ec70d89a18fe68b526c1460f5/realsense2_camera/src/base_realsense_node.cpp#L2204):
```
auto projector_state = f.get_frame_metadata(RS2_FRAME_METADATA_FRAME_LASER_POWER_MODE);
ROS_INFO_STREAM("projector_state : " << projector_state);
```

MartyG-RealSense, 2020-10-02:
Hi @fiorano10   Do you still require assistance with this case, please?  Thanks!

fiorano10, 2020-10-03:
Working on upgrading the library version and ros driver. Will update soon

fiorano10, 2020-10-05:
@doronhi decided to disable the emitter in the base node itself by default.

MartyG-RealSense, 2020-10-05:
Thanks very much for the update @fiorano10

doronhi, 2020-10-05:
I am happy you have a solution that works for you and you can carry on with your project.
I wish I could reproduce, preferably inside the office. I tried getting the camera in and out of big boxes but it didn't happen.
If you ever find the time and get around to it, I would love to see an indoor video of the phenomena with rqt_reconfigure's options visible beside. As I said, if you ever find the spare time :)
Good luck in the future.

fiorano10, 2020-10-26:
@doronhi I seem to be getting errors when trying to get frame metadata for projector data. It only happens on kernel 5.4

`metadata not available`

doronhi, 2020-10-28:
kernel patches are needed. Those will not be available as debian package until version 2.40.0 of librealsense2 (a couple of weeks)
In the meanwhile you can download [librealsense code](https://github.com/IntelRealSense/librealsense.git) and run `./scripts/patch-realsense-ubuntu-lts.sh` (Copied from [librealsense2 installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)). This will update your kernel and the missing metadata entries will become available.

fiorano10, 2021-01-21:
@dorodnic I can't get the projector state even on librealsense v2.41 on kernel 5.3/5.4 without the patch. Do I need the patch if I want to get the frame metadata irrespective of librealsense and kernel version?

MartyG-RealSense, 2021-01-22:
Hi @fiorano10   I have highlighted your re-opened case to Intel to bring it to their attention.

doronhi, 2021-01-24:
> @dorodnic I can't get the projector state even on librealsense v2.41 on kernel 5.3/5.4 without the patch. Do I need the patch if I want to get the frame metadata irrespective of librealsense and kernel version?

The answer is yes. You need the patch to get the frame metadata and you need the metadata to get the projector state.
There is an alternative, build librealsense2 using the FORCE_RSUSB flag to use the rsusb backend but right now this is not considered a stable option.

fiorano10, 2021-01-25:
The patch does the trick, thank you

closed 2021-01-25 by fiorano10

AndreV84, 2021-04-03:
@dorodnic so how to disable the emitter with rose node?

MartyG-RealSense, 2021-04-03:
Hi @AndreV84  The correct name tag for your question of **how to disable the emitter with ROS node?** is @doronhi

gsanya, 2021-10-05:
If someone is still interested:
For me this worked to turn off the emitter:
`<rosparam> /camera/stereo_module/emitter_enabled: 2</rosparam>`
I put it in my own launch file after starting the camera lik this:
```
<include file="$(find realsense2_camera)/launch/rs_camera.launch">  
   <arg name="enable_gyro" value="true" />  
   <arg name="enable_accel" value="true" />  
   <arg name="unite_imu_method" value="linear_interpolation" />  
   <arg name="color_width" value="640" />  
   <arg name="color_height" value="480" />  
   <arg name="color_fps" value="30" />  
   <arg name="enable_infra1" value="true" />  
   <arg name="enable_infra2" value="true" />  
   <arg name="infra_width" value="640" />  
   <arg name="infra_height" value="480" />  
   <arg name="infra_fps" value="30" />  
</include>  
<rosparam> /camera/stereo_module/emitter_enabled: 2</rosparam>
```
If I understand correctly, now the emitter_enabled param is an enum, and 2 means off. This param doesn't exists before setting it like this, so you will only find It in the GUI, after creating it in some lounch file.

MartyG-RealSense, 2021-10-05:
Thanks very much @gsanya for sharing your solution   :)

zhuoqun-chen, 2024-09-15:
> If someone is still interested: For me this worked to turn off the emitter: `<rosparam> /camera/stereo_module/emitter_enabled: 2</rosparam>` I put it in my own launch file after starting the camera [...]

This trick is so helpful for our setup! thx!