realsenseai/realsense-ros #1381: An error has occurred during frame callback: map::at (RealSense L515 ROS)

blue-20, 2020-09-12:
I'm trying to launch the L515 with a ROS node. I installed the latest drivers and ROS node in accordance with IntelRealSense' guide.
https://github.com/IntelRealSense/librealsense/releases 

I tried to start the node with the launch file.
roslaunch realsense2_camera rs_camera.launch
But I get the following same errors after it starts.

An error has occurred during frame callback: map::at
An error has occurred during frame callback: map::at
...

I found similar questions in past issues.
The solutions were advised to lower the FPS or resolution because the CPU can not catch up with the processing, 
But I could not solve the error by this.
I do not understand the reason, but sometimes it starts up successfully without errors.
Could you give me a solution?
Best regards.

The following is the display after launching.

SUMMARY

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 20
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: True
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 6
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 20
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: melodic
 * /rosversion: 1.14.9

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [10771]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7a071510-f50e-11ea-b03c-5c514fca907c
process[rosout-1]: started with pid [10782]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [10789]
process[camera/realsense2_camera-3]: started with pid [10790]
[ INFO] [1599925312.462250768]: Initializing nodelet with 4 worker threads.
[ INFO] [1599925312.703738821]: RealSense ROS v2.2.16
[ INFO] [1599925312.703825925]: Built with LibRealSense v2.38.1
[ INFO] [1599925312.703902348]: Running with LibRealSense v2.38.1
[ INFO] [1599925312.717950693]:  
[ INFO] [1599925312.794219599]: Device with serial number 00000000f0190528 was found.

[ INFO] [1599925312.794321664]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/video4linux/video2 was found.
[ INFO] [1599925312.794376671]: Device with name Intel RealSense L515 was found.
[ INFO] [1599925312.795373450]: Device with port number 3-3 was found.
[ INFO] [1599925312.798207460]: getParameters...
[ INFO] [1599925312.843187942]: setupDevice...
[ INFO] [1599925312.843235476]: JSON file is not provided
[ INFO] [1599925312.843256429]: ROS Node Namespace: camera
[ INFO] [1599925312.843296127]: Device Name: Intel RealSense L515
[ INFO] [1599925312.843327055]: Device Serial No: 00000000f0190528
[ INFO] [1599925312.843351097]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/video4linux/video2
[ INFO] [1599925312.843374753]: Device FW version: 01.05.00.00
[ INFO] [1599925312.843402544]: Device Product ID: 0x0B64
[ INFO] [1599925312.843423818]: Enable PointCloud: On
[ INFO] [1599925312.843446173]: Align Depth: Off
[ INFO] [1599925312.843467781]: Sync Mode: On
[ INFO] [1599925312.843534661]: Device Sensors: 
 13/09 00:41:52,843 WARNING [140110045705984] (backend-v4l2.cpp:1269) Pixel format 20204746-0000-0010-8000-00aa003 likely requires patch for fourcc code FG  !
[ INFO] [1599925312.844220965]: L500 Depth Sensor was found.
[ INFO] [1599925312.846770323]: RGB Camera was found.
[ INFO] [1599925312.846968878]: Motion Module was found.
[ INFO] [1599925312.847009412]: Add Filter: pointcloud
[ INFO] [1599925312.847657797]: num_filters: 1
[ INFO] [1599925312.847690205]: Setting Dynamic reconfig parameters.
[ INFO] [1599925312.892086216]: Done Setting Dynamic reconfig parameters.
[ INFO] [1599925312.892543306]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1599925312.893078101]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1599925312.893116841]: setupPublishers...
[ INFO] [1599925312.895413551]: Expected frequency for depth = 30.00000
[ INFO] [1599925312.924182186]: Expected frequency for color = 30.00000
[ INFO] [1599925312.943652548]: setupStreams...
[ INFO] [1599925312.995127069]: insert Depth to L500 Depth Sensor
[ INFO] [1599925312.995262088]: insert Color to RGB Camera
 13/09 00:41:53,003 WARNING [140110045705984] (backend-v4l2.cpp:1269) Pixel format 20204746-0000-0010-8000-00aa003 likely requires patch for fourcc code FG  !
[ INFO] [1599925313.043207867]: SELECTED BASE:Depth, 0
[ INFO] [1599925313.158681555]: RealSense Node Is Up!
[ERROR] [1599925314.573377482]: An error has occurred during frame callback: map::at
[ERROR] [1599925314.633908041]: An error has occurred during frame callback: map::at
[ERROR] [1599925314.641481910]: An error has occurred during frame callback: map::at
[ERROR] [1599925314.703811995]: An error has occurred during frame callback: map::at
....

RealSenseSupport, 2020-09-14:
Hi @blue-20 

I believe you are experiencing the same issue as https://github.com/IntelRealSense/realsense-ros/issues/1348 . Can you try and see if the workaround suggested there helps?

If so this is a known issue and a fix is pending in https://github.com/IntelRealSense/librealsense/pull/7269.

Thanks

blue-20, 2020-09-14:
Thank you for your answer.
When I tried the solution suggested in [#1348](https://github.com/IntelRealSense/realsense-ros/issues/1348), it worked without any errors. I'm sorry for the duplication of the question.
Thank you for your cooperation.

closed 2020-09-14 by blue-20