realsenseai/realsense-ros #1385: Unable to launch D455 camera

fiorano10, 2020-09-15:
It is connected as a USB 2.1 device
Initializing nodelet with 12 worker threads.
**RealSense ROS v2.2.9**
**LibRealSense v2.38.1**

```
[ INFO] [1600142682.158637580]:  
[ INFO] [1600142682.171507518]: Device with serial number 026222257541 was found.
[ INFO] [1600142682.171530119]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/video4linux/video24 was found.
[ INFO] [1600142682.171545876]: Device with name Intel RealSense D455 was found.
[ INFO] [1600142682.172878672]: Device with port number 1-7 was found.
[FATAL] [1600142682.172974831]: Unsupported device! Product ID: 0x0B5C
terminate called after throwing an instance of 'rs2::wrong_api_call_sequence_error'
  what():  stop_streaming() failed. UVC device is not streaming!
[camera/realsense2_camera_manager-1] process has died [pid 21870, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=$HOME/.ros/log/14620d34-f6ab-11ea-9378-00012999c330/camera-realsense2_camera_manager-1.log].
log file: $HOME/.ros/log/14620d34-f6ab-11ea-9378-00012999c330/camera-realsense2_camera_manager-1*.log
```

MartyG-RealSense, 2020-09-15:
Hi @fiorano10   It is important to match the librealsense SDK version as closely as possible with the SDK version recommended in the notes for a particular ROS wrapper version on the wrapper's **Releases** page.  

librealsense** SDK version **2.38.1** should be matched with RealSense ROS wrapper **2.2.17**

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.17

fiorano10, 2020-09-15:
Yup, I didn't rebuild after merging the updates. The camera still doesn't want to publish frames, only color stream enabled.

```
Device FW version: 05.12.06.00
color stream is enabled - width: 640, height: 480, fps: 15
RealSense ROS v2.2.17
Built with LibRealSense v2.38.1
Running with LibRealSense v2.38.1
 15/09 12:09:31,663 WARNING [140689838786304] (backend-v4l2.cpp:1034) Frames didn't arrived within 5 seconds

```

MartyG-RealSense, 2020-09-15:
Thanks for the update.  Does it make a difference if you add the term **initial_reset:=true** (a hardware reset) to your roslaunch statement?

fiorano10, 2020-09-15:
It doesn't

MartyG-RealSense, 2020-09-15:
Infrared, IMU and fisheye are disabled by default in ROS wrapper version 2.2.15 onward anyway.

I actually just ran some tests at 640x480 and 15 FPS in the RealSense Viewer in SDK 2.38.1 and was getting a continuous stream of errors.  I updated the firmware to the new 5.2.7 one in the Viewer and all of the errors disappeared.  So I was about to suggest a firmware upgrade, but you got there first.

fiorano10, 2020-09-15:
It works on a USB 3.1 port but can't get even get just the color stream on USB 2.0? What else can be tried?

MartyG-RealSense, 2020-09-16:
As you have updated to the latest SDK and wrapper versions since you had a conflict with the D455's Product ID (PID) code 0x0B5C in the log in your opening comment and that conflict no longer seems to be present, could you confirm whether you experience the same problem on D435i (as past cases of yours such as https://github.com/IntelRealSense/realsense-ros/issues/1379 indicate that you own some).

fiorano10, 2020-09-23:
I will check and confirm

MartyG-RealSense, 2020-10-02:
Hi @fiorano10   Do you still require assistance with this case, please?  Thanks!

fiorano10, 2020-10-03:
Decided to use a USB 3 hub with another realsense, seems to get detected as a USB 3.2 device.

closed 2020-10-03 by fiorano10