realsenseai/realsense-ros #1386: How to publish pointcloud from ROS as pointxyzi pointcloud type

sonumathur, 2020-09-15:
There are following type of points in pointcloud data as in image:
[image]

Using ROS i want to produce PointXYZI pointcloud.
I run following Commands as:
rosrun rviz rviz
roslaunch realsense_ros_camera rs_rgbd.launch
rosrun pcl_ros pointcloud_to_pcd input:=/camera/depth_registered/points

under the topic /camera/depth_registered/points pointcloud produced is PointXYZRGB pointcloud.

How i can produce PointXYZI pointclud......?

MartyG-RealSense, 2020-09-15:
Hi @sonumathur   The only librealsense ROS GitHub case involving obtaining a point cloud with intensity information that I know of is in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1046

There was a research dissertation using RealSense and ROS that used PCL to convert the **pointcloud2** message into an XYZI point cloud.  The bottom of page 29 of the document explains this.

https://lib.ugent.be/fulltxt/RUG01/002/785/837/RUG01-002785837_2019_0001_AC.pdf

MartyG-RealSense, 2020-09-26:
Hi @sonumathur   Do you still require assistance with this case, please?  Thanks!

sonumathur, 2020-09-26:
Hi @MartyG-RealSense .  Thanks for query.

I have see a page on web at:
[](http://docs.ros.org/indigo/api/jsk_pcl_ros_utils/html/classjsk__pcl__ros__utils_1_1PointCloudXYZRGBToXYZ.html)

Installed jsk_pcl_ros_utils in ROS but but not understand how to load that code.

Thanks

sonumathur, 2020-09-26:
Hi @MartyG-RealSense, I have One query about a Documentation at link:

http://docs.ros.org/melodic/api/librealsense2/html/files.html

At this link various Function and files that are part of Realsense2 are provided.

My problum can be solved by changing the senser msg types given at:

http://docs.ros.org/melodic/api/librealsense2/html/point__field__conversion_8h_source.html 

As
/*
  * Software License Agreement (BSD License)
  *
  * Robot Operating System code by the University of Osnabrück
  * Copyright (c) 2015, University of Osnabrück
  * All rights reserved.
  *
  * Redistribution and use in source and binary forms, with or without
  * modification, are permitted provided that the following conditions
  * are met:
  *
  *   1. Redistributions of source code must retain the above 
  *      copyright notice, this list of conditions and the following
  *      disclaimer.
  *
  *   2. Redistributions in binary form must reproduce the above 
  *      copyright notice, this list of conditions and the following
  *      disclaimer in the documentation and/or other materials provided
  *      with the distribution.
  *
  *   3. Neither the name of the copyright holder nor the names of its
  *      contributors may be used to endorse or promote products derived
  *      from this software without specific prior written permission.
  *
  *
  * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
  * "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED
  * TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR
  * PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR
  * CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL,
  * EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO,
  * PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS;
  * OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY,
  * WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR
  * OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  * ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
  *
  *
  *
  * point_field_conversion.h
  *
  * Created on: 16.07.2015
  *  Authors: Sebastian Pütz <spuetz@uni-osnabrueck.de>
  */
 
 #ifndef SENSOR_MSGS_POINT_FIELD_CONVERSION_H
 #define SENSOR_MSGS_POINT_FIELD_CONVERSION_H
 
 namespace sensor_msgs{
   template<int> struct pointFieldTypeAsType {};
   template<> struct pointFieldTypeAsType<sensor_msgs::PointField::INT8>    { typedef int8_t   type; };
   template<> struct pointFieldTypeAsType<sensor_msgs::PointField::UINT8>   { typedef uint8_t  type; };
   template<> struct pointFieldTypeAsType<sensor_msgs::PointField::INT16>   { typedef int16_t  type; };
   template<> struct pointFieldTypeAsType<sensor_msgs::PointField::UINT16>  { typedef uint16_t type; };
   template<> struct pointFieldTypeAsType<sensor_msgs::PointField::INT32>   { typedef int32_t  type; };
   template<> struct pointFieldTypeAsType<sensor_msgs::PointField::UINT32>  { typedef uint32_t type; };
   template<> struct pointFieldTypeAsType<sensor_msgs::PointField::FLOAT32> { typedef float    type; };
   template<> struct pointFieldTypeAsType<sensor_msgs::PointField::FLOAT64> { typedef double   type; };
   
[... 57 lines cut ...]
           writePointCloud2BufferValue<sensor_msgs::PointField::UINT16, T>(data_ptr, value);
           break;
         case sensor_msgs::PointField::INT32:
           writePointCloud2BufferValue<sensor_msgs::PointField::INT32, T>(data_ptr, value);
           break;
         case sensor_msgs::PointField::UINT32:
           writePointCloud2BufferValue<sensor_msgs::PointField::UINT32, T>(data_ptr, value);
           break;
         case sensor_msgs::PointField::FLOAT32:
           writePointCloud2BufferValue<sensor_msgs::PointField::FLOAT32, T>(data_ptr, value);
           break;
         case sensor_msgs::PointField::FLOAT64:
           writePointCloud2BufferValue<sensor_msgs::PointField::FLOAT64, T>(data_ptr, value);
           break;
       }
     }
 }
 
 #endif /* point_field_conversion.h */

I think are related to output type msg type from ros

How I can change the these types as given in:

[image]

MartyG-RealSense, 2020-09-26:
I could not find any librealsense references related to **jsk_pcl_ros_utils**.  If you are aiming to generate an xyzrgb point cloud through RealSense ROS, I wonder if it may be easier to use **depth_image_proc/point_cloud_xyzrgb**.

http://wiki.ros.org/depth_image_proc#depth_image_proc.2Fpoint_cloud_xyzrgb
https://github.com/IntelRealSense/realsense-ros/issues/277#issuecomment-365368917

Regarding your question about whether the sensor msgs highlighted in red above can have their type changed, I will refer that question to @doronhi the RealSense ROs wrapper developer.

closed 2020-10-12 by MartyG-RealSense