realsenseai/realsense-ros #1391: Cannot set the use_sim_time parameter on the realsense2_node node.

garyrightbot, 2020-09-16:
I started a simulation for a robot in an empty world and wanted to test my navigation obstacle detection using ```VoxelLayer``` plugin (nav2 stack) with the Realsense D435i camera.

However, despite passing the ```use_sim_time:=True``` parameter, the msgs still have epoch stamp.

```ros2 run realsense2_node realsense2_node use_sim_time:=True```

garyrightbot, 2020-09-16:
More info on the same:

My launch file where I start gazebo as well as the realsense node:

```
def generate_launch_description():
    use_sim_time = LaunchConfiguration('use_sim_time', default='true')
    world_file_name = 'picker_empty/' + 'empty.world'
    world = os.path.join(get_package_share_directory('turtlebot3_gazebo'), 'worlds', world_file_name)
    launch_file_dir = os.path.join(get_package_share_directory('picker'), 'launch')
    lidar_launch_file_dir = os.path.join(get_package_share_directory('rplidar_ros'), 'launch')
    realsense_launch_file_dir = os.path.join(get_package_share_directory('realsense2_camera'), 'launch')

    return LaunchDescription([
        ExecuteProcess(
            cmd=['gazebo', '--verbose', world, '-s', 'libgazebo_ros_init.so'],
            output='screen'),
            
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource([launch_file_dir, '/picker_state_publisher.launch.py']),
            launch_arguments={'use_sim_time': use_sim_time}.items(),
        ),
        
	IncludeLaunchDescription(
	    PythonLaunchDescriptionSource([lidar_launch_file_dir, '/rplidar.launch.py']),
	),
	
	IncludeLaunchDescription(
	    PythonLaunchDescriptionSource([realsense_launch_file_dir, '/rs.launch.py']),
	),
  
    ])
```

My rs.launch.py file:

```

garyrightbot, 2020-09-17:
Was able to solve this in the ```ros2_intel_realsense``` package by performing the below steps. Couldn't quite do so in this one as it was causing the node to crash:

https://github.com/intel/ros2_intel_realsense/issues/157

doronhi, 2020-09-18:
@garyrightbot ,
Thanks for testing and notifying. I tried your suggestion and it didn't crush for me. I did notice that the timestamps were wrong whether you use `use_sim_time` or not so I fixed the conversion from milliseconds to nanoseconds as well and now the timestamps seem ok to me.
Would you care to test PR #1403 please?

garyrightbot, 2020-09-19:
Hi @doronhi I noticed that you have raised the PR for eloquent while I am using foxy. I'll replicate the changes locally on foxy in some time and let you know.

RealSenseSupport, 2020-09-27:
@garyrightbot ROS wrapper for eloquent now released - https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.1  Could you please have a try to see if you still have the issue? Thanks!

RealSenseSupport, 2020-10-10:
@garyrightbot  Any update from your side? Thanks!

RealSenseSupport, 2020-10-27:
@garyrightbot Do you still have any questions or issue for this? Looking forward to your update. Thanks!

doronhi, 2020-10-28:
The ros2_intel_realsense package is on its was to be replaced by realsense_ros and by now can already, I think, can be considered obsolete.

RealSenseSupport, 2020-11-06:
@doronhi Thanks for your clarification! 
@garyrightbot Sorry that we didn't hear from you for weeks. Close it at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-11-06 by RealSenseSupport