realsenseai/realsense-ros #1395: Image problems (depth map, rendering map, point cloud map)

dingbaiyin, 2020-09-17:
1. ubuntu  Pi3B +
2.librealsense2.16.1
3.D430
After the depth map is rendered, it can be seen that the target in the image has obvious non-edge boundaries, showing a semi-transparent state. It can be seen from the point cloud that the corresponding position is concave.
No filtering function is added when capturing images.
Many of the same situations occur in our products, please help analyze the cause of the problem.
The following are depth map, rendering map and point cloud map.
[image]
[image]

[CSB017_314_205_Pig_0_31.00_2020-09-10_15-22-31_pointcloud.zip](https://github.com/IntelRealSense/realsense-ros/files/5235625/CSB017_314_205_Pig_0_31.00_2020-09-10_15-22-31_pointcloud.zip)

MartyG-RealSense, 2020-09-17:
Hi @dingbaiyin  Your image has similarities to a recent case on the librealense GitHub where the edges seem to 'bleed' instead of being sharpy defined.

https://github.com/IntelRealSense/librealsense/issues/7021

[image]

A few improvement suggestions that emerged from that discussion were:

-  Using the **High Density** visual preset (or Medium Density if a balance between image quality and accuracy is needed).
-  Increasing the **Laser Power** value, which was reported by the RealSense user in that case to make surfaces less "blobby"
-  Applying the Advanced Mode function **Second Peak Threshold** (found under the **Depth Controls** sub-category of the **Advanced Controls** options in the RealSense Viewer's side-panel) to replicate the depth confidence-matching that the SR30x cameras have.

dingbaiyin, 2020-09-18:
Thank you very much for your suggestion. At the same time, when I was testing on site, I found that the question I asked had a lot to do with camera exposure. The effect is improved after blocking the sun's rays. I will test your suggestions to achieve better results.

MartyG-RealSense, 2020-09-26:
Hi @dingbaiyin   Do you have an update for us about the tests that you were going to perform, please?  Thanks!

dingbaiyin, 2020-09-27:
No,thanks,I will colose the  issue.

MartyG-RealSense, 2020-09-27:
Thanks very much, I shall close it for you.

closed 2020-09-27 by MartyG-RealSense