realsenseai/realsense-ros #1397: how could I fix this error? when I install the realsense-ros,and I run catkin_make clean ,then I got the error like this.

1556260456, 2020-09-17:
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - realsense_camera_msgs (plain cmake)
-- ~~  - realsense2_camera (plain cmake)
-- ~~  - realsense2_node (plain cmake)
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:67 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/adam/Dektop/realsense_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/adam/Dektop/realsense_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

MartyG-RealSense, 2020-09-17:
Hi, your problem has similarities to the one mentioned in the link below:

https://github.com/IntelRealSense/realsense-ros/issues/395#issuecomment-398310122

The title of your case informs me that you are trying to install the RealSense ROS wrapper.  Assuming that you are doing so by building it from source code as the use of CMake suggests, may I confirm please that the steps below have been completed:

1.  The librealsense SDK is installed and working (for example, the RealSense Viewer program wors without problems).

2.  You have successfuly installed ROS Kinetic, as your logs in your opening comment suggest that Kinetic is the branch of ROS that you are using.

3.  If steps 1 and 2 are completed and you are ready to install the RealSense ROS wrapper, are you using the instructions in the wrapper's dcocumentation for building the wrapper from source, please?  This involves creating a new empty catkin workspace called **catkin_ws**.

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

denesh-botsync, 2020-09-18:
-- ~~ - realsense_camera_msgs (plain cmake)
-- ~~ - realsense2_camera (plain cmake)
-- ~~ - realsense2_node (plain cmake)

If you are trying to catkin_make clean these packages from ROS1 Kinetic, it probably won't work (these packages are meant for ROS2). Follow the steps in https://github.com/IntelRealSense/realsense-ros but omit the line: **git checkout `git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1`** and it should work

doronhi, 2020-09-18:
I would like to emphasize what @denesh-botsync already mentioned: These packages are meant for ROS2-eloquent. `catkin` is not the build tool. If you move from ROS1 to ROS2 you should begin your work on a different environment and cleaning of the ROS1 version could be done using the ROS1 source code.

1556260456, 2020-09-18:
@doronhi @MartyG-RealSense 
thanks for your help and reply my answr ,I think the problem is that :when I install and build the **librealsense**,I used:
[image],and run **cmake ../ -DBUILD_EXAMPLES=true**,which changed the **Cmakelists**。today,I found a demo on internet which **Cmakelists**is like this:`cmake_minimum_required(VERSION 3.1.0)

project(alldata)

set( CMAKE_CXX_COMPILER "g++" )
set( CMAKE_BUILD_TYPE "Release" )
set( CMAKE_CXX_FLAGS "-std=c++11 -O3" )
# Enable C++11
set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED TRUE)

FIND_PACKAGE(OpenCV REQUIRED)
FIND_PACKAGE(realsense2 REQUIRED)

include_directories( ${OpenCV_INCLUDE_DIRS} )
include_directories( ${realsense2_INCLUDE_DIRS})
add_executable(alldata GetAlldata.cpp)

target_link_libraries( alldata ${OpenCV_LIBS} ${realsense2_LIBRARY})`
It is defferent from the librealsense demo `# License: Apache 2.0. See LICENSE file in root directory.
# Copyright(c) 2019 Intel Corporation. All Rights Reserved.
#  minimum required cmake version: 3.1.0
cmake_minimum_required(VERSION 3.1.0)

project(RealsenseExamplesHelloRealSense)

add_executable(rs-hello-realsense rs-hello-realsense.cpp)
set_property(TARGET rs-hello-realsense PROPERTY CXX_STANDARD 11)
target_link_libraries(rs-hello-realsense ${DEPENDENCIES})
include_directories(rs-hello-realsense)
set_target_properties (rs-hello-realsense PROPERTIES FOLDER "Examples")

install(TARGETS rs-hello-realsense RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})`
the Cmakelists which I found from internet make the librealsense demo work normally.
I do not know cmake document very well ,but I think the problem is that!

doronhi, 2020-09-20:
@1556260456 , I'm not sure what the issue with building librealsense2 is. In your first comment I saw you got the following message:
"This workspace contains non-catkin packages in it, and catkin cannot build"

This means that librealsense2 was in the catkin_ws directory. It also means you tried to build librealsense2 using catkin_make which is not possible and the message clearly says so. It also says that using `catkin_make_isolated` will solve the issue (again, on ROS1. Download the ROS1 wrapper if you want to use ROS1).

If by now librealsense2 is not in the catkin_ws directory and `make` command fails, you can post the build error you get and then someone will be able to address those issues. I think that in that case, it will be beneficial to start a separate issue on the [librealsense2 repository](https://github.com/IntelRealSense/librealsense)

MartyG-RealSense, 2020-09-29:
Hi @1556260456   Do you still require assistance with this case, please?  Thanks!

SaiHaneeshAllu, 2020-09-29:
> -- ~~ - realsense_camera_msgs (plain cmake)
> -- ~~ - realsense2_camera (plain cmake)
> -- ~~ - realsense2_node (plain cmake)
>
> If you are trying to catkin_make clean these packages from ROS1 Kinetic, it [...]

This worked for me. Thanks

MartyG-RealSense, 2020-09-29:
Thanks very much @SaiHaneeshAllu 

@1556260456  Is the above advice by @SaiHaneeshAllu help ful for you please?

MartyG-RealSense, 2020-10-17:
Case closed due to no further comments received.

closed 2020-10-17 by MartyG-RealSense

Ericloving, 2021-12-02:
> -- ~~ - realsense_camera_msgs (plain cmake) -- ~~ - realsense2_camera (plain cmake) -- ~~ - realsense2_node (plain cmake)
>
> If you are trying to catkin_make clean these packages from ROS1 Kinetic, it [...]

I did like this but the problem still be there. librealsense 2.48.0 librealsense ros 3.2.2.   ros metric I don't know the reason

gurselturkeri, 2022-06-24:
> -- ~~ - realsense_camera_msgs (plain cmake) -- ~~ - realsense2_camera (plain cmake) -- ~~ - realsense2_node (plain cmake)
>
> If you are trying to catkin_make clean these packages from ROS1 Kinetic, it [...]

Thank you. this work for me

jason-lee-webdev, 2024-06-25:
> -- ~~ - realsense_camera_msgs (plain cmake) -- ~~ - realsense2_camera (plain cmake) -- ~~ - realsense2_node (plain cmake)
>
> If you are trying to catkin_make clean these packages from ROS1 Kinetic, it [...]

Thank you. It was a great help to me.