realsenseai/realsense-ros #1398: How to enbale confidence map for L515?

Pomevak, 2020-09-17:
I have found that the confidence map for L515 can be very helpful, and I want to record it together with the depth map. However, it seems that realsense-ros doesn't publish it as a topic, or I didn't find any documentation on it. So I wonder if there is any way that I could get the confidence maps published so that I can use them in my application. Thanks.

doronhi, 2020-09-18:
There is no way yet. I'll attend to it as soon as possible.

RealSenseSupport, 2021-01-03:
Hi,

Will you be needing further help with this? This is now supported as of the latest release:

https://github.com/IntelRealSense/realsense-ros/releases

*If we don’t hear from you in 7 days, this issue will be closed.*

Thanks

closed 2021-01-11 by RealSenseSupport

doronhi, 2021-01-12:
I would like to point out that the option to publish confidence map for L515 is available since version 2.2.21 for ROS1 and since version 3.1.3 for ROS2-foxy. The topics `/camera/confidence/image_rect_raw` and `/camera/confidence/camera_info` are made available using the launch parameters `enable_confidence`, `confidence_width`, `confidence_height` and `confidence_fps`.